eclipse-mraa/mraa #547: g++ v4.8.2 (Intel Edison) error: 'Result' not a member of 'mraa'

bhorn, 2016-07-21:
When I try to compile some code using MRAA on my Edison g++ tell's me there is no 'Result' in namespace mraa.

Compiling (NOT LINKING!) with: 
`g++ -std=gnu++11 -Wall -c GPIO.cpp -o GPIO.o`

Failes with:
> GPIO.cpp:57:7: error: 'Result' is not a member of 'mraa'
`mraa::Result response = gpio->dir(mraa::DIR_OUT);`

The same code compiles fine on my Mac and with Bitbake/Crosscompiling. But on Target it failes.

arfoll, 2016-07-23:
Can you post your full code as a gist? At a glance I'm a bit confused

alext-mkrs, 2016-07-24:
@bhorn indeed please provide your code - I've just tried the [c++ blink example](https://github.com/intel-iot-devkit/mraa/blob/master/examples/c%2B%2B/Blink-IO.cpp), which has the same string as the one in your error message (so I guess you took the example and modified it?) - and it compiled just fine:
```bash
root@edison1:~# g++ -std=gnu++11 -Wall -c mraa-test-cpp.cpp -o mraa-test-cpp.o
root@edison1:~# opkg list-installed |grep mraa
mraa - 1.1.2
```

bhorn, 2016-07-25:
It's hard to post some Code. Basically it's just the c++ blinkio example. 
I can't even compile the example. Not the blinkio, not the pwm.
Same error every time.

```

root@edison:~/devel/mraa/examples/c++# g++ -std=gnu++11 -Wall Blink-IO.cpp -o io
e++.cpp: In function 'int main(int, char**)':
e++.cpp:60:16: error: 'ERROR_UNSPECIFIED' is not a member of 'mraa'
         return mraa::ERROR_UNSPECIFIED;
                ^
e++.cpp:62:5: error: 'Result' is not a member of 'mraa'
     mraa::Result response = gpio->dir(mraa::DIR_OUT);
     ^
e++.cpp:62:18: error: expected ';' before 'response'
     mraa::Result response = gpio->dir(mraa::DIR_OUT);
                  ^
e++.cpp:63:9: error: 'response' was not declared in this scope
     if (response != mraa::SUCCESS) {
         ^
e++.cpp:63:21: error: 'SUCCESS' is not a member of 'mraa'
     if (response != mraa::SUCCESS) {
                     ^
e++.cpp:69:9: error: 'response' was not declared in this scope
         response = gpio->write(1);
         ^
e++.cpp:75:12: error: 'response' was not declared in this scope
     return response;
            ^
``` 
MRAA got installed via Yocto.

arfoll, 2016-07-25:
Essentially it looks so wrong that either I'd guess there's an error
previous to that one that is causing g++ to go crazy. Or you have an issue
with the headers. What headers are you including and are they there &
contain the right stuff? Also what version of mraa is this?

bhorn, 2016-07-25:
I use the mraa.hpp like in the example. The headers are in /usr/include on my Edison (/usr/include/mraa/ is there, too).
This mraa.hpp includes 
```
#include "mraa/common.hpp"
#include "mraa/pwm.hpp"
#include "mraa/aio.hpp"
#include "mraa/gpio.hpp"
#include "mraa/i2c.hpp"
#include "mraa/spi.hpp"
#include "mraa/uart.hpp"
```
All the enums are defined in types.h, right? Shouldn't be there a types.hpp of some sort with the enums in a `namespace mraa{enum Result {[...]};[...]}`?
Like I said, everything was installed by Yocto (using the official Intel-Sources).

arfoll, 2016-07-25:
Try get the version with:
`Is -l /usr/lib/libmraa*`

There's lots of yocto based images flying around what does `cat
/etc/os-release` say?

Types.hpp gets included from those headers see common.hpp and others so
that's why it's not in mraa.hpp it's just not necessary.

This is really odd, can you try compile mraa from git on the board or
compile an example verbatim?

alext-mkrs, 2016-07-25:
It might be an older version. What does opkg command like the one in my previous post return on your board?

bhorn, 2016-07-25:
Oh. `/usr/lib/libmraa.so.0.6.1`.
So the Yocto-Reciepe installs a (very) old Version. I'll update the recipes...
This is a bit embarrassing. I should have checked this first.

closed 2016-07-25 by bhorn