eclipse-mraa/mraa #549: MRAA on Win 10

m-obi, 2016-07-22:
Hello,

I have a LattePanda. That's a board with an X-Z8300 on it and the OS is Win 10 64 Bit.
Is MRAA ready for Win 10?

My IDE is Qt Creator and I will program in C++.

arfoll, 2016-07-22:
Unfortunately no mraa on windows currently only work in `mock` mode. Its on the to-do list but as of yet no clear timeline. Lattepanda looks like a very interesting board though it seems like it would be a good target for mraa on windows.

m-obi, 2016-07-22:
I will try with OpenFile and DeviceIoControl. Does Intel have any h-Files with the Constants? And which sys-File I have to use? iai2c.sys, iai2ce.sys, iaLPSS2i_I2C.sys or iaLPSSi_I2C.sys? Or I have to access over the interface GUID?

alext-mkrs, 2016-07-24:
As Windows has a very different HAL than the one mraa is used to on Linux (the sysfs), I think you'll need to figure it out by digging into Win API docs. And that's going to be quite different from th existing one.

alext-mkrs, 2016-08-03:
I did some reading on LattePanda and it turns out that's a completely different setup than I originally thought of. It's a normal Atom-based computer with ATmega32u4 core snapped onto the board and communicating with the software on tha main core (Arduino IDE) by means of Firmata library. That's quite a different setup and use case than mraa is about.

There are several I2C connections coming directly from the Atom SoC, which *theoretically* could work with mraa if proper HAL existed, but looks like the OS on LP does not support enabling those at all, see [this thread](http://www.lattepanda.com/forum/viewtopic.php?f=6&t=261&start=0)

All in all, it seems to me that this specific question is about the board, which has no means of enabling mraa even if we prepared a proper HAL - because of two facts
1. It has non-IoT edition of Windows (thus no respective enabling layers/HAL we could use for HAL)
1. Its "IoT" capabilities (except a couple of I2C conectors) are coming from the actual ATmega32u4 MCU, which mraa does not handle by design.

m-obi, 2016-08-03:
I know the thread. I'm m-obi.

But I found the driver for the I2C in the Device Manager.
[image]
So I think, it will work.

The  ATmega32u4 its just an Arduino Leonardo. I know thats mraa doesnt handle this.

So I'm still working on this and this issue is not closed for me. But first I have to wait for the 1,8V-5V-Levelshifter.

alext-mkrs, 2016-08-04:
Well, having a driver is good, but unless there's an API for it, it's unlikely to be helpful.

But if you are anyway going to check the topic out and potentially contribute back to mraa (in one way or the other, e.g. with information and not necessarily coding), that's most welcome, let me reopen this one as an enhancement.

sjournaux, 2018-12-04:
you can use SPB framework
https://docs.microsoft.com/en-us/windows-hardware/drivers/spb/spb-framework-extension
and 
https://docs.microsoft.com/en-us/windows/uwp/devices-sensors/enable-usermode-access