eclipse-mraa/mraa #550: MRAA 0.9.0 Version C Files In Eclipse IDE Luna SR2 (4.4.2)

cnvery, 2016-07-22:
Hi all,

I wish to find MRAA 0.9.0 C Files as per listed below:
> aio.c
> common.c
> gpio.c
> i2c.c
> iio.c
> iio_kernel_headers.c
> ioctl.c
> mraa.c
> mraa_adv_func.c
> mraa_internal_types.c
> pwm.c
> spi.c
> types.c
> uart.c

I dont face any issues on compiling the C codes in Eclipse IDE.

Now, I wish compile the project itself with these C files with Microsoft Azure IoT C SDK on Intel Edison without utilizing Eclipse IDE.

However, current Github MRAA version is 1.1.2, not 0.9.0.

Thus, I need someone post these files for my further action.

Thank you!

arfoll, 2016-07-22:
I'm not entirely sure how having the c files will help you, end of the day headers you cross compile against will have to match the dynamic library on Edison. Or otherwise compile mraa statically and link against it that way?

In any case if you want v0.9.0 simply use git tags. `git checkout v0.9.0` will give you what you want. You can also download tarballs of all the versions by just hitting the releases tab on github.

cnvery, 2016-07-22:
Hi @arfoll,

I actually compile the C code in 64-bit Ubuntu 15.10 version due to issues of [Microsoft Azure IoT Hub SDK](https://github.com/Azure/azure-iot-sdks/issues/435).

Then transfer the executable file from Ubuntu to Intel Edison via SSH.

Hence, I need to compile mraa statically on same folder in Ubuntu.

Anyway, I found the mraa v0.9.0 in this [link](https://github.com/intel-iot-devkit/mraa/tree/v0.9.0), suggested by you.

Thanks!

cnvery, 2016-07-22:
Hi @arfoll,

I tried to compile C code in Ubuntu, but it failed.

The same code is successfully complied in Eclipse IDE, but not for cross compiling in Ubuntu.

The error is shown as per [this image link](https://s32.postimg.org/h5r0thdw5/123213.jpg).

> /home/osboxes/Source/azure-iot-sdks/c/serializer/samples/sample2/gpio.c: In function 'mraa_gpio_init_internal':
> /home/osboxes/Source/azure-iot-sdks/c/serializer/samples/sample2/gpio.c:75:18: warning: passing argument 1 of 'dev->advance_func->gpio_init_internal_replace' makes pointer from integer without a cast
>          status = dev->advance_func->gpio_init_internal_replace(pin);
>                   ^
> /home/osboxes/Source/azure-iot-sdks/c/serializer/samples/sample2/gpio.c:75:18: note: expected 'mraa_gpio_context' but argument is of type 'int'
> /home/osboxes/Source/azure-iot-sdks/c/serializer/samples/sample2/gpio.c:75:18: error: too few arguments to function 'dev->advance_func->gpio_init_internal_replace'
> serializer/samples/sample2/CMakeFiles/sample2.dir/build.make:110: recipe for target 'serializer/samples/sample2/CMakeFiles/sample2.dir/gpio.c.o' failed
> make[2]: *** [serializer/samples/sample2/CMakeFiles/sample2.dir/gpio.c.o] Error 1
> CMakeFiles/Makefile2:2557: recipe for target 'serializer/samples/sample2/CMakeFiles/sample2.dir/all' failed
> make[1]: *** [serializer/samples/sample2/CMakeFiles/sample2.dir/all] Error 2
> Makefile:138: recipe for target 'all' failed
> make: *** [all] Error 2

Any suggestion on this?

Thanks in advance.

arfoll, 2016-07-24:
Ok so you're copying mraa's sourcefiles into a new project in the azure-iot-sdk? Without telling you what to do, that's not the way I would do it.

Mraa is meant to be used as library, you build it as a .so and then link your application against it with the linker. At runtime the loader will resolve the link on your board and use the onboard lib. That's how pretty much everything in Linux works. You can also compile mraa as a static library .a and include it in your binary. This generates a 'fat' binary with everything included.

Copying random bits of C in will just cause you massive pain, I'm not sure what your exact issue is there but likely you have diffeernt versions of stuff because gpio_init_internal_replace takes a mraa_gpio_context and an int as arguments and the version of gpio.c you have seems to not do that. I'm guessing you've got some files from v0.9 and others from master.

cnvery, 2016-07-25:
Hi @arfoll,

Thanks for pointing the issue on different files for mraa. I have tried again using same version but still unable to make it works.

Then, I found some guidelines on compile mraa using **cmake** from [this link](https://github.com/intel-iot-devkit/mraa/issues/303).

Since I am using [Intel Edison SDK - Cross Compile Tools](https://software.intel.com/en-us/iot/hardware/edison/downloads), I found that actually there is mraa library folder under _/opt/opky-edison/1.7.2/sysroots/core2-32-poky-linux/usr/include_.

Finally, I am able to compile the project with following **CMakeLists.txt** configuration:
> #Copyright (c) Microsoft. All rights reserved.
> #Licensed under the MIT license. See LICENSE file in the project root for full license information.
> 
> #this is CMakeLists.txt for sample2
> 
> compileAsC99()
> 
> if(NOT ${use_amqp})
> 	message(FATAL_ERROR "Test being generated without amqp support")
> endif()
> 
> set(sample2_c_files
> Application.c
> Utility.c
> )
> 
> set(sample2_c_files ${sample2_c_files} ./Main.c)
> 
> set(sample2_h_files
> Application.h
> Utility.h
> )
> 
> include_directories(. ${SHARED_UTIL_INC_FOLDER} ${IOTHUB_CLIENT_INC_FOLDER})
> link_directories(${whatIsBuilding}_dll ${SHARED_UTIL_LIB_DIR})
> 
> add_executable(sample2 ${sample2_c_files} ${sample2_h_files})
> 
> target_link_libraries(sample2 serializer iothub_client iothub_client_amqp_transport mraa)
> 
> linkSharedUtil(sample2)
> linkUAMQP(sample2)

You may close this issue.

Thanks for your great assistance!

closed 2016-07-25 by cnvery