eclipse-mraa/mraa #551: Intel Edison Arduino-Board gpio214 missing, no PWM possible

bhorn, 2016-07-26:
I can't use PWM on my Intel Edison Arduino board. The TRI_STATE-GPIO is not exportable (No such device). 
This triggers libmraa to think it's running on an Intel Edison Miniboard.

Tried running the PWM-Example:
```
root@edison:~/devel/mraaexp# ./pwm 
terminate called after throwing an instance of 'std::invalid_argument'
  what():  Error initialising PWM on pin
Aborted
root@edison:~/devel/mraaexp# journalctl -f
-- Logs begin at Sat 2000-01-01 00:00:14 UTC. --
Jul 26 09:09:48 edison libmraa[299]: libmraa version v1.1.2 initialised by user 'root' with EUID 0
Jul 26 09:09:48 edison libmraa[299]: gpio214: init: Failed to write to 'export': No such device
Jul 26 09:09:48 edison libmraa[299]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Jul 26 09:09:48 edison libmraa[299]: libmraa initialised for platform 'Intel Edison' of type 2
Jul 26 09:09:48 edison libmraa[299]: pwm_init: pin 6 not capable of pwm
```

What causes the TRI_STATE-GPIO to be messing?

bhorn, 2016-07-26:
Ok, Arduino/Board broken. I have shut down the Edison 3 times and made it powerless (to avoid the shield-reset-bug).
Applying my Image and Software to a brand new Edison and everything is fine.

Well, thats my 3. dead Edison. First one failed Barrel-Power, second one every GPIO-Pin and this one broke TRI_STATE...

closed 2016-07-26 by bhorn

arfoll, 2016-07-26:
First off really sorry about your boards. Whilst the first one sounds dead, arduino breakouts are known to crash or enter weird states leading to this tristate issue. Try plugging it back in after a few hours when the power is completely drained and it may come back to life. The tristate gpio missing usually means the gpio expander (used for muxing) has crashed and on a 'warm' reboot can essentially keep this crashed state.

Also please return your board and ask for a new one here: https://customercare.intel.com. If you have any issues cc me.