eclipse-mraa/mraa #555: npm install error "fatal error: json-c/json.h: No such file or directory"

deadprogram, 2016-08-03:
I am unable to install the mraa 1.2.0 release under Ubuntu 14.04 using Node.js 4.2.2

The following error output is received:

```
../src/json/jsonplatform.c:31:25: fatal error: json-c/json.h: No such file or directory
 #include <json-c/json.h>
                         ^
compilation terminated.
make: *** [Release/obj.target/mraa/src/json/jsonplatform.o] Error 1
```

Thanks!

Hbrinj, 2016-08-03:
Also are you possibly compiling the json platform loading version?

Hbrinj, 2016-08-03:
Do you have json c installed? Are you by Chance trying to using imraa? How are you running cmake?

arfoll, 2016-08-03:
> Do you have json c installed? How are you running cmake?
>
> Sent from my iPhone

Try 1.2.1 😋

tingleby, 2016-08-04:
Node 6 seems good at 1.2.1 too!
```
tom@heim ~/test-npm % node -v
v6.3.1
tom@heim ~/test-npm % npm -v
3.10.5
```

alext-mkrs, 2016-08-17:
This should be fixed and re-fixed :smiley: in 1.2.3, so closing this one.

closed 2016-08-17 by alext-mkrs