eclipse-mraa/mraa #556: Python bindings fail to load with latest MRAA on WRLinux

Propanu, 2016-08-04:
Fresh build on WRLinux with:
* SWIG 3.0.10
* Cmake 3.6.1
* GCC 4.9.1
* Python 2.7.3

Getting this upon init:
```
Python 2.7.3 (default, Jan  1 2016, 03:45:12)
[GCC 4.9.1] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/mraa.py", line 182, in <module>
    """
AttributeError: 'module' object has no attribute 'INTEL_GALILEO_GEN1_swigconstant'
```
The message though is very likely misleading. I thought SWIG was the issue and tried rebuilding with 3.0.8, then 3.0.2. Loading will still fail, albeit on a different type enum every time the version changed.

arfoll, 2016-08-04:
Check the linker resolves mraa.so in your python site dir to the correct
/usr/lib/libmraa.so.1.0.0 and that's there's no other versions around.

If you can email me some login details to that box would be curious to have
a look.

Hbrinj, 2016-08-04:
I've got wrl setup will try to replicate.

arfoll, 2016-08-05:
Houman was confused. He does not have wind river linux (WRLinux). Apparently windows runtime linux is also WRLinux :100:

Propanu, 2016-08-05:
Linker seems to resolve fine, I have a stale 1.1.2 but I double checked from Node.js and when I init MRAA it picks up the right version:
```
root@WR-IDP-3E23:~# ldconfig -p | grep mraa
        libmraajava.so (libc6,x86-64) => /usr/lib64/libmraajava.so
        libmraa.so.1 (libc6,x86-64) => /usr/lib64/libmraa.so.1
        libmraa.so (libc6,x86-64) => /usr/lib64/libmraa.so
root@WR-IDP-3E23:~# ls -la /usr/lib64/libmraa*
lrwxrwxrwx 1 root root     12 Aug  5 01:01 /usr/lib64/libmraa.so -> libmraa.so.1
lrwxrwxrwx 1 root root     16 Aug  5 04:51 /usr/lib64/libmraa.so.1 -> libmraa.so.1.2.1
-rwxr-xr-x 1 root root 244454 Aug  5 00:36 /usr/lib64/libmraa.so.1.1.2
-rwxr-xr-x 1 root root 244641 Aug  5 05:55 /usr/lib64/libmraa.so.1.2.1
-rwxr-xr-x 1 root root 163890 Aug  5 05:55 /usr/lib64/libmraajava.so
```
Unfortunately this happened on one of the demos and it has to be shipped out so I will setup another WR box in a similar fashion to see if I can reproduce again.

Propanu, 2016-08-05:
And:
```
root@WR-IDP-3E23:~# ldd /usr/lib/python2.7/site-packages/_mraa.so
ldd: warning: you do not have execution permission for `/usr/lib/python2.7/site-packages/_mraa.so'
        linux-vdso.so.1 (0x00007fff08d46000)
        libpython2.7.so.1.0 => /usr/lib64/libpython2.7.so.1.0 (0x00007f80a02d1000)
        libmraa.so.1 => /root/mraa/build/src/libmraa.so.1 (0x00007f80a009c000)
        libpthread.so.0 => /lib64/libpthread.so.0 (0x00007f809fe80000)
        libdl.so.2 => /lib64/libdl.so.2 (0x00007f809fc7c000)
        libstdc++.so.6 => /usr/lib64/libstdc++.so.6 (0x00007f809f971000)
        libm.so.6 => /lib64/libm.so.6 (0x00007f809f66d000)
        libgcc_s.so.1 => /lib64/libgcc_s.so.1 (0x00007f809f45b000)
        libc.so.6 => /lib64/libc.so.6 (0x00007f809f0b8000)
        libutil.so.1 => /lib64/libutil.so.1 (0x00007f809eeb5000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f80a08d8000)
```
So things look right.

alext-mkrs, 2016-08-05:
Looks pretty much like it. Have you set any RPATH-related CMake options? It looks like you've used `make install`, so by default (as described [here](https://cmake.org/Wiki/CMake_RPATH_handling#Default_RPATH_settings)) it should have removed the RPATH upon installation... Maybe try setting it manually?

alext-mkrs, 2016-08-18:
@Propanu, is this still actual and have you had a chance to check those RPATH settings?

Propanu, 2016-08-19:
Sorry I was busy with IDF. It's still an issue and it started when we decided to stop doing static linking for the bindings a while back. Seems like it shows up on occasion in certain build environments. I'm very familiar with the CMake page describing RPATH settings and most likely, the proper way to fix it is to set the RPATH during build based on the install path rather than relying on default settings. Will look into it some more when I get back in the office.

alext-mkrs, 2016-09-06:
@Propanu, anything on this one?

Propanu, 2016-09-06:
I think I found a fix, adding `BUILD_WITH_INSTALL_RPATH TRUE` to the python targets seems to do it. Will test and will submit a PR sometime this week.

Propanu, 2016-09-08:
Submitted #575, looks OK with it.
Before
```
mtpanu@flax:/project/mtpanu/IoT/mraa/build$ chrpath lib/python2.7/site-packages/_mraa.so
lib/python2.7/site-packages/_mraa.so: RPATH=/project/mtpanu/IoT/mraa/build/src
```
After
```
mtpanu@flax:/project/mtpanu/IoT/mraa/build$ chrpath lib/python2.7/site-packages/_mraa.so
lib/python2.7/site-packages/_mraa.so: no rpath or runpath tag found.
```
As far as I recall only the Python bindings had this issue. Also, I only tested this with 2.7 but it all works as expected now.

alext-mkrs, 2017-04-08:
This is already fixed with #575

closed 2017-04-08 by alext-mkrs