eclipse-mraa/mraa #557: JSON and replace functions

alext-mkrs, 2016-08-07:
@Hbrinj, after talking to @arfoll I wanted to add a JSON example file for Mock platform. But looks like that's not a supported case as only pin definitions are being set and all the replace functions are left out (as naturally there isn't a way to specify that in the JSON file).

So the question I have is - is this functionality only targeted at boards, which do not require replace functions or is there any plan to add this piece later, somehow?

That's what I get when running an I2C read byte data test (generally that could be any of them) after building with both mock andb JSONPLAT enabled:
```syslog
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: libmraa version v1.2.1 initialised by user 'alt' with EUID 1000
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: init_json_platform: SPI wasn't found in the json file
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: init_json_platform: UART wasn't found in the json file
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: init_json_platform: PWM wasn't found in the json file
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: init_json_platform: Platform MRAA mock platform initialised via json
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: libmraa initialised for platform 'MRAA mock platform' of type 97
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: i2c_init: Selected bus 0
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: i2c0_init: Failed to open requested i2c port /dev/i2c-0: No such file or directory
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: i2c_init: Selected bus 0
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: i2c0_init: Failed to open requested i2c port /dev/i2c-0: No such file or directory
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: i2c_init: Selected bus 0
Aug 07 16:23:44 bl-fed21.md.local libmraa[25312]: i2c0_init: Failed to open requested i2c port /dev/i2c-0: No such file or directory
```

BTW, some documentation on the JSON file format would be quite nice to have as I had to have both the Turbot example file and `jsonplatform.c` in front of me when preparing my one :smiley:

Hbrinj, 2016-08-08:
@alext-mkrs, initially this was aimed towards simple boards, it would be quite useful if we could develop a nice way to handle more complex boards. With regards to the documentation, i apologise i hadn't had time recently to create the docs but it's on my list to do before the end of this week.

arfoll, 2016-08-09:
That is a very good point, I hadn't thought of that at all when we talked about mraa mock w/jsonplat. Sorry! The problem is I have no idea how to make replace functions work without massive hacks in json. This really only works for 'simple' boards with no advance functions.

alext-mkrs, 2016-08-09:
No worries and thank you both for comments. I understood it quite that way, but just wanted to discuss it so that anyone could refer to this in the future. And I guess that would also be in the docs :smiley:

alext-mkrs, 2016-08-17:
Just to make it clear - let's keep this open until docs are added, so @Hbrinj you can close it from the commit adding them.

alext-mkrs, 2016-08-23:
Docs added in 0630c5f5050eaff18cbd2f96f7fd0707a8ab8503, closing this one.

closed 2016-08-23 by alext-mkrs