eclipse-mraa/mraa #558: npm install fail on Edison Yocoto 3.10.98 node v4.4.4

windmillknight, 2016-08-09:
Hello 
      When I instill mraa use npm install mraa on my edison .it said that 

mraa.target.mk:135: recipe for target 'Release/obj.target/mraa/src/json/jsonplatform.o' failed
make: *** [Release/obj.target/mraa/src/json/jsonplatform.o] Error 1
make: Leaving directory '/home/root/node_modules/mraa/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 3.10.98-poky-edison+
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/root/node_modules/mraa
gyp ERR! node -v v4.4.4
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
npm ERR! Linux 3.10.98-poky-edison+
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "mraa"
npm ERR! node v4.4.4
npm ERR! npm  v2.15.1
npm ERR! code ELIFECYCLE

npm ERR! mraa@1.2.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@1.2.1 install script 'node-gyp rebuild'.
npm ERR! This is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs mraa
npm ERR! Or if that isn't available, you can get their info via:
npm ERR! 
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/root/node_modules/npm-debug.log

what can I do ?please help me

arfoll, 2016-08-09:
Weird, this should have been fixed with 1.2.1, 1.2.2 just got pushed and should fix this by simply removing jsonplat.o from being built. completely. It's just been pushed.

If you're on an edison note that latest mraa can be downloaded via opkg rather than npm which is better since it updates the full mraa pkg not just install a freestanding npm module. opkg update && opkg upgrade should work as long as the intelgalactic repo is installed correctly.

windmillknight, 2016-08-09:
thanks for your help!
but what happen now?

Installing mraa (1.2.2) to root...
Downloading http://iotdk.intel.com/repos/3.5/intelgalactic/opkg/i586/mraa_1.2.2_i586.ipk.
Collected errors:
 * check_data_file_clashes: Package mraa wants to install file /usr/lib/python2.7/site-packages/mraa.py
        But that file is already provided by package  * libmraa1
 * check_data_file_clashes: Package mraa wants to install file /usr/lib/python2.7/site-packages/_mraa.so
        But that file is already provided by package  * libmraa1
 * check_data_file_clashes: Package mraa wants to install file /usr/lib/libmraa.so.1
        But that file is already provided by package  * libmraa1
 * opkg_install_cmd: Cannot install package mraa.

arfoll, 2016-08-09:
@windmillknight damnit :/ easiest thing to do is to use opkg to remove libmraa1 and upm and reinstall them from that repo. opkg remove should work. The latest UPM is provided in that repo too.

@pylbert? this is an issue...

alext-mkrs, 2016-08-09:
That's a bit strange, if that's release 3.5 (judging by the kernel version), there shouldn't be a libmraa1 as the image has I think 1.0.0, which is called simply mraa.

@windmillknight, what image version do you have and how did you install it?

windmillknight, 2016-08-10:
Oh, I download it from https://github.com/jiangzidong/image4hope. I think it is a version for iot-sol.
And I have another question . I want to use the jsupm_apa102 moudle in upm.How can I  install it?
just use "npm install jsupm_apa102 "?

windmillknight, 2016-08-10:
I have already install the upm moudle . 
But there is a problem .when I run my program for some times .It would get stuck and  I have to reboot my edison .this is my code :

function go (i,end){
if(i<end){
if(i !== 0){
var j = (i-1);
ledStrip.setLed(j,1,0,0,0);

}
ledStrip.setLed(i,1,255,255,255);
ledStrip.pushState();
setTimeout(go,500,++i,end);
}
else{
ledStrip.setAllLeds(1,0,0,0);
return ;
} 
}
var lib = require('jsupm_apa102');
var ledStrip = new lib.APA102(30,0,true);

go(0,30);

Is it something wrong about SPI ?

arfoll, 2016-08-10:
So tbh I would recommend installing the official latest image from software.intel.com/iot since I know that one is good.

For the UPM module issue you'll have to ask on intel-iot-devkit/upm. It could be the SPI bus, edison's SPI is not it's strongest point but I can't say for sure with the apa102 module.

windmillknight, 2016-08-10:
ok .I know. Thank you very much~

alext-mkrs, 2016-08-17:
Looks like this one is fully fixed/answered, so I'm closing it. In a nutshell - it's recommended to use Intel's official images instead of those made by third-parties (or at least don't be surprised at problems - in this case we had new image with very old mraa) and UPM questions/issues are better addressed in the UPM repo bugtracker.

closed 2016-08-17 by alext-mkrs