eclipse-mraa/mraa #559: All npm installs of 1.2.x failing

jakehewitt, 2016-08-10:
npm install was failing for 1.2.0 and 1.2.1 due to the previously mentioned ```fatal error: json-c/json.h: No such file or directory```. npm install is successful for 1.2.2 but when the package is actually imported throws an error ```<ommited>/node_modules/mraa/build/Release/mraa.node: undefined symbol: mraa_init_json_platform```

System is running node 6.3.1, npm 3.10.3 on a debian image

arfoll, 2016-08-10:
Urgh sorry about this. For some reaosn my system lets me get away with this, this is the problem we have with so many different combinations. This will be fixed in an hour or so.

arfoll, 2016-08-10:
1.2.3 is out and hopefully fixes this for you (works here). Travis builds have been updated to make sure this doesn't happen, they know check the npmpkg builds and loads.

jakehewitt, 2016-08-10:
Wow that was quick, yeah 1.2.3 fixed it

closed 2016-08-10 by jakehewitt