eclipse-mraa/mraa #562: Is Uart's time out function not complete yet?

Royalor, 2016-08-17:
I try to use setTimeout of javascript function on nodejs, 
setTimeout(0, 0, 5);
and 
readStr(200);
it not work for time out function.

I found the function of mraa_uart_set_timeout(...int write, int interchar) which the parameter, write &  interchar is not use in code, so... am i right?

alext-mkrs, 2016-08-17:
As far as I can see in the code, only the read timeout is implemented (in commit 63604ef5cc8a2f4da44f75d39a6b2363dd10eff8), before that this functionality was not implemented at all, so write and inter-char are still not implemented.

Royalor, 2016-08-22:
Thanks for your response, so.. if we want to use write and inter-char timeout function, we must create by self?

alext-mkrs, 2016-08-22:
Yes  pretty much. Or add thin functionality to mraa and a pull request ;-)

arfoll, 2016-08-22:
If someone would like to do this, they're more than welcome to, otherwise I'll take a stab at it later this week. Apoligies this isn't documented better, to be perfectly honest I'd completely forgotten about this feature and the lack of full implementation.