eclipse-mraa/mraa #566: Building LibMraa using Node.js installed using Ubuntu apt-get results in error

vraoresearch, 2016-08-20:
**Build System:** Ubuntu 16.04.1
**Install Command:** sudo apt-get install nodejs-dev
Installing nodejs-dev installs Node.js to the following directories.
**Executable Name:** ```nodejs```
**Executable Path:** ```/usr/bin```
**Include Path:** ```/usr/include/nodejs/src```  where node.h can be found.
**Dependency Path:** ```/usr/include/nodejs/deps/uv/include/``` where uv.h and v8.h can be found.

when building LibMraa, upon running cmake you get an error 
Could NOT find Nodejs (missing: NODEJS_INCLUDE_DIRS)

ERROR - node.h not found
ERROR - uv.h not found

If you build Node.js from source, you can build LibMraa without this error.
The difference is that
**Executable Name:** ```node```
**Executable Path:** ```/usr/local/bin/```
**Include Path:** ```/usr/local/include/node/``` This directory has the includes all in one folder i.e. node.h, uv.h and v8.h

By tweaking  the cmake module FindNodejs.cmake a little, it is possible build LibMraa with the apt-get installed nodejs package. In the code block shown below just remove the leading ```include/```
as shown by the  **include/**  below:

```
# Now look for node. Flag an error if not found
find_path (NODE_ROOT_DIR "**include/**node/node.h" "**include/**src/node.h"
  PATHS /usr/include/nodejs /usr/local/include/nodejs /usr/local/include)
if (NODE_ROOT_DIR)
  add_include_dir(${NODE_ROOT_DIR}/**include/**src)
  add_include_dir(${NODE_ROOT_DIR}/**include/**node)
  add_include_dir(${NODE_ROOT_DIR}/**include/**deps/v8/include)
  add_include_dir(${NODE_ROOT_DIR}/**include/**deps/uv/include)
else()
  unset(NODEJS_INCLUDE_DIRS)
  message(ERROR " - node.h not found")
endif()
```

With the above changes, it is possible to install all LibMraa dependencies using only apt-get.

closed 2016-09-12 by arfoll

skiselev, 2017-03-03:
Same problem on Ubuntu 16.10

arfoll, 2017-03-03:
It's because ubuntu doesn't ship the node headers (even when you install nodejs-dev). You can either get them by using nvm (see .travis.yml) or compiling from source or by using a distro that packages things properly ;-)

commit 139d42d0 (2016-09-12): FindNodejs.cmake: added search paths for Ubuntu 16.04

Closes #566.