eclipse-mraa/mraa #569: Using GP19 as GPIO on Edison

mogar, 2016-08-24:
I'm using mraa version 1.2.3 in Python on an Intel Edison.

I'm trying to use GP19 as a GPIO pin. It can also serve as the I2C-1-SCL.

Whenever I set GP19 as an output, the pin toggles as if it's clocking an I2C bus. It does this even if I haven't written any data to the bus yet.

For reference, I'm able to use GP20 (the I2C-1-SDA pin) as a GPIO without any problems.

Thanks!

alext-mkrs, 2016-08-24:
Any code samples? Edison and which expansion board? This sounds like the pin is not really switched to the GPIO mode...

arfoll, 2016-08-24:
Thanks for reporting, can you give us the value of /sys/class/gpio/19/curent_pinmux as you run your code (expecting it to work)? If 0 is not the reply can you try:
```
echo 0 > /sys/class/gpio/19/pinmux
```

mogar, 2016-08-25:
I'm using a custom block, but mraa pin 19 just has a level shifter attached to it. There's nothing that should be interfering with the pin.

Here's what I'm doing in python:

>>> import mraa as m
>>> DC = m.Gpio(19)
>>> DC.dir(m.DIR_OUT)
0
>>> # pin 19 starts clocking with 84us period
>>> DC.write(1)
0
>>> # logic level high
>>> DC.write(0)
0
>>> # pin clocking with 94us period

I actually don't see pin 19 in GPIO:

root@EdisonExplorer:~# cat /sys/class/gpio/19/current_pinmux
cat: /sys/class/gpio/19/current_pinmux: No such file or directory
root@EdisonExplorer:~# cat /sys/class/gpio/19/pinmux
cat: /sys/class/gpio/19/pinmux: No such file or directory
root@EdisonExplorer:~# cd /sys/class/gpio/
root@EdisonExplorer:/sys/class/gpio# ls
export   gpio125  gpio127  gpio129  gpio131  gpio133  gpiochip0
gpio124  gpio126  gpio128  gpio130  gpio132  gpio134  unexport

alext-mkrs, 2016-08-25:
Could you please show the journalctl output for the time you run your program? Looks like the GPIO is not exported at all and that's one of the first steps during init. If it didn't work, there's most probably something in the log.

arfoll, 2016-08-25:
So I've tried this out with mraa 1.0.0 from the latest image for edison and GP19 behaves as expected with my LA on 500MS/s, I can't see anything funny going on. Will see if latest mraa makes a difference but I can't imagine that really. Have you tried using GP19 on the reference edison miniboard to see if it's something weird on your board?

mogar, 2016-08-25:
Looks like my previous testing wasn't done correctly. I didn't actually have that python script running in the background. I wrote up a python script that just toggles pin19 once per second, and I have that running in the background.

    root@EdisonExplorer:/sys/class/gpio# ls
    export   gpio125  gpio127  gpio129  gpio131  gpio133  gpio19     unexport
    gpio124  gpio126  gpio128  gpio130  gpio132  gpio134  gpiochip0

So I have gpio19 now (not exported with the same name as yours though).

    root@EdisonExplorer:/sys/class/gpio# cd gpio19
    root@EdisonExplorer:/sys/class/gpio/gpio19# ls
    active_low  device  direction  edge  power  subsystem  uevent  value

There's no pinmux file inside the gpio19 directory.

I also tried re-flashing my Edison to see if that made a difference.  I've now tested it with mraa 1.0.0, and I get the same behavior.

If I look at journalctl, I see a lot of lines like this:

    Aug 25 15:31:33 EdisonExplorer libmraa[399]: [[1;39mlibmraa initialised for platform 'Intel Edison' of type 2[[0m
    Aug 25 15:31:39 EdisonExplorer libmraa[403]: [[1;39mlibmraa version v1.0.0 initialised by user 'root' with EUID 0[[0m
    Aug 25 15:31:39 EdisonExplorer libmraa[403]: [[1;31mgpio214: init: Failed to write to 'export': No such device[[0m
    Aug 25 15:31:39 EdisonExplorer libmraa[403]: [[1;39medison: Arduino board not detected, assuming Intel Edison Miniboard[[0m

So it looks like it can't find the device?

arfoll, 2016-08-25:
Those errors/warnings are completely normal for a miniboard. We essentially probe for a arduino breakout hence the slightly odd message. I ran this for a while and then analysed the trace, it looks perfect to me. Are you able to try on the mini breakout or bypassing your level shifter? I honestly can't think of anything else. Even if i2c was active, pulses on SCL should only happen when data is transfered so it seems pretty odd to get leakage.

```
>>> import mraa as m
>>> x = m.Gpio(19)
>>> x.dir(m.DIR_OUT)
0
>>> import time
>>> while(True):
...   x.write(0)
...   time.sleep(1)
...   x.write(1)
...   time.sleep(1)
```

mogar, 2016-08-25:
The only other breakout board I have for the Edison that makes GPIO19 even slightly accessible is the Arduino board. I tried using that, but I still don't get the right behavior (it's different, but that may just be because mraa thinks that board should only use i2c).

What yocto image are you using? It seems like you expect the pin exports to happen differently than is happening on my board. I've been using the latest yocto image, which I think is from June 6.

arfoll, 2016-08-26:
I'm using IOTDK 3.5 on a new board so it should be the same as you. I'm a little lost since I have no idea how to replicate this.

alext-mkrs, 2016-08-26:
@mogar, if you're up to that, I'd suggest you to try initializing and toggling that pin fully from sysfs. Code in gpio.c and mraa.c should show you how.

If you don't think you know how to do that, I should be able to prepare an instruction tomorrow or day after tomorrow.

mogar, 2016-08-26:
I'll take a look at gpio.c and mraa.c tomorrow and see if I can test things out with sysfs directly. I'll let you know what I see.

Another hint about what might be going on is that if I set the pin as input first, mraa seems to give me an error when I try to set it as output:

    >>> import mraa as m
    >>> DC = m.Gpio(19)
    >>> DC.dir(m.DIR_IN)
    0
    >>> DC.dir(m.DIR_OUT)
    7
    >>> DC.write(0)
    7
    >>> DC.write(1)
    7

mogar, 2016-08-27:
I finally got a hold of a mini breakout for the Edison. Using the same Edison and the same python script, I see exactly what I expect on GPIO19. Looks like the problem is my hardware >.<

Thanks for all your help!

closed 2016-08-27 by mogar

arfoll, 2016-08-27:
Well that's very reassuring, thanks for getting back to us and good luck!