eclipse-mraa/mraa #570: Contradictory GPIO behavior on Joule: Error: Illegal arguments...

rwaldron, 2016-08-24:
The following illustrates an issue that I've consistently encountered (as you can see): 

```
root@intel-corei7-64:~/js# node
> var m = require("mraa");
undefined
> m.pinModeTest(100, m.PIN_GPIO)
true
> new m.Gpio(100);
Error: Illegal arguments for construction of _exports_Gpio
    at Error (native)
    at repl:1:1
    at REPLServer.defaultEval (repl.js:262:27)
    at bound (domain.js:287:14)
    at REPLServer.runBound [as eval] (domain.js:300:12)
    at REPLServer.<anonymous> (repl.js:427:12)
    at emitOne (events.js:82:20)
    at REPLServer.emit (events.js:169:7)
    at REPLServer.Interface._onLine (readline.js:211:10)
    at REPLServer.Interface._line (readline.js:550:8)
> m.pinModeTest(100, m.PSegmentation fault (core dumped)
root@intel-corei7-64:~/js# node
> var m = require("mraa");
undefined
> m.pinModeTest(101, m.PIN_GPIO)
true
> new m.Gpio(101);
Error: Illegal arguments for construction of _exports_Gpio
    at Error (native)
    at repl:1:1
    at REPLServer.defaultEval (repl.js:262:27)
    at bound (domain.js:287:14)
    at REPLServer.runBound [as eval] (domain.js:300:12)
    at REPLServer.<anonymous> (repl.js:427:12)
    at emitOne (events.js:82:20)
    at REPLServer.emit (events.js:169:7)
    at REPLServer.Interface._onLine (readline.js:211:10)
    at REPLServer.Interface._line (readline.js:550:8)
> m.pinModeTest(101, m.PIN_GPIO)Segmentation fault (core dumped)
root@intel-corei7-64:~/js# node
> var m = require("mraa");
undefined
> m.pinModeTest(102, m.PIN_GPIO)
true
> new m.Gpio(102);
Error: Illegal arguments for construction of _exports_Gpio
    at Error (native)
    at repl:1:1
    at REPLServer.defaultEval (repl.js:262:27)
    at bound (domain.js:287:14)
    at REPLServer.runBound [as eval] (domain.js:300:12)
    at REPLServer.<anonymous> (repl.js:427:12)
    at emitOne (events.js:82:20)
    at REPLServer.emit (events.js:169:7)
    at REPLServer.Interface._onLine (readline.js:211:10)
    at REPLServer.Interface._line (readline.js:550:8)
```

rwaldron, 2016-08-24:
Running this: 

```
#Blink LED on Python with MRAA
#First, run python by entering 
#python
#in the command line
#You should see a prompt similar to
#Python 2.7.11
#>>> 
#start by importing mraa
import mraa
#instantiates a GPIO object using pin 101, one of the on-board LEDs
led = mraa.Gpio(101)       
#sets the GPIO to an output
led.dir(mraa.DIR_OUT) 
#The on-board LED should turn off (see picture below if you're not sure which one it is), it is active low, you should also see a '0' returned on the console, which means MRAA success
led.write(1)                    
#turn the LED back on
led.write(0) 
```

(from https://software.intel.com/en-us/node/672541)

Doesn't actually do anything (ie. the LED doesn't change, it's on the whole time), but the return codes are 0: 

```
root@intel-corei7-64:~/js# python
Python 2.7.11 (default, Jul 22 2016, 20:41:20)
[GCC 5.4.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> led = mraa.Gpio(101)
>>> led.dir(mraa.DIR_OUT)
0
>>> led.write(1)
0
>>> led.write(0)
0
>>>
```

arfoll, 2016-08-24:
If you have the latest firmware those indeed don't work. It's in the
release notes somewhere but I still don't have a good situation. It should
be 100-102. 103 should work.

Sorry, will try fix asap.

rwaldron, 2016-08-24:
> If you have the latest firmware those indeed don't work. It's in the release notes somewhere 

I can't find it here https://github.com/intel-iot-devkit/mraa/blob/master/docs/changelog.md or https://github.com/intel-iot-devkit/mraa/releases/tag/v1.2.3 —is that what you meant, or did I misunderstand?

Just now confirmed that 103 worked, but then the board crashed. I can't say that the two are related.

arfoll, 2016-08-25:
No it should be somewhere on the Intel site as an errata. The problem is it worked in earlier versions of the OS and broke right before release :/

alext-mkrs, 2016-08-25:
I guess @arfoll meant Joule image release notes,not mraa's

rwaldron, 2016-08-25:
Thanks for the follow ups, much appreciated

deadprogram, 2016-08-26:
The Joule OS image that shipped on the board at IDF does not appear to have any inherent firmware limitations on using those LEDs.

I am able to run Gobot code on it that flashes all of them:https://vine.co/v/5OEtPilJDnh

Note that Gobot does not use MRAA, it uses a direct sysfs interface.

Anyhow, I hope that is helpful in some way.

arfoll, 2016-08-26:
interesting that is helpful - can you give me the output of /etc/os-release on the board you got? Does mraa also not work for you but gobot does?

deadprogram, 2016-08-26:
```
root@intel-corei7-64:~# cat /etc/os-release
ID="ostro-xt"
NAME="Ostro OS XT"
VERSION="1.0+snapshot-20160805"
VERSION_ID="1.0+snapshot-20160805"
PRETTY_NAME="Ostro OS XT 1.0+snapshot-20160805"
BUILD_ID="2016-08-05_14-11-41-build-166"
```

I am just getting around to testing mraa on the board. I'll get back to you on that.

arfoll, 2016-08-26:
@deadprogram can you check `/sys/devices/virtual/dmi/id/bios_version` and `/sys/devices/virtual/dmi/id/bios_date`? I've had a few people check this and all are reporting different things :/

arfoll, 2016-08-30:
Right this should all be fixed in master, tested on A16 bios (latest public) with Ostro-XT 172 (latest). mraa 1.3.0 will be tagged as soon as travis pases the tests.

arfoll, 2016-08-30:
pushed 1.3.0. Please re-open if bug is still there!

closed 2016-08-30 by arfoll