eclipse-mraa/mraa #572: issue with Gpio on Edison with ubilinux

mogar, 2016-08-27:
Hi,

We installed mraa onto ubilinux using the SparkFun instructions here:
https://learn.sparkfun.com/tutorials/installing-libmraa-on-ubilinux-for-edison

Now when we try to use any gpio, we get an error:

edison@edison-ti ~/src/915MHzEdisonExplorer_SW/CC_Debug $ python
Python 2.7.9 (default, Mar  1 2015, 18:22:53)
[GCC 4.9.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa as m
>>> rst = m.Gpio(36)
Traceback (most recent call last):
 File "<stdin>", line 1, in <module>
 File "/usr/local/lib/python2.7/dist-packages/mraa.py", line 592, in __init__
   this = _mraa.new_Gpio(*args)
ValueError: Invalid GPIO pin specified

We've tried this with a few different pins (which all work under yocto) and still get this error.

Any ideas?

arfoll, 2016-08-27:
> on-ubilinux-for-edison
> File "", line 1, in

Check /var/log/messages and past the mraa related output here are you
running under root?

mogar, 2016-08-27:
I'm not running under root (and wouldn't like to). 

/var/log/messages has the following, repeated many times
~~~~
Aug 27 15:08:32 edison-ti libmraa[2377]: libmraa initialised for platform 'Intel Edison' of type 2
Aug 27 15:08:33 edison-ti libmraa[2386]: libmraa version v1.2.3-7-gbdbbfd0 initialised by user 'edison' with EUID 1002
Aug 27 15:08:33 edison-ti libmraa[2386]: edison: Arduino board not detected, assuming Intel Edison Miniboard
~~~~

mogar, 2016-08-27:
Oh, if I run it with sudo it works. Is there something I can do to get this working with a non-root user?

arfoll, 2016-08-27:
So if you don't want to run as root you'll have to fix the permissions in
sysfs to let you do that. Imraa was made for that purpose but it doesn't
quite function like that yet so easiest way is probably udev.

alext-mkrs, 2016-08-28:
@mogar, check out https://github.com/intel-iot-devkit/mraa/issues/380, there was some work done by @tingleby on enabling exactly this.

RudyAramayo, 2017-05-31:
I ran the install mraa script and it worked after this... had same error for intel joule

sudo was also necessary to run this code

nikitshah, 2017-08-02:
@RudyAramayo  which mraa script?