eclipse-mraa/mraa #573: Error when initializing A4/A5 at Intel Galileo Gen 2

SaschaWeisenberger, 2016-08-31:
When some voltage is connected to A4 or A5 and e.g. I call the function mraa_aio_init(4), the whole I²C bus is clamped to that voltage and you cannot use any other gpio from this moment on, because the gpio expanders connected to that bus aren't reachable anymore.

The reason is, that gpio60s (b->pins[18].aio.mux[1].pin = 60; int "intel_galileo_rev_g.c") direction is set to MRAA_GPIO_OUT. This causes the mux to have an output value of 0, which connects the A4/A5 connectors to the I²C bus. 
I solved this by using MRAA_GPIO_OUT_HIGH for pin 60 on MRAA_INTEL_GALILEO_GEN2.

[fix_mux_init.patch.txt](https://github.com/intel-iot-devkit/mraa/files/446714/fix_mux_init.patch.txt)

Can you please add this fix?

arfoll, 2016-08-31:
Interesting, I can't merge this as is, changing the way the muxing works for everything is not a good idea, that could break many other platforms.

How about just using the pincmd value to set direction to MRAA_GPIO_OUT_HIGH instead? I'm happy to merge that and should work the same way
```
diff --git a/src/x86/intel_galileo_rev_g.c b/src/x86/intel_galileo_rev_g.c
index d0fe8d0..3f4e258 100644
--- a/src/x86/intel_galileo_rev_g.c
+++ b/src/x86/intel_galileo_rev_g.c
@@ -808,9 +808,9 @@ mraa_intel_galileo_gen2()
     b->pins[18].aio.mux[0].pincmd = PINCMD_SET_DIRECTION;
     b->pins[18].aio.mux[0].pin = 57;
     b->pins[18].aio.mux[0].value = MRAA_GPIO_IN;
-    b->pins[18].aio.mux[1].pincmd = PINCMD_SET_OUT_VALUE;
+    b->pins[18].aio.mux[1].pincmd = PINCMD_SET_DIRECTION;
     b->pins[18].aio.mux[1].pin = 60;
-    b->pins[18].aio.mux[1].value = 1;
+    b->pins[18].aio.mux[1].value = MRAA_GPIO_OUT_HIGH;
     b->pins[18].aio.mux[2].pincmd = PINCMD_SET_VALUE;
     b->pins[18].aio.mux[2].pin = 78;
     b->pins[18].aio.mux[2].value = 0;
```

SaschaWeisenberger, 2016-08-31:
Hi, unfortunatly this doesn't help. The problem are these lines

```
ret = mraa_gpio_dir(mux_i, MRAA_GPIO_OUT);
if(ret == MRAA_SUCCESS)
    ret = mraa_gpio_write(mux_i, meta.mux[mi].value);
```
The value for pin 60 is correct! But you set the direction first to MRAA_GPIO_OUT, this means in fact you set the pin to zero hard coded, because thats the default value for output pins...
 After that, the value is written, but there is a very short time between these commands. 
Instead I suggest to do this in one command instead of two commands:

`mraa_gpio_dir(mux_i, meta.mux[mi].value ? MRAA_GPIO_OUT_HIGH : MRAA_GPIO_OUT_LOW);`

This doesn't change the behavior for any other board, except there won't be this very short time of mismatch between default value and assigned value (if assigned value is 1).

arfoll, 2016-08-31:
Have you tried it the patch I sent? The code stops setting the value for pin60 but sets the direction instead to MRAA_GPIO_OUT_HIGH wich is exactly what the patch I sent you does ;-) Instead of going into that code PINCMD_SET_OUT_VALUE it'll go in PINCMD_SET_DIRECTION, which doesn't touch the gpio value.

Sadly this would change the behaviour, some boards don't support setting direction OUT_HIGH/OUT_LOW :(

SaschaWeisenberger, 2016-08-31:
oh, oops, didn't see both changes... You don't set ..._OUT_VALUE but ..._DIRECTION
okay, great! I'm totally happy with that ;)
Thanks!

arfoll, 2016-08-31:
@SaschaWeisenberger no problem at all - can you confirm it fixes the issue for you and works and I'll merge it? I'll keep it open till merged.

SaschaWeisenberger, 2016-08-31:
Oh change it for A4 AND A5! And also for the gpio function! Respectivly everywhere where the pin 60 is set to 1. Thanks!

arfoll, 2016-08-31:
Like this?

```
diff --git a/src/x86/intel_galileo_rev_g.c b/src/x86/intel_galileo_rev_g.c
index d0fe8d0..35533c0 100644
--- a/src/x86/intel_galileo_rev_g.c
+++ b/src/x86/intel_galileo_rev_g.c
@@ -808,9 +808,9 @@ mraa_intel_galileo_gen2()
     b->pins[18].aio.mux[0].pincmd = PINCMD_SET_DIRECTION;
     b->pins[18].aio.mux[0].pin = 57;
     b->pins[18].aio.mux[0].value = MRAA_GPIO_IN;
-    b->pins[18].aio.mux[1].pincmd = PINCMD_SET_OUT_VALUE;
+    b->pins[18].aio.mux[1].pincmd = PINCMD_SET_DIRECTION;
     b->pins[18].aio.mux[1].pin = 60;
-    b->pins[18].aio.mux[1].value = 1;
+    b->pins[18].aio.mux[1].value = MRAA_GPIO_OUT_HIGH;
     b->pins[18].aio.mux[2].pincmd = PINCMD_SET_VALUE;
     b->pins[18].aio.mux[2].pin = 78;
     b->pins[18].aio.mux[2].value = 0;
@@ -819,9 +819,9 @@ mraa_intel_galileo_gen2()
     b->pins[18].gpio.mux[0].pincmd = PINCMD_SET_DIRECTION;
     b->pins[18].gpio.mux[0].pin = 57;
     b->pins[18].gpio.mux[0].value = MRAA_GPIO_IN;
-    b->pins[18].gpio.mux[1].pincmd = PINCMD_SET_OUT_VALUE;
+    b->pins[18].gpio.mux[1].pincmd = PINCMD_SET_DIRECTION;
     b->pins[18].gpio.mux[1].pin = 60;
-    b->pins[18].gpio.mux[1].value = 1;
+    b->pins[18].gpio.mux[1].value = MRAA_GPIO_OUT_HIGH;
     b->pins[18].gpio.mux[2].pincmd = PINCMD_SET_VALUE;
     b->pins[18].gpio.mux[2].pin = 78;
     b->pins[18].gpio.mux[2].value = 1;
@@ -845,9 +845,9 @@ mraa_intel_galileo_gen2()
     b->pins[19].aio.mux[0].pincmd = PINCMD_SET_DIRECTION;
     b->pins[19].aio.mux[0].pin = 59;
     b->pins[19].aio.mux[0].value = MRAA_GPIO_IN;
-    b->pins[19].aio.mux[1].pincmd = PINCMD_SET_OUT_VALUE;
+    b->pins[19].aio.mux[1].pincmd = PINCMD_SET_DIRECTION;
     b->pins[19].aio.mux[1].pin = 60;
-    b->pins[19].aio.mux[1].value = 1;
+    b->pins[19].aio.mux[1].value = MRAA_GPIO_OUT_HIGH;
     b->pins[19].aio.mux[2].pincmd = PINCMD_SET_VALUE;
     b->pins[19].aio.mux[2].pin = 79;
     b->pins[19].aio.mux[2].value = 0;
@@ -856,9 +856,9 @@ mraa_intel_galileo_gen2()
     b->pins[19].gpio.mux[0].pincmd = PINCMD_SET_DIRECTION;
     b->pins[19].gpio.mux[0].pin = 59;
     b->pins[19].gpio.mux[0].value = MRAA_GPIO_IN;
-    b->pins[19].gpio.mux[1].pincmd = PINCMD_SET_OUT_VALUE;
+    b->pins[19].gpio.mux[1].pincmd = PINCMD_SET_DIRECTION;
     b->pins[19].gpio.mux[1].pin = 60;
-    b->pins[19].gpio.mux[1].value = 1;
+    b->pins[19].gpio.mux[1].value = MRAA_GPIO_OUT_HIGH;
     b->pins[19].gpio.mux[2].pincmd = PINCMD_SET_VALUE;
     b->pins[19].gpio.mux[2].pin = 79;
     b->pins[19].gpio.mux[2].value = 1;
```

If you can test this I can merge it today, otherwise will need a few days to get a galileo gen2 up and running and wired up.

SaschaWeisenberger, 2016-08-31:
Tested with my board now. No error. 

Btw. for the completeness, this was the error message: 
```
i2c_designware i2c_designware.0: controller timed out
pca953x i2c-INT3491:02: failed writing register
...
```

commit 8ec4fcb3 (2016-09-02): intel_galileo_rev_g.c: use pincmd to set OUT_HIGH instead of value

The i/o expander looses connection as soon as direction is set so use direction
OUT_HIGH instead of setting value after direction. Closes #573
closed 2016-09-02 by arfoll