eclipse-mraa/mraa #581: Ubuntu Installation from PPA Encounters Errors

vraoresearch, 2016-09-21:
**Test System:**
- Distributor ID:	Ubuntu
- Description:	Ubuntu 16.04.1 LTS
- Release:	16.04
- Codename:	xenial

PPA was added according to documentation and keyring was successfully updated.

```
sudo apt-get install libmraa1 libmraa-dev mraa-gpio mraa-tools python2-mraa python3-mraa
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python2-mraa
```

After removing the two Python packages from the command:

```
sudo apt-get install libmraa1 libmraa-dev mraa-gpio mraa-tools
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libpango1.0-0 libpangox-1.0-0
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  libmraa-dev libmraa1 mraa-gpio mraa-tools
0 upgraded, 4 newly installed, 0 to remove and 4 not upgraded.
Need to get 0 B/90.6 kB of archives.
After this operation, 404 kB of additional disk space will be used.
Selecting previously unselected package libmraa1.
(Reading database ... 172733 files and directories currently installed.)
Preparing to unpack .../libmraa1_1.3.0-2_amd64.deb ...
Unpacking libmraa1 (1.3.0-2) ...
Selecting previously unselected package libmraa-dev.
Preparing to unpack .../libmraa-dev_1.3.0-2_amd64.deb ...
Unpacking libmraa-dev (1.3.0-2) ...
Selecting previously unselected package mraa-gpio.
Preparing to unpack .../mraa-gpio_1.2.3-1_amd64.deb ...
Unpacking mraa-gpio (1.2.3-1) ...
Preparing to unpack .../mraa-tools_1.3.0-2_amd64.deb ...
Unpacking mraa-tools (1.3.0-2) ...
dpkg: error processing archive /var/cache/apt/archives/mraa-tools_1.3.0-2_amd64.deb (--unpack):
 trying to overwrite '/usr/bin/mraa-gpio', which is also in package mraa-gpio 1.2.3-1
Processing triggers for libc-bin (2.23-0ubuntu3) ...
Errors were encountered while processing:
 /var/cache/apt/archives/mraa-tools_1.3.0-2_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

tingleby, 2016-09-21:
Yes, those instructions are not the best. 

The first is beacuse the packages plain don't exist.....
Second, mraa-gpio was replaced by the mraa-tools package as it contains mraa-i2c too!

Let me commit and fix this doc issue.

closed 2016-09-21 by tingleby

vraoresearch, 2016-09-21:
Thanks Tom
Works perfectly now...

commit 92fdb8aa (2016-09-21): README: Update PPA instructions

Fixes #581