eclipse-mraa/mraa #582: Nodejs install with npm fails with node-gyp errors

vraoresearch, 2016-09-21:
**Test System:**

- Distributor ID:	Ubuntu
- Description:	Ubuntu 16.04.1 LTS
- Release:	16.04
- Codename:	xenial

Installed Node.js version v4.2.6 using aptitude
Installed nmp version v3.5.2 using aptitude

Attempted to install mraa using npm
npm install mraa

```
npm install mraa

> mraa@1.3.0 install /home/vraoresearch/node_modules/mraa
> node-gyp rebuild

/bin/sh: 1: node: not found
gyp: Call to 'node -e 'console.log(process.arch)'' returned exit status 127 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:354:16)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.4.0-21-generic
gyp ERR! command "/usr/bin/nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/vraoresearch/node_modules/mraa
gyp ERR! node -v v4.2.6
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/vraoresearch/package.json'
npm WARN vraoresearch No description
npm WARN vraoresearch No repository field.
npm WARN vraoresearch No README data
npm WARN vraoresearch No license field.
npm ERR! Linux 4.4.0-21-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "mraa"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE

npm ERR! mraa@1.3.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@1.3.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs mraa
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/vraoresearch/npm-debug.log
[npm-debug.log.txt](https://github.com/intel-iot-devkit/mraa/files/486053/npm-debug.log.txt)

```

vraoresearch, 2016-09-21:
My bad...
Did not notice the requirement for Compiler and Nodejs-dev.

vraoresearch, 2016-09-21:
Nope still a problem:
Nodejs-dev was already installed
Installed cmake v3.5.1 and clang version 3.8.0-2ubuntu4

```
sudo npm install mraa

> mraa@1.3.0 install /home/vraoresearch/node_modules/mraa
> node-gyp rebuild

/bin/sh: 1: node: not found
gyp: Call to 'node -e 'console.log(process.arch)'' returned exit status 127 while in binding.gyp. while trying to load binding.gyp
gyp ERR! configure error 
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:354:16)
gyp ERR! stack     at emitTwo (events.js:87:13)
gyp ERR! stack     at ChildProcess.emit (events.js:172:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:200:12)
gyp ERR! System Linux 4.4.0-21-generic
gyp ERR! command "/usr/bin/nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/vraoresearch/node_modules/mraa
gyp ERR! node -v v4.2.6
gyp ERR! node-gyp -v v3.0.3
gyp ERR! not ok 
npm WARN enoent ENOENT: no such file or directory, open '/home/vraoresearch/package.json'
npm WARN vraoresearch No description
npm WARN vraoresearch No repository field.
npm WARN vraoresearch No README data
npm WARN vraoresearch No license field.
npm ERR! Linux 4.4.0-21-generic
npm ERR! argv "/usr/bin/nodejs" "/usr/bin/npm" "install" "mraa"
npm ERR! node v4.2.6
npm ERR! npm  v3.5.2
npm ERR! code ELIFECYCLE

npm ERR! mraa@1.3.0 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@1.3.0 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs mraa
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/vraoresearch/npm-debug.log
```

arfoll, 2016-09-22:
Uhm you'll need to simlink /usr/bin/nodejs to /usr/bin/node, seems that the ubuntu guys decided to change the binary name....

vraoresearch, 2016-09-22:
Hmm! I do have a symlink. Probably created it after trying the install...

So trying again and looks like it succeeded with a lot of warnings ....

[mraa_installation.txt](https://github.com/intel-iot-devkit/mraa/files/487733/mraa_installation.txt)
 
Thanks

The preferred way to do this on Ubuntu is as follows:
sudo update-alternatives --install /usr/bin/node node /usr/bin/nodejs 10

I will update the documentation...

closed 2016-09-22 by vraoresearch

arfoll, 2016-09-22:
Great thanks. 1.3.1 should remove alot of warnings