eclipse-mraa/mraa #585: Setting PWMs on Edison to zero duty is weird and unreliable

RiccardoMaffei, 2016-09-28:
I'm currently using mraa v1.3.0 with Java bindings on an Intel Edison and sometimes the PWM pins are not set to zero.

In this version the #515 fix should be already implemented.
This bug may be similar and related to the #91 but I think it's not the same.

PWM pins sometimes (**not always**) fail to go to 0% duty but the failure rate seem related to the difference between the zero and the previous duty value. 
I've made some tests including the three examples provided below.

## Example 1:
The following code has a failure rate of ~100%. A led connected to the pin is almost never turned off.
```java
for(int i = 0; i < 100; i++) {	//a hundred times
	Thread.sleep(500);			//wait half a second
	pwmPin.write(1);			//write 100%
	Thread.sleep(500);			//wait half a second
	pwmPin.write(0.0f);			//write 0%
}
```

## Example 2:
The following code has a failure rate of ~60%:
- ~40% of the time the led works as expected: 100%->50%->0% (each loop)
- ~60% of the time the led is not turned off and goes to 100%: 100%->50%->100% (each loop)
```java
for(int i = 0; i < 100; i++) {	//a hundred times
	Thread.sleep(500);			//wait half a second
	pwmPin.write(1);			//write 100%
	Thread.sleep(500);			//wait half a second
	pwmPin.write(0.5f);			//write 50%
	Thread.sleep(500);			//wait half a second
	pwmPin.write(0.0f);			//write 0%
}
```

## Example 3:
In the following code the two pins fail but not necessarily at the same time:
- sometimes the first pin fails
- sometimes the second pin fails
- sometimes they both fail
- sometimes they both work as expected
```java
for(int i = 0; i < 100; i++) {	//a hundred times
	Thread.sleep(500);			//wait half a second
	pwmPinOne.write(1);			//write 100%
	pwmPinTwo.write(1);			//write 100%
	Thread.sleep(500);			//wait half a second
	pwmPinOne.write(0.5f);		//write 50%
	pwmPinTwo.write(0.5f);		//write 50%
	Thread.sleep(500);			//wait half a second
	pwmPinOne.write(0.0f);		//write 0%
	pwmPinTwo.write(0.0f);		//write 0%
}
```

I've tried using `pwmPin.enable(false)` but the bug is still there and the failure rate is the same.

alext-mkrs, 2016-09-29:
That's exactly the thing I've described in the last paragraph of #515's description and later on I've found out this is most likely a hardware issue, just like it looked to me, so I haven't pursued it any further.

Propanu, 2016-09-29:
This was a big deal with our UPM Buzzer driver too so I did the following:
```
mraa_pwm_period_us(m_pwm_context, 1);
mraa_pwm_write(m_pwm_context, 0);
```
It works well for the Buzzer since it's not audible, but I have doubts it's a flat 0 and you might still see it with an LED. However, since the issue seems to appear only when writing a pulse_width of 0, you can also try the other extreme and set the maximum possible value for the period and 1 us for the width. That might be the closest you could get to a flat 0 reliably.

RiccardoMaffei, 2016-09-29:
The LED is used just for testing purpose.
I've just tried setting a low duty value (e.g. 1% which is ok for my purpose) instead of the 0% but in this way even the 100% duty fails. 

## Example 4:
In the following code I'm experiencing a weird behavior because the LED blinks from ~5% brightness to 0% almost every time.
```java
for(int i = 0; i < 100; i++){	//a hundred times
	pwmPin.write(1.0f);			//write 100%
	Thread.sleep(500);			//wait half a second
	pwmPin.write(0.01f);		//write 1%
	Thread.sleep(500);			//wait half a second
}
```
Unfortunately, at the moment I don't have an oscilloscope to check what the emitted signal actually is.

alext-mkrs, 2016-10-02:
As this is a known HW problem, I don't think we can help with this one as I don't know of any way of alleviating this in SW. So I'd vote for closing this one as wontfix/hardware issue. @arfoll or anyone reading, any other ideas?

arfoll, 2016-10-03:
Unfortunately this is an issue at a much lower level than mraa so there's not much we can do about it. I'm going to guess it's a kernel/bios/hw issue but I can't tell with any certaininty. I'm sorry I can't be more useful.

I'm happy to leave this open for people to discuss workarounds and not be surprised.

alext-mkrs, 2016-10-03:
Ok, let's leave it opened then. I'll also add my findings from the troubleshooting sessions as there are some useful facts about when it's happening more frequently and when less.

belugon, 2016-10-04:
I had similar issue. After some testing, it appears the edison pwm output would stuck at the state when pwm is disabled. E.g. if pwm is outputting high when disable bit is written to the register, the pwm pin will stay at high. This explains why failure rate is higher when switching from 100%->0% than from 50%->0%

One workaround is to set 0% duty cycle before disable the pwm output from MRAA. I created a pull request with the workaround. 

Another place to put this fix is in the pwm driver: https://github.com/01org/edison-linux/blob/edison-3.10.98/drivers/pwm/pwm-intel-mid.c
I have a working fix for that too. But I thought updating mraa would be easier than updating the whole kernel image since the pwm driver is built in.

alext-mkrs, 2016-10-04:
I think I have tested this approach ad it does not alleviate the issue completely, just decreases the fail rate. As for the pull request, I see you are changing the enable/disable logic by removing array updates, I'd be careful with that. Will test your proposal and comment in the PR.

You need to sign-off your commit, see contributing.md

belugon, 2016-10-04:
You are right, the change doesn't work for large pwm period case (>1ms). I added 2 period delay to allow HW to pick up duty cycle change.
PR is updated.

alext-mkrs, 2016-10-04:
So here come my troubleshooting results for the same issue (from the past). For whatever reason GitHub doesn't like my ZIP with all three files, so here they are as gists:

1. [first test script](https://gist.github.com/alext-mkrs/324832d4f2e5384ac619c68a906c3687), uses "enable->set duty->disable" pattern
1. [second test script](https://gist.github.com/alext-mkrs/70c4e8b23b01f9693b055b104a36675f), uses "set duty->enable->disable" pattern
1. [table with some stats](https://gist.github.com/alext-mkrs/6079765ac85adc5ab2a7b1eaa7435360) gathered by running those scripts on Edison's pin 3 and 6

Overall, pin number and patter make difference, but I was not able to reach fully proper functioning.

belugon, 2016-10-05:
Thanks for the cool test script. I run the "enable->set duty->disable" case on pin3 with no failure for all duty cycles after my workaround. I'll try few more case tomorrow.

alext-mkrs, 2016-12-18:
Ok, looks like we are going to have a viable workaround for this one, so removing `wontfix`. Within PR #586 there's some work being done to make this work despite the HW peculiarity.

alext-mkrs, 2017-01-15:
This is now fixed with commits af051d820aa0fb2649f6b093a5286617e1ec035a and 2558866d4add55d590f17b77cbe1b8cd4767d807

closed 2017-01-15 by alext-mkrs