eclipse-mraa/mraa #587: NPM install error on OS X

w4ilun, 2016-10-08:
Error when installing MRAA with NPM on OS X. Running node v6.3.1:

```
> node-gyp rebuild

  CC(target) Release/obj.target/mraa/src/x86/intel_gt_tuchuck.o
In file included from ../src/x86/intel_gt_tuchuck.c:32:
In file included from ../include/x86/intel_gt_tuchuck.h:31:
In file included from ../include/mraa_internal.h:35:
In file included from ../include/mraa_internal_types.h:31:
In file included from ../api/mraa/iio.h:28:
../api/mraa/iio_kernel_headers.h:113:10: fatal error: 'linux/ioctl.h' file not found
#include <linux/ioctl.h>
         ^
1 error generated.
```

w4ilun, 2016-10-08:
Closing. Duplicate of https://github.com/intel-iot-devkit/mraa/issues/546

closed 2016-10-08 by w4ilun

arfoll, 2016-10-11:
If you want this to work you'll have to send me a mac ;-) I'm quite impartial to the new mac air....