eclipse-mraa/mraa #588: How to initialize RTS/CTS pinson Edison?

htot, 2016-10-08:
On Edison (miniboard / ardiono board) when initializing the uart the RX/TX pins are mapped and configured. However, when setting HW flow control (RTS/CTS) the pins are not configured.

Can that be done using mraa? Or should I duplicate code from src/x86/intel_edison_fab_c.c mraa_intel_edison_uart_init_pre and modify that init RTS/CTS?

alext-mkrs, 2016-10-09:
You can use [mraa_uart_set_flowcontrol()](http://iotdk.intel.com/docs/master/mraa/uart_8h.html#abf0e5f97dfcbf227e03a4e1b2225dfd2) function for that, or a respective wrapper if you're using mraa from something else than C.

htot, 2016-10-09:
Yeah, that will initialize the UART to use flow control. It will not enable the pin AFAIK, those are multiplexed with GPIO on Edison, the same way as RX/TX. The mode is changed in src/x86/intel_edison_fab_c.c mraa_intel_edison_uart_init_pre with the following lines:

    ret = mraa_intel_edison_pinmode_change(130, 1); // IO0 RX
    ret = mraa_intel_edison_pinmode_change(131, 1); // IO1 TX

Unfortunately mraa_intel_edison_pinmode_change is a static function.

arfoll, 2016-10-10:
So this commit will hopefully give you a temp fix https://github.com/intel-iot-devkit/mraa/commit/563bf708c636d6068e989d78e1d67e88fd8ae947. However this is quite a complex issue so will need substantial testing before I add something like this in.

htot, 2016-10-11:
Yes, something like this did occur to me. But I think my application will break later when something definite goes in. Thus my suggestion to copy/paste something similar into my application.

The real question is how initializing rtx/cts should fit into mraa best. Currently afaik on edison the pins are never initialized. Your patch will always initialize them (so disable gpio on these pins).

Currently mraa_uart_init does not take an option that would allow conditionally initializing rts/cts. And adding that would break all existing application code. It could be added to mraa_uart_set_flowcontrol(), but that would be a late point to configure pins and probably that function should not depend on hardware/platform.

So what could be a definite solution then? Create a function mraa_pin_option() that should be called before mraa_uart_init? Such a function could be made to have zero effect on other platforms and when not called mraa_uart_init could default to the current not changing the rts/cts pin mode.

alext-mkrs, 2016-10-11:
In general, there's an infrastructure of so called advanced functions, essentially platform-specific ones, which override (`_replace`) or run before (`_pre`) or after (`_post`) the generic one. So `mraa_uart_set_flowcontrol()` could theoretically be overridden for Edison in a way that setting the rts/cts to true would trigger pin reconfiguration. Not sure though if that's possible at arbitrary point in time, haven't looked at what specifically that reconfiguration would entail.