eclipse-mraa/mraa #589: Spurious error when calling mraa_add_subplatform()?

deadprogram, 2016-10-10:
When attempting to call `mraa_add_subplatform(MRAA_GENERIC_FIRMATA, "/dev/ttyACM0")` I am getting back a result code `4` however the call appears to actually succeed, and subsequent MRAA calls work as expected.

I am calling from a NUC (platform 3) to an attached Arduino 101.

arfoll, 2016-10-11:
If you get MRAA_ERROR_INVALID_PARAMETER you need to check syslog, in any case it pretty much guarrantees that the call failed, likely imraa already added the platform so the `mraa_add_subplatform` call was superfluous?

deadprogram, 2016-10-11:
I will check syslog.

Is there a way I can check to see if imraa has already added the subplatform? I tried calling `mraa_get_platform_type()` but that returns the NUC that MRAA is running on.

Perhaps `mraa_add_subplatform()` could return a specific result in the case that imraa has already added the subplatform?

arfoll, 2016-10-11:
Uhm there is `mraa_get_platform_combined_type` which if you give the offset should return the firmata firmware version. We had a combined name thing but it turned out to be a bad idea since it's impossible to parse. There is `mraa_get_platform_combined_type` but it's not exposed via C++ because it's a bit confusing.

We had something like you describe where `mraa_init` would return MRAA_ERROR_PLATFORM_ALREADY_INITIALISED instead of `MRAA_SUCCESS` when already initialised but turned out that was really hard to parse/understand and very error prone. But probably `mraa_add_subplatform` should return `MRAA_SUCCESS` if a platform is already there and used just like `mraa_init` silently succeeds. Does that make sense?

deadprogram, 2016-10-11:
"But probably mraa_add_subplatform should return MRAA_SUCCESS if a platform is already there and used just like mraa_init silently succeeds."

Yes, please. That is the behavior I expected, and the `MRAA_ERROR_INVALID_PARAMETER` confused me into thinking that my param was bad somehow, and not realizing for a while that the subplatform was already available.

arfoll, 2016-10-11:
Done with 980dfce80ed5ee5f2244aaf434d0ef29a5e6babc - if there is a GENERIC_FIRMATA subplatform already present it will return `MRAA_SUCCESS`, if another subplatform is present you will get `MRAA_ERROR_FEATURE_NOT_SUPPORTED` and in all other cases `MRAA_ERROR_INVALID_PARAMETER`

closed 2016-10-11 by arfoll