eclipse-mraa/mraa #591: Is UART working for the up-board

KreativeCode, 2016-10-16:
I'm running the up-board kernel on ubuntu and i can't seem to get my python program to work. when i run it, it gives me the following error.

File "motorController.py", line 8, in <module>
    transmitter = mraa.Uart(1) # i enter here doesn't make a difference.
  File "/usr/local/lib/python2.7/dist-packages/mraa.py", line 1181, in __init__
    this = _mraa.new_Uart(*args)
ValueError: Error initialising UART

Should i be doing something extra? The following is my code up until it breaks.

import mraa
import time
import binascii

#Setup
transmitter = mraa.Uart(1)

Let me know if you need any other info.
Thanks in advance!

arfoll, 2016-10-16:
Are the devices /dev/ttyS1 and /dev/ttyS2 on your device? What's the output
in syslog and do you have the correct user permissions to write on serial
devices?

KreativeCode, 2016-10-16:
I'm not sure how to check if those devices are on my device. The following is my syslog output. Sorry about not putting it up i'm new to this. It seems as though i don't have permission to write on serial devices. How do i go about doing that?

Oct 16 12:59:50 maverick-UP-CHT01 anacron[2850]: Updated timestamp for job `cron.monthly' to 2016-10-16
Oct 16 13:17:01 maverick-UP-CHT01 CRON[3478]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Oct 16 13:33:41 maverick-UP-CHT01 libmraa[3515]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 13:33:41 maverick-UP-CHT01 libmraa[3515]: libmraa initialised for platform 'UP' of type 12
Oct 16 13:33:41 maverick-UP-CHT01 libmraa[3515]: uart: open(/dev/ttyS1) failed: Permission denied
Oct 16 13:33:57 maverick-UP-CHT01 libmraa[3518]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 13:33:57 maverick-UP-CHT01 libmraa[3518]: libmraa initialised for platform 'UP' of type 12
Oct 16 13:33:57 maverick-UP-CHT01 libmraa[3518]: uart: open(/dev/ttyS1) failed: Permission denied
Oct 16 13:34:19 maverick-UP-CHT01 libmraa[3521]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 13:34:19 maverick-UP-CHT01 libmraa[3521]: libmraa initialised for platform 'UP' of type 12
Oct 16 13:34:19 maverick-UP-CHT01 libmraa[3521]: uart: open(/dev/ttyS1) failed: Permission denied
Oct 16 13:42:00 maverick-UP-CHT01 dbus[443]: [system] Activating service name='org.freedesktop.hostname1' (using servicehelper)
Oct 16 13:42:00 maverick-UP-CHT01 kernel: [ 3902.708713] systemd-hostnamed[3565]: Warning: nss-myhostname is not installed. Changing the local hostname might make it unresolveable. Please install nss-myhostname!
Oct 16 13:42:00 maverick-UP-CHT01 dbus[443]: [system] Successfully activated service 'org.freedesktop.hostname1'
Oct 16 13:42:01 maverick-UP-CHT01 dbus[443]: [system] Activating service name='org.freedesktop.PackageKit' (using servicehelper)
Oct 16 13:42:02 maverick-UP-CHT01 AptDaemon: INFO: Initializing daemon
Oct 16 13:42:02 maverick-UP-CHT01 AptDaemon.PackageKit: INFO: Initializing PackageKit compat layer
Oct 16 13:42:02 maverick-UP-CHT01 dbus[443]: [system] Successfully activated service 'org.freedesktop.PackageKit'
Oct 16 13:42:02 maverick-UP-CHT01 AptDaemon.PackageKit: INFO: CreateTransaction() was called
Oct 16 13:42:02 maverick-UP-CHT01 AptDaemon.PackageKit: INFO: Initializing PackageKit transaction
Oct 16 13:42:02 maverick-UP-CHT01 AptDaemon.PackageKit: INFO: GetUpdates() was called
Oct 16 13:42:02 maverick-UP-CHT01 AptDaemon.Worker: INFO: Simulating trans: /org/debian/apt/transaction/b518651c701a4779b1e5b45221014fc6
Oct 16 13:42:02 maverick-UP-CHT01 AptDaemon.Worker: INFO: Processing transaction /org/debian/apt/transaction/b518651c701a4779b1e5b45221014fc6
Oct 16 13:42:09 maverick-UP-CHT01 AptDaemon.Worker: INFO: Finished transaction /org/debian/apt/transaction/b518651c701a4779b1e5b45221014fc6
Oct 16 13:53:02 maverick-UP-CHT01 AptDaemon: INFO: Quitting due to inactivity
Oct 16 13:53:02 maverick-UP-CHT01 AptDaemon: INFO: Quitting was requested
Oct 16 14:04:50 maverick-UP-CHT01 libmraa[3746]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 14:04:50 maverick-UP-CHT01 libmraa[3746]: libmraa initialised for platform 'UP' of type 12
Oct 16 14:04:50 maverick-UP-CHT01 libmraa[3746]: uart: open(/dev/ttyS2) failed: Permission denied
Oct 16 14:12:33 maverick-UP-CHT01 libmraa[3753]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 14:12:33 maverick-UP-CHT01 libmraa[3753]: libmraa initialised for platform 'UP' of type 12
Oct 16 14:12:33 maverick-UP-CHT01 libmraa[3753]: uart: open(/dev/ttyS2) failed: Permission denied
Oct 16 14:17:01 maverick-UP-CHT01 CRON[3760]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Oct 16 14:25:01 maverick-UP-CHT01 libmraa[3767]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'root' with EUID 0
Oct 16 14:25:01 maverick-UP-CHT01 libmraa[3767]: libmraa initialised for platform 'UP' of type 12
Oct 16 14:25:01 maverick-UP-CHT01 libmraa[3767]: uart: tcgetattr(/dev/ttyS2) failed: Input/output error
Oct 16 15:17:01 maverick-UP-CHT01 CRON[3777]: (root) CMD (   cd / && run-parts --report /etc/cron.hourly)
Oct 16 15:27:21 maverick-UP-CHT01 libmraa[3782]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 15:27:21 maverick-UP-CHT01 libmraa[3782]: libmraa initialised for platform 'UP' of type 12
Oct 16 15:27:21 maverick-UP-CHT01 libmraa[3782]: uart8: init: platform has only 2 uarts
Oct 16 15:27:29 maverick-UP-CHT01 libmraa[3784]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 15:27:29 maverick-UP-CHT01 libmraa[3784]: libmraa initialised for platform 'UP' of type 12
Oct 16 15:27:29 maverick-UP-CHT01 libmraa[3784]: uart10: init: platform has only 2 uarts
Oct 16 15:27:43 maverick-UP-CHT01 libmraa[3786]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 15:27:43 maverick-UP-CHT01 libmraa[3786]: libmraa initialised for platform 'UP' of type 12
Oct 16 15:27:57 maverick-UP-CHT01 libmraa[3788]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 15:27:57 maverick-UP-CHT01 libmraa[3788]: libmraa initialised for platform 'UP' of type 12

arfoll, 2016-10-16:
So your serial devices are in /dev/ttyS*. Have a look what group they
belong to usually it's modem, or uccp. Add your user to that group and you
should be good. Note that for other stuff like gpio you'll have to set more
perms. Test with root to see if works then work to fix your permissions.

KreativeCode, 2016-10-16:
i ran ls -l and got the following (most has been cut out):

crw-rw----  1 root dialout   4,  64 Oct 16 16:47 ttyS0
crw-rw----  1 root dialout   4,  65 Oct 16 16:47 ttyS1
crw-rw----  1 root dialout   4,  74 Oct 16 16:47 ttyS10
crw-rw----  1 root dialout   4,  75 Oct 16 16:47 ttyS11
crw-rw----  1 root dialout   4,  76 Oct 16 16:47 ttyS12
crw-rw----  1 root dialout   4,  77 Oct 16 16:47 ttyS13
crw-rw----  1 root dialout   4,  78 Oct 16 16:47 ttyS14
crw-rw----  1 root dialout   4,  79 Oct 16 16:47 ttyS15
crw-rw----  1 root dialout   4,  80 Oct 16 16:47 ttyS16
crw-rw----  1 root dialout   4,  81 Oct 16 16:47 ttyS17
crw-rw----  1 root dialout   4,  82 Oct 16 16:47 ttyS18
crw-rw----  1 root dialout   4,  83 Oct 16 16:47 ttyS19
crw-rw----  1 root dialout   4,  66 Oct 16 16:47 ttyS2

the two serial devices belong to the group dealt so i ran: sudo usermod -a -G dealt maverick

I logged out and logged back in. then i tried to run the program again with 8 as the Uart constructor field and got the same error. However, when i did syslog again i got the following:

Oct 16 17:03:38 maverick-UP-CHT01 libmraa[1862]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 17:03:38 maverick-UP-CHT01 libmraa[1862]: libmraa initialised for platform 'UP' of type 12
Oct 16 17:03:38 maverick-UP-CHT01 libmraa[1862]: uart8: init: platform has only 2 uarts

Then i ran again with 0 as the Uart constructor field (i saw this in an example python program in this repository) and got the following.

Oct 16 17:06:06 maverick-UP-CHT01 libmraa[1875]: libmraa version v1.3.0-7-g8ec4fcb initialised by user 'maverick' with EUID 1000
Oct 16 17:06:06 maverick-UP-CHT01 libmraa[1875]: libmraa initialised for platform 'UP' of type 12
Oct 16 17:06:06 maverick-UP-CHT01 libmraa[1875]: uart: tcgetattr(/dev/ttyS1) failed: Input/output error

any ideas?

KreativeCode, 2016-10-17:
I ran 'setserial -g /dev/ttyS[012]' and got this:

maverick@maverick-UP-CHT01:~$ setserial -g /dev/ttyS[012]
/dev/ttyS0, UART: 16550A, Port: 0x03f8, IRQ: 4
/dev/ttyS1, UART: unknown, Port: 0x02f8, IRQ: 3
/dev/ttyS2, UART: unknown, Port: 0x03e8, IRQ: 4

I'm thinking this is why it's not working. Can anyone tell me how i can set up ttyS[12]?

I tried running sudo setserial /dev/ttyS1 which i'm guessing configured it. It doesn't give me the error anymore however it's not transmitting anything and it hangs. Also when i run 'echo "Hello World" > /dev/ttyS1' it hangs.

Any idea if i'm heading in the right direction? Where do i go from here?

KreativeCode, 2016-10-17:
Thanks guys i've figured it out. The upboard kernel i was using recently got updated to allow for the 40-pin headers to function properly. once i did the update everything worked properly.

Thanks again.

alext-mkrs, 2016-10-18:
No worries, I'll close it. Thanks for reporting back the fix you've found!

closed 2016-10-18 by alext-mkrs