eclipse-mraa/mraa #598: NPM v1.5.1 build failing on Ubilinux

WoodyWoodsta, 2016-11-08:
Building `mraa v1.5.1` fails on Ubilinux on the Intel Edison. I've tried Node `v7.0.0`, `v6.0.0` and the preinstalled `v0.10.29`.

[npm-debug.log](http://pastebin.com/7FKCMK1n)

Installing `mraa@1.4.0` works as expected.

arfoll, 2016-11-09:
Indeed, seems it's broken on node v7. Will diagnose & fix asap.

arfoll, 2016-11-09:
So here is the issue, it's a v8 API brak - https://github.com/swig/swig/issues/825
For now you'll have to stick with node < 7

alext-mkrs, 2017-05-13:
https://github.com/swig/swig/issues/804 is now the leading one for this issue and looks like folks are getting close to the resolution there, so keep your fingers crossed :)

alext-mkrs, 2018-06-21:
The newest SWIG thread on this is at https://github.com/swig/swig/pull/968

alext-mkrs, 2026-02-15:
Closing this one as very obviously no longer relevant. We do have some Node-related build fixing to do, but that's a different story these days.