eclipse-mraa/mraa #599: SIGSEGV with ISR in thread

ihabZhaika, 2016-11-08:
Hi,
I am using python 2.7 , mraa  v1.4.0 , on Intel GT Tuchuck
I created a class to lunch interrupts for pins , but when the interrupt happen and i call the method to execute , it throws SIGSEGV , the interrupt method is 'recoredRpm' 
Tried to run it without thread and no problem

main :
```
left = WheelEncoder("left wheel",ConstData.LEFT,ConstData.ECNODER_LEFT_PIN)
left.start()
while True:
    print "Wait"
    time.sleep(1)`

```
The class
```
import threading
import mraa

class WheelEncoder(threading.Thread):
```
    def __init__(self, name, direction, pin):
        threading.Thread.__init__(self)
        self.name = name
        self.direction = direction
        self.rpm = 0
        self.pin = pin
        self.encoder = None

```
    def recoredRpm(self):
        print "inside "

    def startEncoder(self):
        try:
            self.encoder = mraa.Gpio(self.pin)
            print("Starting ISR for pin " + repr(self.pin))
            self.encoder.dir(mraa.DIR_IN)
            self.encoder.isr(mraa.EDGE_BOTH, self.recoredRpm,None)
        except ValueError as e:
            print(e)

    def run(self):
        self.startEncoder()

alext-mkrs, 2016-11-08:
1. There's no start() method in your class, but you call it in main. Do you mean run?
2. Please post the whole stack trace/error message you get, that would help troubleshooting.

ihabZhaika, 2016-11-08:
@alext-mkrs 
1- the start method are inherited from (threading.Thread) , which call run
2- Process finished with exit code 139 (interrupted by signal 11: SIGSEGV)

alext-mkrs, 2016-11-08:
Okay, so something tries to access the memory outside of what OS thinks the address space of the process is. The ISR code runs in a separate thread in mraa, so probably something is getting out of scope somewhere.

What exactly do you mean by "if i take out the method under the class , and call it with it self , it works"? Is the code you posted a working version or not? I'm not with any of my boards right now, so I'd need you to post the necessary pieces from your runs to help me troubleshoot this, for now.

In addition to the above, please post the following two things (as attachments or something like pastebin, please don't paste that into the comment itself, it becomes unreadable):

1. output of the `journalctl -f` command run when the segmentation fault happens
1. output of the `strace -ffF -tt -s 512 python <your script name>` command

ihabZhaika, 2016-11-09:
HI, 
for what i meant , check the two files , one do work and one dont,
the difference that i removed the meothod from the class scope and put it in the script scope then access it
download the code :
https://drive.google.com/open?id=0B_AFMME1kY1oNjBwYlhtRzNFUWM
1-
```
Nov 09 08:17:34 ubuntu3 kernel: python2.7[3995]: segfault at 8 ip 00000000004e115f sp 00007faec8da4e10 error 6 in python2.7[400000+2e9000]
Nov 09 08:18:07 ubuntu3 libmraa[4017]: libmraa version v1.4.0 initialised by user 'root' with EUID 0
Nov 09 08:18:07 ubuntu3 libmraa[4017]: libmraa initialised for platform 'Intel GT Tuchuck' of type 13
Nov 09 08:18:08 ubuntu3 libmraa[4017]: gpio: PyEval_CallObject failed
Nov 09 08:18:08 ubuntu3 libmraa[4017]: gpio: the error was <type 'exceptions.AttributeError'>:'function' object has no attribute '_Verbose__verbose':<traceback object at 0x7f369f1c1a70>
Nov 09 08:18:08 ubuntu3 kernel: python2.7[4021]: segfault at 8 ip 00000000004e115f sp 00007f369f1bce10 error 6 in python2.7[400000+2e9000]

```

2- 
```
[pid  4053] 08:19:46.948852 select(0, NULL, NULL, NULL, {1, 0} <unfinished ...>
[pid  4057] 08:19:47.334987 <... poll resumed> ) = 1 ([{fd=4, revents=POLLPRI|POLLERR}])
[pid  4057] 08:19:47.335204 read(4, "\n", 1) = 1
[pid  4057] 08:19:47.335608 sendto(5, "<139>Nov  9 08:19:47 libmraa[4053]: gpio: PyEval_CallObject failed", 66, MSG_NOSIGNAL, NULL, 0) = 66
[pid  4057] 08:19:47.335972 sendto(5, "<139>Nov  9 08:19:47 libmraa[4053]: gpio: the error was <type 'exceptions.AttributeError'>:'function' object has no attribute '_Verbose__verbose':<traceback object at 0x7f6a13e3c9e0>", 182, MSG_NOSIGNAL, NULL, 0) = 182
[pid  4057] 08:19:47.336300 --- SIGSEGV {si_signo=SIGSEGV, si_code=SEGV_MAPERR, si_addr=0x8} ---
[pid  4057] 08:19:47.717668 +++ killed by SIGSEGV (core dumped) +++
[pid  4056] 08:19:47.718734 +++ killed by SIGSEGV (core dumped) +++
08:19:47.718853 +++ killed by SIGSEGV (core dumped) +++

```

alext-mkrs, 2016-11-11:
Thanks, but the strace output is too short, I need the full one (that's wny I mentioned pastebin - it will be long).

alext-mkrs, 2016-11-19:
@ihabzhaika, will it be possible for you to post a full log?

alext-mkrs, 2016-11-27:
So I have spent half ofthe day today looking into this and reproduced this thing to a certain degree on my Edison. I haven't got SIGSEGV, but a SIGABRT + an assert within Python, though.
I've noticed that:

1. When the ISR is declared as a usual method or a `@classmethod` of a class, it fails with an error message in the `journalctl` and the following assert:

```
root@edison1:~/mraa-599# python main.py 
MainThread: <MyTestClass(Thread-1, initial)>.start(): starting thread
Thread-1: <MyTestClass(Thread-1, started -1219224768)>.__bootstrap(): thread started
Hello from the main thread
 Before initMraa()
All set
After initMraa()
Thread-1: <MyTestClass(Thread-1, started -1219224768)>.__bootstrap(): normal return
Before PyEvalCallObject
After PyEvalCallObject
before decref(arglist)
after decref(arglist)
before pygilstate_release
python: Objects/classobject.c:2360: instancemethod_dealloc: Assertion `g->gc.gc_refs != (-2)' failed.
Aborted
root@edison1:~/mraa-599#
```

2. If the ISR is defined as a standalone module-level function or a `@staticmethod` of a class, it works

3. If not `threading.Thread` is used as the ISR wrapper class ancestor, but I don't go through the `run()` method there's only an assert and the ISR is actually getting run before it aborts at the assert:
```
root@edison1:~/mraa-599# python testisr.py 
Hello from the main thread
Before PyEvalCallObject
ISR was executed with value: 5
python: Objects/classobject.c:2360: instancemethod_dealloc: Assertion `g->gc.gc_refs != (-2)' failed.
Aborted
```

The code is:
```python
import mraa
import time
import threading

class DummyClass(threading.Thread):
    def myIsr(self, val):
        print("ISR was executed with value: %d" %val)

c = DummyClass()

g = mraa.Gpio(3)
g.dir(mraa.DIR_IN)
g.isr(mraa.EDGE_BOTH, c.myIsr, 5)

while True:
    print("Hello from the main thread")
    time.sleep(5)
```

So all in all there seems to be two problems here, not one, if we assume that SIGSEGV you @ihabZhaika observes is another form or caused by the same thing as my SIGABRT + an assert:

1. due to all the interesting things `threading` does inside with its internal inheritance, it's not compatible with `PyEval_CallObject` (or our - pretty much textbook - way of using it, anyway)

2. the same `PyEval_CallObject` doesn't seem to like the usual (or class-) method of classes and apparently fails when trying to deallocate them after executing the ISR.

To be honest, given the number of layers and thread interactions we have here, I'd say it may be simpler just to take the above specifics into account and move on. I don't see any specific problems in our use of Python C API, but all the examples and illustrations out there are really simplistic and our case is more complex. I'll maybe look into this further, later on, but right now the actual reason seems to be somewhere deep in Python.