eclipse-mraa/mraa #600: Sending zero not working when using mraa.Pwm

edwardsp, 2016-11-09:
I have a motor connected to pin3.  Here is my code:
```
var m = require("mraa");
var pin = new m.Pwm(3);
pin.enable(true);
pin.period_us(2000);
pin.write(1);
// motor is now on
pin.write(0);
// motor is still on
```
The only way around this to actually stop the motor in my case is to send a low value, e.g. 0.01, and then followed by another call to set the value to 0.0.

alext-mkrs, 2016-11-10:
Is that on Edison? If yes, then it's a hardware issue, setting duty to 0 doesn't fully switch it off. That was discussed in #585 and earlier, there's a workaround for that in the code, which doesn't 100% of the time due to another glitch, which is worked around in pending PR #586.

If that's a different platform, let's see which one.

edwardsp, 2016-11-10:
Yes, this is the edison and I have my own workaround by setting it to a low
number and then zero... but it is not ideal!

alext-mkrs, 2016-11-11:
Yep, that's exactly what's being proposed as a workaround in #586. Let me close this one then as a duplicate of #585 and feel free to watch it and the PR #586 for the solution.

closed 2016-11-11 by alext-mkrs