eclipse-mraa/mraa #605: Build cmake with FT4222

GuzMp, 2016-11-15:
Hi,
I want to use this library with FT4222, so I had to add FTDI4222=ON USBPLAT=ON. 
What is command line please ?
I try  _cmake .. -DFTDI4222=ON -DUSBPLAT=ON_ but it don't found library for FT4222 

-- The C compiler identification is GNU 5.4.0
-- The CXX compiler identification is GNU 5.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Performing Test HAS_C__Wall
-- Performing Test HAS_C__Wall - Success
-- Performing Test HAS_C__Werror_main
-- Performing Test HAS_C__Werror_main - Success
-- Performing Test HAS_C__Wformat
-- Performing Test HAS_C__Wformat - Success
-- Performing Test HAS_C__Wmain
-- Performing Test HAS_C__Wmain - Success
-- Performing Test HAS_C__Wuninitialized
-- Performing Test HAS_C__Wuninitialized - Success
-- Performing Test HAS_C__Winit_self
-- Performing Test HAS_C__Winit_self - Success
-- Performing Test HAS_C__Werror_implicit
-- Performing Test HAS_C__Werror_implicit - Success
-- Performing Test HAS_C__Werror_missing_parameter_type
-- Performing Test HAS_C__Werror_missing_parameter_type - Success
-- Performing Test HAS_CXX__Wall
-- Performing Test HAS_CXX__Wall - Success
-- Performing Test HAS_CXX__Werror_main
-- Performing Test HAS_CXX__Werror_main - Success
-- Performing Test HAS_CXX__Wformat
-- Performing Test HAS_CXX__Wformat - Success
-- Performing Test HAS_CXX__Wmain
-- Performing Test HAS_CXX__Wmain - Success
-- Performing Test HAS_CXX__Wuninitialized
-- Performing Test HAS_CXX__Wuninitialized - Success
-- Performing Test HAS_CXX__Winit_self
-- Performing Test HAS_CXX__Winit_self - Success
-- Performing Test HAS_CXX__Wnon_virtual_dtor
-- Performing Test HAS_CXX__Wnon_virtual_dtor - Success
-- Performing Test HAS_CXX__Woverloaded_virtual
-- Performing Test HAS_CXX__Woverloaded_virtual - Success
-- Performing Test HAS_CXX__Wreorder
-- Performing Test HAS_CXX__Wreorder - Success
CMake Warning at CMakeLists.txt:93 (message):
   - Install git to compile a production libmraa!

-- INFO - libmraa Version v1.5.1
-- INFO - cmake Version 3.5.1
-- INFO - Target arch is x86_64
-- Found PythonInterp: /usr/bin/python2.7 (found suitable version "2.7.12", minimum required is "2.7") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython2.7.so (found suitable exact version "2.7.12") 
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3") 
-- Could NOT find PythonLibs (missing:  PYTHON_LIBRARIES PYTHON_INCLUDE_DIRS) (Required is exact version "3.5.2")
-- INFO - Adding onewire backend support
-- INFO - Adding support for platform ALL
-- INFO - Adding support for all platforms
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
INFO - Adding x86 platforms
-- INFO - Adding USB platforms
CMake Error at src/CMakeLists.txt:146 (message):
  Enabled FTDI4222 support but library not found

INFO - FTDI4222
-- Found SWIG: /usr/bin/swig3.0 (found version "3.0.8") 
PYTHON2 attempting to build!
-- Found Nodejs: /usr/bin/nodejs  
INFO - Node version is 4.2.6
INFO - Node using v8 4.5.103.35
-- INFO - swig Version 3.0.8
-- INFO - CXX compiler Version GNU 5.4.0
-- INFO - Using V8 version > 3 so requiring C++11 compiler
-- INFO - install NODE modules to /usr/include/nodejs/lib/node_modules/mraa/
-- Configuring incomplete, errors occurred!
See also "/home/teosit/mraa-master/build/CMakeFiles/CMakeOutput.log".
See also "/home/teosit/mraa-master/build/CMakeFiles/CMakeError.log".

Thank you.

arfoll, 2016-11-15:
If you read here: https://github.com/intel-iot-devkit/mraa/blob/master/docs/ftdi_ft4222.md did you install the library from ftdi?

GuzMp, 2016-11-16:
Libft4222 was not installed correctly, it work now thank you. But now when I test this program : 

int
main(int argc, char** argv)
{
    	mraa_init();
    //! [Interesting]
    printf("MRAA Version: %s\n\n", mraa_get_version() );
    mraa_spi_context spi;
    spi = mraa_spi_init(0);
	mraa_spi_mode(spi, MRAA_SPI_MODE0);
    if (spi == NULL) {
        printf("Initialization of spi failed, check syslog for details, exit...\n");
        exit(1);
    }

    printf("SPI initialised successfully\n");
	
    mraa_spi_frequency(spi, 10000);
    mraa_spi_lsbmode(spi, 0);
    uint8_t data[1] = { 0x03 };
    uint8_t* recv;
    
    recv = mraa_spi_write_buf(spi, data, 1);
    printf("Writing -%i", data[0]);
    printf("RECIVED-%i-\n", recv[0]);
    usleep(100000);
     
	mraa_spi_stop(spi);
}

And in my syslog : 
Nov 16 09:51:28 teo-System-Product-Name libmraa[7252]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Nov 16 09:51:28 teo-System-Product-Name libmraa[7252]: Platform not supported, not initialising
Nov 16 09:51:28 teo-System-Product-Name libmraa[7252]: libft4222.so not found, skipping
Nov 16 09:51:28 teo-System-Product-Name libmraa[7252]: libmraa initialised for platform 'Unknown platform' of type 98
Nov 16 09:51:28 teo-System-Product-Name libmraa[7252]: spi: no spi buses defined in platform
Nov 16 09:51:28 teo-System-Product-Name libmraa[7252]: spi: mode: context is invalid

arfoll, 2016-11-16:
Ok so that indicates mraa didn't manage to dlopen libft42222.so as done here: ```src/usb/ftdi_ft4222.c:121:    void *func = dlsym(libft4222_lib, symbol);```

Where is libft4222.so on your system and has ldconfig been refreshed?

GuzMp, 2016-11-16:
libft4222.so is on /usr/local/lib. I have juste made : sudo ldconfig and now in my syslog : 

Nov 16 10:15:58 teo-System-Product-Name libmraa[8883]: libmraa version v1.5.1
 initialised by user 'root' with EUID 0
Nov 16 10:15:58 teo-System-Product-Name libmraa[8883]: Platform not supported, not initialising
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: FT_GetDeviceInfoList returned 2 devices
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: mraa_ftdi_ft4222_init completed successfully
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: FT4222_GetVersion 42220200 01030178
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: FT4222_I2CMaster_Read failed for address 0x20. Code 50
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: FT4222_I2CMaster_Write failed address 0x27
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: FT4222_I2CMaster_Read failed for address 0x27. Code 50
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: FT4222_I2CMaster_Read failed for address 0x70. Code 50
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: libmraa initialised for platform 'Unknown platform + FTDI FT4222' of type 98
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: spi: no spi buses defined in platform
Nov 16 10:16:01 teo-System-Product-Name libmraa[8883]: spi: mode: context is invalid

arfoll, 2016-11-16:
ok well that looks much better. Uhm unfortunately SPI functionality isn't implemented in mraa for the FT4222 yet see docs/ftdi_ft4222.md. @whbruce do you have it stashed in a WIP branch or was it never done?

You may have to implement that part if you want to use it with mraa, although most of the ground work is already there so it shouldn't be too hard. We'd obviously welcome the contribution :)

whbruce, 2016-11-16:
I never had SPI connected sensors to test so didn't add support. Maybe @Propanu did some work on this?

Propanu, 2016-11-16:
Unfortunately I never added SPI support for the same reason: we had no need for SPI sensors for the demos that use the FT4222 expander.

arfoll, 2016-11-16:
Closing as probably a feature request at this point ;-) Sorry!

closed 2016-11-16 by arfoll