eclipse-mraa/mraa #606: Reinitializing a GPIO pin will return wrong read values with a Firmata subplatform

Propanu, 2016-11-17:
So I came across this weird one while updating a demo on a Wind River 7 RCPL13 Gateway. It seems like if you close an app and reinitialize a GPIO pin, the second time around it will read bad values, but only until switching states. I haven't been able to trace this yet but one potential workaround (at least for Node, doesn't seem to work in Java) is to set the direction on the pin to out, and then in again, before the read. Steps to reproduce:
``` js
root@WR-IDP-7EE5:~# node
> var m = require('mraa');
undefined
> var pin = new m.Gpio(515);
undefined
> pin.dir(m.DIR_IN);
0
> pin.read()
1
> var pin = new m.Gpio(515);
undefined
> pin.dir(m.DIR_IN);
0
> pin.read()
1
>
(To exit, press ^C again or type .exit)
>
root@WR-IDP-7EE5:~# node
> var m = require('mraa');
undefined
> var pin = new m.Gpio(515);
undefined
> pin.dir(m.DIR_IN);
0
> pin.read()
0
> pin.dir(m.DIR_OUT);
0
> pin.dir(m.DIR_IN);
0
> pin.read()
1
```
Same behavior with MRAA 1.3 and 1.4. The pin has a simple switch attached, pulled low when open and high when closed.

alext-mkrs, 2016-12-07:
Setting this to "triaging" to indicate this is a potential bug no one has positively verified yet, but as long as I don't have access to the HW I unfortunately can't help myself and unless @arfoll or @tingleby have one, we would probably want to set "help wanted" as well.

alext-mkrs, 2018-06-10:
This is quite ancient, @Propanu, do you think we still want to keep it open? Apparently no one bothered by this too much :)

Propanu, 2018-06-22:
I'm fine if we close this, I wasn't able to reproduce it with other Firmata setups since.

closed 2018-06-22 by Propanu