eclipse-mraa/mraa #607: No i2c buses defined in platform FT4222

GuzMp, 2016-11-18:
Hi,
I try to test mraa-i2c.c with FT4222 with nothink in i2c bus but it fail. In my syslog : 

Nov 18 13:20:10 teo-System-Product-Name libmraa[19513]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Nov 18 13:20:10 teo-System-Product-Name libmraa[19513]: Platform not supported, not initialising
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: FT_GetDeviceInfoList returned 2 devices
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: mraa_ftdi_ft4222_init completed successfully
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: FT4222_GetVersion 42220200 01030178
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: FT4222_I2CMaster_Read failed for address 0x20. Code 50
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: FT4222_I2CMaster_Write failed address 0x27
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: FT4222_I2CMaster_Read failed for address 0x27. Code 50
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: FT4222_I2CMaster_Read failed for address 0x70. Code 50
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: libmraa initialised for platform 'Unknown platform + FTDI FT4222' of type 98
Nov 18 13:20:13 teo-System-Product-Name libmraa[19513]: Loglevel 7 is set
Nov 18 13:20:38 teo-System-Product-Name libmraa[19513]: i2c_init: Selected bus 0
Nov 18 13:20:38 teo-System-Product-Name libmraa[19513]: i2c_init: No i2c buses defined in platform

There is a failure during _mraa_i2c_init(bus)_ function. We can see that i2c_bus_count (in i2c.c) stay at 0 but I don't know why..
And where is located mraa_set_log_level() function ? 

Thank you.

arfoll, 2016-11-18:
You need to add the platform offset of 512. So it's mraa_i2c_init(512).

mraa_set_log_level is in mraa/common.h, in the C++ API it's mraa::setLogLevel()

GuzMp, 2016-11-18:
Thank you. 
And now I adding spi in libft4222.c, what I have to add for mraa_spi_init(?) ?

arfoll, 2016-11-18:
So look here https://github.com/intel-iot-devkit/mraa/blob/master/src/usb/ftdi_ft4222.c#L1003. That's the function table added for i2c. You'll need to add new functions and overrides for all the features in src/spi/spi.c so that it goes through the D2xx API

closed 2016-11-18 by arfoll

GuzMp, 2016-11-18:
Yes, I already added this functions in ftdi_ft4222.c through the D2xx API. But I don't understand how you choose the bus

arfoll, 2016-11-18:
ah yes for SPI there's a bit of code missing in spi.c
See: https://github.com/intel-iot-devkit/mraa/blob/master/src/i2c/i2c.c#L144
You'll want to add something like it here: https://github.com/intel-iot-devkit/mraa/blob/master/src/spi/spi.c#L68

Once the 'board' var is flipped to the subplatform then you should end up calling the overrides. You'll probably have to add a mraa_spi_init_internal so you can do a full _replace of the mraa_spi_init function like what is done in i2c https://github.com/intel-iot-devkit/mraa/blob/master/src/i2c/i2c.c#L98

GuzMp, 2016-11-18:
Ok thank you very much.

GuzMp, 2016-11-18:
And what is the platform offset for spi ?

arfoll, 2016-11-18:
platform offset is always the same, 512 for a subplatform. 512+spibus_number where spibus_number is defined in the ft4222 subplatform