eclipse-mraa/mraa #609: Issue installing mraa with Travis CI

resolutedreamer, 2016-11-20:
Hello, thank you for a good repo. I have a project that I am trying to use Travis CI with, and I am getting an error trying to make it install mraa for my repo.

This is the log from the error:
https://gist.github.com/resolutedreamer/997a2e07c20ccb0835e17abf64ecf76a

and this is my .travis.yml file:
https://gist.github.com/resolutedreamer/3d4019eb8c486ea98952a153fb77aea4

tingleby, 2016-11-21:
Hello,
The PPA is only built for xenial (16.04) series of Ubuntu.

arfoll, 2016-11-30:
@tingleby any idea how much effort it would be to build it for 14.04? Might be handy if just for travis. Maybe just C? If it's a 30min type effort might be a good training for me on the PPA build system :)

resolutedreamer, 2016-12-02:
Hi everyone,

I found a workaround. It turns out that in addition to the PPA for 'xenial', there appears to also be a PPA built for 'trusty' at the time of writing. I edited my .travis.yml file with the lines in the instruction in the reference here: https://docs.travis-ci.com/user/trusty-ci-environment/ 

dist: trusty
sudo: required

and it seems to work. Travis CI does not seem to offer xenial as far as I can see right now, so please leave the PPA for 'trusty' up and don't take it down please. : )

alex-ter, 2016-12-07:
@resolutedreamer, could you please share the URL of that PPA you're referring to? I'm just trying to tie @tingleby's note about it being built only for 16.04 and your one about one being available for 14.04 together :)

resolutedreamer, 2016-12-07:
ppa.launchpad.net/mraa/mraa/ubuntu/dists/

alext-mkrs, 2016-12-11:
Okay, that one doesn't seem to be an actively supported. There's only one package, 1.3.0 (https://launchpad.net/~mraa/+archive/ubuntu/mraa/+sourcepub/6959377/+listing-archive-extra), and it was copied from the 16.04 by @tingleby around the end of August, so I guess that was an attempt to setup a 14.04 build, or just an ad-hoc test, which didn't go any further.

alext-mkrs, 2017-05-20:
Let's finalize this one - as 14.04 will still be supported until 2019 - do we want to setup a PPA for it as well, @tingleby, @arfoll?

tingleby, 2017-05-29:
Ill look into how well it packages for 14.04, I think for at least the base it should be ok but things like the imraa service would need an init script instead.

closed 2026-02-15 by alext-mkrs