eclipse-mraa/mraa #610: Intel edison sparkfun 9dof block

spein46, 2016-11-24:
Hi, I installed mraa library in my intel edison with 9dof sparkfun block.
I m trying to compile SparkFun_9DOF_Edison_Block_Example.cpp but when I try to compile SFE_LSM9DS0.cpp, I get this error:
`SFE_LSM9DS0.cpp: In member function ‘void LSM9DS0::gReadBytes(uint8_t, uint8_t*, uint8_t)’:
SFE_LSM9DS0.cpp:514:9: error: ‘class mraa::I2c’ has no member named ‘readBytesReg’
SFE_LSM9DS0.cpp: In member function ‘void LSM9DS0::xmReadBytes(uint8_t, uint8_t*, uint8_t)’:
SFE_LSM9DS0.cpp:524:7: error: ‘class mraa::I2c’ has no member named ‘readBytesReg’`
what do I wrong?
Thanks

alext-mkrs, 2016-11-24:
How specifically did you install the mraa library, via opkg or from sources? It looks like the compilation doesn't properly pick the i2c header, because the function is actually there.

spein46, 2016-11-24:
Hi, thanks for answer, anyway I installed mraa through sources, master vesion, and I used rhia command:
./bootstrap
Make
Make install

Anyway I open i2c.c file and there aren t members .

spein46, 2016-11-24:
I use ubilinux in my intel edison

arfoll, 2016-11-24:
Uhm mraa uses cmake so you can't compile it like that? What's rhia?

So you want the i2c.hpp C++ header. What version of mraa are you on? Calling mraa_get_version() or mraa::getVersion() from C/C++/python etc would be super helpful. Also give us the out put of `ls -l /usr/lib/libmraa*` then we can see what's installed.

spein46, 2016-11-24:
Sorry, I wrong to type, "these commands:". I installed last version, master, from github.

arfoll, 2016-11-24:
Can you give us any of the logs I mentioned? Without those I'm really struggling to understand what's gone wrong

spein46, 2016-11-24:
`root@ubilinux:~# ls -l /usr/lib/libmraa*
lrwxrwxrwx 1 root staff     12 Nov 24 22:11 /usr/lib/libmraa.so -> libmraa.so.1
lrwxrwxrwx 1 root staff     16 Nov 24 22:11 /usr/lib/libmraa.so.1 -> libmraa.so.1.5.1
-rw-r--r-- 1 root staff 211200 Nov 24 22:10 /usr/lib/libmraa.so.1.5.1
`

I use my mak file:
`CC=g++
CFLAGS=-c -Wall
LDFLAGS=
SOURCES=SparkFun_9DOF_Edison_Block_Example.cpp  SFE_LSM9DS0.cpp
OBJECTS=$(SOURCES:.cpp=.o)
EXECUTABLE=EdisonV3

all: $(SOURCES) $(EXECUTABLE)

$(EXECUTABLE): $(OBJECTS)
        $(CC) $(LDFLAGS) $(OBJECTS) -o $@

.cpp.o:
        $(CC) $(CFLAGS) $< -o $@

`

when I run my make file this is output:
`g++  SparkFun_9DOF_Edison_Block_Example.o SFE_LSM9DS0.o -o EdisonV3
SFE_LSM9DS0.o: In function `mraa::I2c::I2c(int, bool)':
SFE_LSM9DS0.cpp:(.text._ZN4mraa3I2cC2Eib[_ZN4mraa3I2cC5Eib]+0x1b): undefined reference to `mraa_i2c_init_raw'
SFE_LSM9DS0.cpp:(.text._ZN4mraa3I2cC2Eib[_ZN4mraa3I2cC5Eib]+0x2f): undefined reference to `mraa_i2c_init'
SFE_LSM9DS0.o: In function `mraa::I2c::address(unsigned char)':
SFE_LSM9DS0.cpp:(.text._ZN4mraa3I2c7addressEh[_ZN4mraa3I2c7addressEh]+0x1d): undefined reference to `mraa_i2c_address'
SFE_LSM9DS0.o: In function `mraa::I2c::readReg(unsigned char)':
SFE_LSM9DS0.cpp:(.text._ZN4mraa3I2c7readRegEh[_ZN4mraa3I2c7readRegEh]+0x1f): undefined reference to `mraa_i2c_read_byte_data'
SFE_LSM9DS0.o: In function `mraa::I2c::readBytesReg(unsigned char, unsigned char*, int)':
SFE_LSM9DS0.cpp:(.text._ZN4mraa3I2c12readBytesRegEhPhi[_ZN4mraa3I2c12readBytesRegEhPhi]+0x2b): undefined reference to `mraa_i2c_read_bytes_data'
SFE_LSM9DS0.o: In function `mraa::I2c::writeReg(unsigned char, unsigned char)':
SFE_LSM9DS0.cpp:(.text._ZN4mraa3I2c8writeRegEhh[_ZN4mraa3I2c8writeRegEhh]+0x2b): undefined reference to `mraa_i2c_write_byte_data'
collect2: error: ld returned 1 exit status
make: *** [EdisonV3] Error 1
`

spein46, 2016-11-25:
I added -lbmraa at LDFLAGS and I solved it. Sorry for my question.
Thanks

arfoll, 2016-11-25:
No problem, glad it's all fixed. FYI I'd really recommend getting your LDFLAGS via pkgconfig rather than manually setting them.

closed 2016-11-25 by arfoll