eclipse-mraa/mraa #615: i2c mraa python

PaulB2Code, 2016-12-06:
Dear Mraa Team, 

We want to implement i2c with python. 

We have developped a programm on raspberry pi using smbus.

`import smbus
import time

bus = smbus.SMBus(1)
address = 0x74

LEDst0 = [ 0x08, 0x41, 0x00, 0x12, 0x4E, 0x04 ,0x52]
DEVICE_REG_LEDOUT0 = 0xA5

time.sleep(2)
print "Envoi de la valeur "
print LEDst0
res = bus.write_i2c_block_data(address, DEVICE_REG_LEDOUT0, LEDst0)
print "Fin envoi de la valeur "

time.sleep(1)
reponse = bus.read_byte(address)
`
that we want to use on edison with mraa.

`import mraa as m
import time

x = m.I2c(1)
x.address(0x74)

print("Envoi de la valeur ")
x.write(bytearray(b'0xA5x08x41x00x12x4Ex04x52'))
print("Lecture Valeur")
time.sleep(1)
d = x.read(2)

print(str(d))
print(int.from_bytes(d, byteorder='little'))
`

But we have the following issue
Envoi de la valeur
Lecture Valeur
Traceback (most recent call last):
  File "testComMCP39F521-mraa.py", line 17, in <module>
    d = x.read(2)
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1164, in read
    return _mraa.I2c_read(self, data)
IOError: [Errno 11] Resource temporarily unavailable

Do you know what is wrong?

Looking forward. 

Paul

arfoll, 2016-12-06:
I suggest you put mraa on the RPI and use m.I2c(1, True) to check that your code works as you expect.

Then let's have a look at edison :) Are you on the miniboard or the arduino board? If it's the arduino then try m.I2c(0), if it's the miniboard are you sure that you've done the voltage level shifting correctly? Can you describe your circuit a little?

PaulB2Code, 2016-12-06:
Hi Arfoll, 

I just try on Pi and I get no errorno11 but i did not have any return value from the sensor. 
But if I used d = x.readByte(), I get an answer.

When i use x.readByte() on edison I get an other error.

`root@edison:~/# i2cdetect -y -r 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- 74 -- -- --
root@edison:~/# python test.py
Envoi de la valeur
Lecture Valeur
Traceback (most recent call last):
  File "test.py", line 17, in <module>
    d = x.readByte()
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1153, in readByte
    return _mraa.I2c_readByte(self)
ValueError: Unknown error in I2c::readByte()
`

I use a miniboard with a sparkfun i2c adaptors  (https://www.sparkfun.com/products/13034) and my sensor. I check the level shiffting and I can see the level at the entrance of the sensor. 

I can do other tests if you need more information or if you have some idear for the issue as I am a new with mraa.

Thank you

Paul

arfoll, 2016-12-07:
ok well what version of mraa are you on? Have you tried updating to v1.5.1?

PaulB2Code, 2016-12-07:
We are already on v.1.5.1 on edison. 

print(m.getVersion())
v1.5.1

alext-mkrs, 2016-12-07:
@PaulB2Code, what sensor is that? How do you power it, using VSYS, 3.3V or VOUT pins on the I2C Block?

PaulB2Code, 2016-12-07:
Hi Alex,

I used 3,3V on thé pin i2c and thé sensor is à microchip sensor (mcp39f521).

As it works on rapsberry Pi. I do not understand why it does not on Edison
if it is an mraa issue or something else.

I can do tests.

Thank you for your help

alext-mkrs, 2016-12-08:
Thanks for the information. The fact it works on RasPi is good, that means we have a known working config we can compare against.

Such generic error in mraa's I2C code typically (but not always) means low-level bus problems, but we'll see.

How do you power your Edison?

PaulB2Code, 2016-12-08:
Ok. 

I power the edison with a USB cable plug in a Iphone Base and the sparkfun base block (SparkFun Block for Intel® Edison - Base). No other power are used. 

I have a uart base block (from sparkfun) that I can power to 3.3V with an external generator if you think this could help.

PaulB2Code, 2016-12-09:
I then used the 3.3 V and Gnd to power the chip and I connected the i2c  to the sensor

alext-mkrs, 2016-12-11:
Okay, so that's a bit different story then. I see you have a Base block, then the I2C one and finally a GPIO one. And you are using GPIO block to power your sensor, right? Not sure why, if you have the same voltage and ground on the I2C Block, but that should not theoretically hurt anyway.

What i don't like is that `i2cdetect` is not showing anything - that rather confirms that there's a connection- (or HW-) level problem and not the library one.

The simplest thing to do would probably be what @arfoll has suggested earlier - use your RPi again, but test your mraa-based program there (mraa supports RPi). Then if it works - that should definitely be something in the connection or the specific sensor breakout's compatibility with Edison's I2C bus and we could check it out further. How exactly your sensor breakout looks like, do you have any links to schematics?

Checking Edison's syslog could be helpful as well, please post `journalctl -f` output you see when running your program.

PaulB2Code, 2016-12-27:
Hi Alex, 
Sorry for my long delay. 

Yes we re using GPIO block to power your sensor. And it is the same as using the i2C voltage and ground of the I2C block but we try to see in ase the sensor consume too much energy derating the i2c signal. 

I test the same program on raspberry Pi and on intel Edison:

Programm:
import mraa as m
import time

x = m.I2c(1,True)
x.address(0x74)

#LEDst0 = [ 0x08, 0x41, 0x00, 0x12, 0x4E, 0x04 ,0x52]
#DEVICE_REG_LEDOUT0 = 0xA5

print("Envoi de la valeur ")
x.write(bytearray(b'0xA5x08x41x00x12x4Ex04x52'))
print("Lecture Valeur")
#time.sleep(2)
d = x.readByte()

print(str(d))
d = x.read(10)

print(str(d))
#print(int.from_bytes(d, byteorder='little'))

Result on Pi:
date
Tue 27 Dec 10:33:14 CET 2016
pi@raspberrypi ~/testMraa $ python test.py
Envoi de la valeur
Lecture Valeur
21

pi@raspberrypi ~/testMraa $ sudo journalctl -f
-- Logs begin at Sun 2016-12-25 14:19:01 CET. --
Dec 27 10:28:36 raspberrypi libmraa[8033]: libmraa version v1.5.1-11-gc1017bb initialised by user 'pi' with EUID 1000
Dec 27 10:28:36 raspberrypi libmraa[8033]: libmraa initialised for platform 'Raspberry Pi 2 Model B Rev 1' of type 5
Dec 27 10:28:45 raspberrypi sudo[8035]: pi : TTY=pts/0 ; PWD=/home/pi/testMraa ; USER=root ; COMMAND=/bin/journalctl -f
Dec 27 10:28:45 raspberrypi sudo[8035]: pam_unix(sudo:session): session opened for user root by pi(uid=0)
Dec 27 10:28:47 raspberrypi sudo[8035]: pam_unix(sudo:session): session closed for user root
Dec 27 10:30:07 raspberrypi dhcpcd[577]: eth0: fe80::1ab4:30ff:fe13:eecb router available
Dec 27 10:33:16 raspberrypi libmraa[8079]: libmraa version v1.5.1-11-gc1017bb initialised by user 'pi' with EUID 1000
Dec 27 10:33:16 raspberrypi libmraa[8079]: libmraa initialised for platform 'Raspberry Pi 2 Model B Rev 1' of type 5
Dec 27 10:33:18 raspberrypi sudo[8080]: pi : TTY=pts/0 ; PWD=/home/pi/testMraa ; USER=root ; COMMAND=/bin/journalctl -f
Dec 27 10:33:18 raspberrypi sudo[8080]: pam_unix(sudo:session): session opened for user root by pi(uid=0)

Result on Edison: 

 date
Tue Dec 27 09:29:50 UTC 2016
root@MyDevice:~/testMraa# journalctl -f
-- Logs begin at Sat 2000-01-01 00:00:20 UTC. --
Dec 27 09:02:09 MyDevice sshd[392]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
Dec 27 09:02:09 MyDevice sshd[392]: lastlog_openseek: Couldn't stat /var/log/lastlog: No such file or directory
Dec 27 09:03:08 MyDevice systemd-timesyncd[163]: interval/delta/delay/jitter/drift 256s/+0.005s/0.018s/0.002s/-5ppm (ignored)
Dec 27 09:07:24 MyDevice systemd-timesyncd[163]: interval/delta/delay/jitter/drift 512s/+0.012s/0.028s/0.005s/-5ppm (ignored)
Dec 27 09:07:37 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 27 09:07:38 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 27 09:07:49 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 27 09:13:43 MyDevice systemd[1]: Starting Cleanup of Temporary Directories...
Dec 27 09:13:43 MyDevice systemd[1]: Started Cleanup of Temporary Directories.
Dec 27 09:15:56 MyDevice systemd-timesyncd[163]: interval/delta/delay/jitter/drift 1024s/+0.013s/0.013s/0.012s/+0ppm
^C
root@MyDevice:~/testMraa# python test.py
Envoi de la valeur
Lecture Valeur
Traceback (most recent call last):
  File "test.py", line 15, in <module>
    d = x.readByte()
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1153, in readByte
    return _mraa.I2c_readByte(self)
ValueError: Unknown error in I2c::readByte()
root@MyDevice:~/testMraa# journalctl -f
-- Logs begin at Sat 2000-01-01 00:00:20 UTC. --
Dec 27 09:13:43 MyDevice systemd[1]: Started Cleanup of Temporary Directories.
Dec 27 09:15:56 MyDevice systemd-timesyncd[163]: interval/delta/delay/jitter/drift 1024s/+0.013s/0.013s/0.012s/+0ppm
Dec 27 09:30:00 MyDevice libmraa[416]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Dec 27 09:30:00 MyDevice libmraa[416]: gpio214: init: Failed to write to 'export': No such device
Dec 27 09:30:00 MyDevice libmraa[416]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Dec 27 09:30:00 MyDevice libmraa[416]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 27 09:30:00 MyDevice libmraa[416]: i2c1: write: Access error: Resource temporarily unavailable
Dec 27 09:30:00 MyDevice libmraa[416]: i2c1: read_byte: Access error: Resource temporarily unavailable
Dec 27 09:30:00 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 27 09:30:00 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration

This issue always appears but I do not know why. 

 kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration

Do you have an explanation ? I can do more tests if need.

Thank you for your help

Paul

PaulB2Code, 2016-12-27:
If I changed my Code a little bit I can have no issue sometimes:

Code: 
import mraa as m
import time

x = m.I2c(1,True)
x.address(0x74)

print("Envoi de la valeur ")
x.write(bytearray(b'0xA5x08x41x00x12x4Ex04x52'))
print("Lecture Valeur")

d = x.read(10)

print(str(d))

Response:
python test.py
Envoi de la valeur
Lecture Valeur

root@MyDevice:~/testMraa# journalctl -f
-- Logs begin at Sat 2000-01-01 00:00:20 UTC. --
Dec 27 11:19:15 MyDevice libmraa[586]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 27 11:19:15 MyDevice libmraa[586]: i2c1: write: Access error: Resource temporarily unavailable
Dec 27 11:19:15 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 27 11:19:15 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 27 11:19:33 MyDevice libmraa[590]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Dec 27 11:19:33 MyDevice libmraa[590]: gpio214: init: Failed to write to 'export': No such device
Dec 27 11:19:33 MyDevice libmraa[590]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Dec 27 11:19:33 MyDevice libmraa[590]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 27 11:19:33 MyDevice libmraa[590]: i2c1: write: Access error: Resource temporarily unavailable
Dec 27 11:19:33 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
^C
root@MyDevice:~/testMraa# python test.py
Envoi de la valeur
Lecture Valeur

root@MyDevice:~/testMraa# python test.py
Envoi de la valeur
Lecture Valeur

root@MyDevice:~/testMraa# python test.py
Envoi de la valeur
Lecture Valeur
Traceback (most recent call last):
  File "test.py", line 18, in <module>
    d = x.read(10)
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1164, in read
    return _mraa.I2c_read(self, data)
IOError: [Errno 11] Resource temporarily unavailable
root@MyDevice:~/testMraa# journalctl -f
-- Logs begin at Sat 2000-01-01 00:00:20 UTC. --
Dec 27 11:19:56 MyDevice libmraa[597]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 27 11:19:56 MyDevice libmraa[597]: i2c1: write: Access error: Resource temporarily unavailable
Dec 27 11:19:56 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 27 11:20:00 MyDevice libmraa[600]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Dec 27 11:20:00 MyDevice libmraa[600]: gpio214: init: Failed to write to 'export': No such device
Dec 27 11:20:00 MyDevice libmraa[600]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Dec 27 11:20:00 MyDevice libmraa[600]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 27 11:20:01 MyDevice libmraa[600]: i2c1: write: Access error: Resource temporarily unavailable
Dec 27 11:20:01 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 27 11:20:01 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration

PaulB2Code, 2016-12-27:
When I tried on raspberry Pi, I get no error but it seems that I can not write properly on the bus because the device do not undestand the request sent. 
If I do not used mraa it work perfectly. 

But if I used mraa then i got  the same answer.

arfoll, 2016-12-27:
Using raw i2c bus 1 on edison is a bad idea. What is this i2c device you are connecting to? Maybe there is something simple we can check with a i2c.readReg() of a whoami type value?

PaulB2Code, 2016-12-27:
The chip is a mcp39F521,  http://ww1.microchip.com/downloads/en/DeviceDoc/20005442A.pdf  . 

Could we use read.Reg if we need to send many frame before?

alext-mkrs, 2016-12-27:
On Edison you'd rather use x = m.I2c(1), i.e. without True, which means raw bus init and is applicable to RPi, but not Edison. Please try it to see if that helps.

That "lost arbitration" thing generally means bus-level connectivity problems, like wrong pullups or something slong these lines. But that could be caused by the raw init, so try changing it first.

PaulB2Code, 2016-12-28:
Hi Alex, 
I remove the True, but i did not change anything. 

It is like I can read the bus, and read information from the chip but i can not send the first command. 

Dec 28 10:34:16 MyDevice libmraa[404]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Dec 28 10:34:16 MyDevice  libmraa[404]: gpio214: init: Failed to write to 'export': No such device
Dec 28 10:34:16 MyDevice libmraa[404]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Dec 28 10:34:16 MyDevice libmraa[404]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 28 10:34:16 MyDevice libmraa[404]: i2c_init: Selected bus 1
Dec 28 10:34:16 MyDevice libmraa[404]: i2c1: write: Access error: Resource temporarily unavailable
Dec 28 10:34:16 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration

Find the read signal but not write signal

[image]

but if I remove the write command, then i get the following error and noting can be read. 

Envoi de la valeur
Lecture Valeur
Traceback (most recent call last):
  File "test.py", line 16, in <module>
    d = x.readByte()
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1153, in readByte
    return _mraa.I2c_readByte(self)
ValueError: Unknown error in I2c::readByte()

Dec 28 10:44:04 MyDevice libmraa[449]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Dec 28 10:44:04 MyDevice libmraa[449]: gpio214: init: Failed to write to 'export': No such device
Dec 28 10:44:04 MyDevice libmraa[449]: edison: Arduino board not detected, assuming Intel Edison Miniboard
Dec 28 10:44:04 MyDevice libmraa[449]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 28 10:44:04 MyDevice libmraa[449]: i2c_init: Selected bus 1
Dec 28 10:44:04 MyDevice  libmraa[449]: i2c1: read_byte: Access error: Resource temporarily unavailable
Dec 28 10:44:04 MyDevice kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration

# **Moreover If i send two (2)  write command then SOMETIMES i can send a command:** 

**If it do not work:**

Dec 28 10:52:19 MyDevice libmraa[484]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 28 10:52:19 MyDevice  libmraa[484]: i2c_init: Selected bus 1
Dec 28 10:52:19 MyDevice  libmraa[484]: i2c1: write: Access error: Resource temporarily unavailable
Dec 28 10:52:19 MyDevice  kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 28 10:52:19 MyDevice  kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration
Dec 28 10:52:19 MyJouleBox libmraa[484]: i2c1: write: Access error: Resource temporarily unavailable

**if it work:** 

Dec 28 10:52:31 MyDevice libmraa[491]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 28 10:52:31 MyDevice libmraa[491]: i2c_init: Selected bus 1
Dec 28 10:52:31 MyDevice libmraa[491]: i2c1: write: Access error: Resource temporarily unavailable
Dec 28 10:52:31 MyDevice  kernel: i2c-designware-pci 0000:00:08.0: i2c_dw_handle_tx_abort: lost arbitration

As you can see I get only 1 error. 

The code stays the same:

import mraa as m
import time

x = m.I2c(1)
x.address(0x74)

print("Envoi de la valeur ")
x.write(bytearray(b'0x00x01x01'))
x.write(bytearray(b'0xA50x080x410x000x120x4E0x040x52'))
print("Lecture Valeur")
#time.sleep(2)
#d = x.readByte()

#print(str(d))
d = x.read(10)

print(d)
print(str(d))
#print(int.from_bytes(d, byteorder='little'))

It is like the busy is somethimes busy.. sometimes not..
I can test in C if you think it can work differently. (or directly with ioctl if you thing i could help us to find out the issues)

For the pull-up we have a 2,1 kOhm on our board. 

thank you for your help

PaulB2Code, 2016-12-28:
I tried 

print("Envoi de la valeur ")
x.writeReg(0xFF,0xFF)
x.writeReg(0xFF,0xF0)

And it send sometimes (one time every 5 or 10 try) a two trame with exactly my second request 0x74 0xFF 0xF0.

PaulB2Code, 2017-01-06:
Hi Alex, 

How could we test to know if it is an electrical issue like wrong pullups ?

thank you for your help

alext-mkrs, 2017-01-06:
Well, attaching a scope, like you did in the previous comment, could be a good idea. That photo you've posted has all the important details like voltages and which color is what invisible, so I cant comment on that one, sorry. If you could make a more frontal one, capturing the whole screen and preferably without that glare, making parts of the picture invisible, I might be able so suggest something more meaningful.

PaulB2Code, 2017-01-09:
Hi Alex, 
The scale for the two signals is 1,00V per div. And the voltage is 3,3V for each signal.  For the next photo, i will pay attention to get all informations. 

This signal has been taken at the input of our sensor. 

I can do test as requested. 

thank you.

PaulB2Code, 2017-01-10:
Hi Alex, 

We buy a new I2C level shifter to do tests the PCA9306 (http://www.farnell.com/datasheets/1805226.pdf?_ga=1.189531691.1407728301.1477071045) .

We have 2 resistors of 10 kOhm between the compute module and the level shifter and the we have 2 resistors of 2.1 kOhm between the level shifter and the sensor. 

We analysed the signal and we arrived to received and send a signal sometimes. 
[image]

But we never arrived to send the first signal. 

[image]

Do you have an idea how we could solve it?

Thank you.

alext-mkrs, 2017-01-15:
This seems to be the Edison side of the translator, judging by the voltage being around 1.8V, right? How does the signal look like at the sensor side?

Not sure I understand your second screenshot, could you please elaborate?

All in all, I guess you might be better off asking in the Intel's support forum at https://communities.intel.com/, as support folks there may advise on the electronics side of it (maybe they even have the same sensor to try) as this does not look like anyting mraa-related.

alext-mkrs, 2017-02-25:
Closing for no activity for more than a month and actually rather a support forum question than mraa issue.

closed 2017-02-25 by alext-mkrs