eclipse-mraa/mraa #616: Compile .c in one command line

GuzMp, 2016-12-07:
Hi,

Actually, I use LoRa_SX1272 library (a library to use a LORA module) with my FTDI FT4222. So I had to modify LoRa_SX1272 in order to works with the libft4222.h and ftd2xx.h. And after I created a file .c (test.c) that use LoRa_SX1272 functions. So to compile test.c, I write this command line : "gcc -o test test.c LoRa_SX1272.c -lft4222", with this all works correctly.

Now I want to use mraa to switch between FT4222 and intel edison but SPI is not developed for ft4222 in mraa so I add some functions SPI for ft4222 in mraa, it works.

Now I want to use these functions (mraa_spi_frequency(spi, 1000) or mraa_spi_write_buf(spi, data, 2)) in  LoRa_SX1272 library so I add the mraa function in LoRa_SX1272 library and I creating a new file .c (LoRaMraa.c) that use LoRa_SX1272 library so that use also mraa library.

But I don't know how to compile LoRaMraa.c because normaly to compile I write "make" in mraa/build and it compile all file .c in mraa/examples but now to compile LoRaMraa.c I have to add LoRa_SX1272.c.

So do you have an idea to compile LoRaMraa.c with LoRa_SX1272.c and mraa ?
And how can I send you new code with SPI functions for ft4222 ?

Thank you.

arfoll, 2016-12-07:
So I'm not sure what you mean, likely you need to either add stuff in examples/CMakeLists.txt and remember to link against mraa (with -lmraa)? Have you got a concrete example of the code and what is failing with the compilation line etc?

To submit new code create a fork, make a branch and submit a pull request. More details are available here: https://github.com/intel-iot-devkit/mraa/blob/master/CONTRIBUTING.md.

GuzMp, 2016-12-07:
Ok so first if I want to compile spi_mcp4261.c alone in mraa-master/examples, how can I do without use _make_ ?
Because when I write _gcc -o spi_mcp4261  spi_mcp4261.c -lmraa_, it generates an executable but when I execute, it says _Initialization of spi failed..._

arfoll, 2016-12-07:
Ok so that's good, but it means your spi_mcp4261.c is failing initialisation of the spi bus. Without knowing your code it's hard to really tell you more. Maybe the syslog logs will be of help? Or use gdb to see where mraa_spi_init is failing

GuzMp, 2016-12-07:
Whereas if I compile in /build with make it works perfectly

arfoll, 2016-12-07:
ah ok, well maybe your linker is finding the wrong version of mraa? Check with `ldd` of both executables

GuzMp, 2016-12-07:
Yes you are right.
Executable in /build /examples: _libmraa.so.1 => /home/GuzMp/mraa-master/build/src/libmraa.so.1_
Executable in /examples : _/usr/local/lib/libmraa.so.1_

So how I can change the path when I compile with this _gcc -o spi_mcp4261 spi_mcp4261.c -lmraa_ ?

arfoll, 2016-12-07:
Good news :) Well you have a number of options, you could install libmraa.so.1 to /usr/local to replace it, you could simply copy the lib over, or you could play with /etc/ld.so.conf.d/ or LD_PRELOAD or LD_LIBRARY_PATH to get your loader to change it's mind about which lib to load!

GuzMp, 2016-12-07:
At final I use this _gcc -o spi_mcp spi_mcp4261.c -lmraa -Wl,-rpath,/home/GuzMp/mraa-master/build/src_ and yes it works perfectly, thank you very much!

alext-mkrs, 2016-12-07:
Good to see it's working for you now and thanks for reporting the solution back!

closed 2016-12-07 by alext-mkrs