eclipse-mraa/mraa #617: C++ isrExit returns MRAA_ERROR_INVALID_HANDLE

dhrynyk, 2016-12-08:
I have a C++ application running on Edison (firmware release 3.5; mraa version 1.5.1) with an ISR. When terminating the application, and releasing the resources, the isrExit function occasionally returns an “MRAA_ERROR_INVALID_HANDLE” and the thread, presumably to manage the interrupt routine, has not been terminated. This error is raised approximately 1 out 15 times the application is run.  

Steps I have taken to try to determine the cause include:
1.	Gdb trace of application to see if there are logic error;
2.	Review journal log for any related message (none were found); 
3.	Valgrind to find memory violations (none were found); and 
4.	Code walk-throughs. 

I am interested in knowing if there are any views on the appropriate actions to take when ending an ISR.  I did note one open issue in this ‘repository’ **mraa_gpio_isr_exit does not allow ISR thread to shutdown cleanly #437** however there were no follow ups to the original question. 
Also, as a suggestion, is there any opportunity to log the specific error codes for the mraa_gpio_isr_exit function as a way to try to locate the root cause?

Suggestion and questions welcomed.

Thanks
David

arfoll, 2016-12-08:
So #437 really only deals with isr overriding in subplatforms and is not relevant for Edison. We've seen issues like this in python but not in C++. You could try compile without HAVE_PTHREAD_CANCEL and see if that works better for you, that codepath was done for musl c lib support.

However since you get MRAA_ERROR_INVALID_HANDLE I'm guessing your code falls over at the first if (dev == NULL)? which means you may have called mraa_gpio_close() before mraa_gpio_isr_exit() which would invalidate your context and cause exactly this issue?

dhrynyk, 2016-12-13:
I would like to update you on my investigation into the “MRAA_ERROR_INVALID_HANDLE” issue.

- I confirmed that the mraa_gpio_isr_exit function was being called before the mraa_gpio_close function.
- I downloaded the gpio.c module and inserted diagnostics.  I found the issue was that pthread_close was returning an error code of ESRCH.  
- Using GDB I trapped any calls to pthread_close and pthread_join and confirmed that the only calls were coming from the mraa_gpio_isr_exit function. I observed only one call to the pthread_close function.
- Using GDB to trace pthread_close, I found a test of the value of pd->tid (using macro INVALID_TD_P) at the beginning of pthread_close.  The value of pd->tid less than or equal to zero results in the function returning the value of ESRCH. The value of pd->tid is consistently zero whenever there is a value of “MRAA_ERROR_INVALID_HANDLE” returned by the mraa_gpio_isr_exit function.
- Reviewing the contents of the structure pointed to by pd (thread data structure) did not reveal and obvious data corruption. 
- Using the GDB watch functionality I observed the value of pd->tid being set by the OS. 

Some thoughts and speculations I have with respect to the issue:

- I believe the tid value is dynamic and based on scheduling at the OS level.  This could explain why this error cannot be consistently reproduced. In some ways, the issue appears to be a form of a data race. 
- The C/C++ community may not be reporting the issue because they are not consistently testing the return codes, particularly on application termination.
- I believe the issue may rest in the multi-threading logic on the platform.  However, investigating the issue is beyond my knowledge of the OS internals.

Finally, as part of the investigation I developed syslog code to report out in the journal the error codes return by pthread_close and pthread_join.  I am happy to share the code. 

Please let me know your thoughts and suggestions.

Thank you,
David

alext-mkrs, 2017-04-08:
This has fallen through the cracks a bit, so let me check: @dhrynyk, is this still relevant? Have you been able to try @arfoll's suggestions about {{HAVE_PTHREAD_CANCEL}}?

alext-mkrs, 2017-08-10:
Closing forno response - feel free to comment if still actual.

closed 2017-08-10 by alext-mkrs

dhrynyk, 2017-08-10:
Hi Alex:

When I last tested the issue in MRAA version 1.7.0 it still occurred. 

I have found an alternative device which does not require an interrupt handle so closing the issue is not a problem. 

Thank you,
David

alext-mkrs, 2017-08-10:
There were no specific fixes for this one, as it's unclear if this is a bug at all, so 1.7.0 would make no difference against 1.5.1.

If anyone else faces this - please try @arfoll's suggestion of compiling without HAVE_PTHREAD_CANCEL first, to see if that makes a difference.