eclipse-mraa/mraa #618: npm install mraa is not working

chanhyeoni, 2016-12-11:
I got the following message 

```
> mraa@1.5.1 install /Users/xxx/node_modules/mraa
> node-gyp rebuild

  CC(target) Release/obj.target/mraa/src/x86/intel_gt_tuchuck.o
In file included from ../src/x86/intel_gt_tuchuck.c:32:
In file included from ../include/x86/intel_gt_tuchuck.h:31:
In file included from ../include/mraa_internal.h:35:
In file included from ../include/mraa_internal_types.h:31:
In file included from ../api/mraa/iio.h:28:
../api/mraa/iio_kernel_headers.h:113:10: fatal error: 'linux/ioctl.h' file not found
#include <linux/ioctl.h>
         ^
1 error generated.
make: *** [Release/obj.target/mraa/src/x86/intel_gt_tuchuck.o] Error 1
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:204:12)
gyp ERR! System Darwin 16.1.0
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /Users/changhyunlee/Desktop/Programming/JavaScript/iotapp-template-analog-read/node_modules/mraa
gyp ERR! node -v v6.0.0
gyp ERR! node-gyp -v v3.3.1
gyp ERR! not ok 
npm ERR! Darwin 16.1.0
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "mraa"
npm ERR! node v6.0.0
npm ERR! npm  v3.8.6
npm ERR! code ELIFECYCLE

npm ERR! mraa@1.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@1.5.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs mraa
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /Users/xxx/iotapp-template-analog-read/npm-debug.log
```

I tried the downgraded version of node, and nothing worked.

arfoll, 2016-12-11:
You don't seem to have the linux kernel headers installed?

arfoll, 2016-12-12:
As @tingleby points out, this is on mac OS X `gyp ERR! System Darwin 16.1.0`. We don't support installing libmraa on non linux targets.

closed 2016-12-12 by arfoll

alext-mkrs, 2016-12-12:
...and that's probably the build is run on Mac OS ("Darwin" is reported by npm in the error log). I don't think we support this yet - the mock mode could be built there, but there's no support for that on trunk (search the issues section for a discussion we had on that a while ago).