eclipse-mraa/mraa #619: mraa.Gpio causes core dump in SetAlignedPointerInInternalField

maihde, 2016-12-11:
While attempting to follow the IoT Joule getting started guide:

`
> var mraa = require('mraa')
undefined
> var led = mraa.Gpio(101)
FATAL ERROR: v8::Object::SetAlignedPointerInInternalField() Internal field out of bounds
Aborted (core dumped)
`

This is with the following MRAA version 

`
# mraa-gpio version
Version v1.5.1-11-gc1017bb on Intel GT Tuchuck
`

The relevant portion of the backtrace is:

`
#5  0x00000000005b9c91 in v8::Object::SetAlignedPointerInInternalField(int, void*) ()
#6  0x00007ffff458729b in SWIGV8_SetPrivateData(v8::Local<v8::Object>, void*, swig_type_info*, int) () from /usr/lib/node_modules/mraa/mraa.node
#7  0x00007ffff459f7b2 in _wrap_new_Gpio__SWIG_2(v8::FunctionCallbackInfo<v8::Value> const&, V8ErrorHandler&) () from /usr/lib/node_modules/mraa/mraa.node
#8  0x00007ffff459fb99 in _wrap_new_Gpio(v8::FunctionCallbackInfo<v8::Value> const&) () from /usr/lib/node_modules/mraa/mraa.node
#9  0x00000000005d760a in v8::internal::FunctionCallbackArguments::Call(void (*)(v8::FunctionCallbackInfo<v8::Value> const&)) ()
`

alext-mkrs, 2016-12-13:
Mraa version looks like if you've compiled it on your own. Have you?

Have you tried any other pins, e.g. 102? i don't have Joule to check this out, but Gpio under Node functionality is very basic and very well tested, so it looks a bit strange at the first glance.

maihde, 2016-12-13:
Per the Joule first-time setup guide, I followed the steps as described
here:

https://software.intel.com/en-us/updating-libraries-for-joule

And then attempted:

https://software.intel.com/en-us/node/672541

I've also tried other GPIO pins and had the same issue.

claytongulick, 2016-12-18:
I'm experiencing the same issue on the Edison

alext-mkrs, 2016-12-18:
@claytongulick, could you please share the source of the program you are running when you get that error, as well as describe how do you run/upload it (e.g. XDK, manually from the SSH console, etc)?

alext-mkrs, 2016-12-18:
Ok, never mind, I can see that as well on Edison, nothing fancy to reproduce, so looks like something broke this basic functionality:

```javascript
root@edison1:~/mraa-619# node
> var m = require('mraa')
undefined
> m.setLogLevel(7)
0
> var g = m.Gpio(3)
FATAL ERROR: v8::Object::SetAlignedPointerInInternalField() Internal field out of bounds
Aborted
```

Nothing overly interesting in the syslog:
```
Dec 18 20:03:17 edison1 libmraa[6917]: libmraa version v1.5.1-11-gc1017bb initialised by user 'root' with EUID 0
Dec 18 20:03:17 edison1 libmraa[6917]: edison: Arduino board detected
Dec 18 20:03:17 edison1 libmraa[6917]: libmraa initialised for platform 'Intel Edison' of type 2
Dec 18 20:03:28 edison1 libmraa[6917]: Loglevel 7 is set
Dec 18 20:03:47 edison1 libmraa[6917]: gpio219: owner: Set owner to 0
```

alext-mkrs, 2016-12-18:
Well, that's actually just a typo in the instruction. There's `new` missing in the `var led = mraa.Gpio(101)` line (and in my test as well). So nothing is broken and if you run the bundled example, e.g. `examples/javascript/GPIO_DigitalWrite.js`, changing the pin number inside to the one you are interested in, you'll see it works fine.

closed 2016-12-18 by alext-mkrs

claytongulick, 2016-12-18:
@alext-mkrs thanks, I'll try that! BTW - are you all accepting pull requests for this? Using new in js is sometimes considered a bit of an antipattern in an API, I could tweak the Gpio function to detect how it's being called and work either way, if you'd like.

maihde, 2016-12-18:
Thanks Alan.  Is there a mechanism to get the documentation authors to
update their docs?

Regards,
Michael

arfoll, 2016-12-19:
@claytongulick sure that'd be awesome. We've always had this 'new' requirement in mraa's JS API, I know it's considered an antipattern by some, i'd argue JS just isn't a very nice & clean language anyways so arguing about antipatterns is pretty pointless!

alext-mkrs, 2016-12-19:
@maihde, I guess that was question for me, as you're replying to my comnent? ;)

I have no idea, most probably there's some way for authors to correct their texts and looks like Intel support folks have already addresed that as necessary, they reported back in the thread I had started.