eclipse-mraa/mraa #620: mraa.Gpio / mraa-gpio list / pins not visible on Raspberry PI3

g-vidal, 2016-12-12:
Hi,
I am compiling a stretch image with kernel 4.8 for my raspi3 using @drtyhlpr  [rpi23-gen-image](https://github.com/drtyhlpr/rpi23-gen-image). `$ uname -a
Linux raspife3 4.8.13-v7+ #1 SMP Sun Dec 11 21:40:00 CET 2016 armv7l GNU/Linux
I` Compile node-6.9.2 and swig-3.0.10, then clone from here compile and install mraa. With previous versions (some time ago there was no mraa-gpio command) it worked perfectly but pins are no more accessible :

 *  either with python
```shell
sudo python
Python 2.7.12+ (default, Nov 22 2016, 00:48:54) 
[GCC 6.2.1 20161118] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
>>> x = mraa.Gpio(32)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 657, in __init__
    this = _mraa.new_Gpio(pin, owner, raw)
ValueError: Invalid GPIO pin specified
>>> x = mraa.Gpio(35)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/dist-packages/mraa.py", line 657, in __init__
    this = _mraa.new_Gpio(pin, owner, raw)
ValueError: Invalid GPIO pin specified
>>> 
```
 * or mraa-gpio 
```shell
mraa-gpio list
No Pins
mraa-gpio list
No Pins
```
GPIO devices are present as usual :
```shell
ls /dev/gpio*
/dev/gpiochip0  /dev/gpiochip1  /dev/gpiomem
```
but the /sys/class/gpio is slightly different from the previous versions
```shell
raspife3:/sys/class/gpio$ ls -al 
total 0
drwxr-xr-x  2 root root    0 janv.  1  1970 .
drwxr-xr-x 48 root root    0 janv.  1  1970 ..
--w-------  1 root root 4096 janv.  1  1970 export
lrwxrwxrwx  1 root root    0 janv.  1  1970 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
lrwxrwxrwx  1 root root    0 janv.  1  1970 gpiochip100 -> ../../devices/gpiochip1/gpio/gpiochip100
--w-------  1 root root 4096 janv.  1  1970 unexport
```
I have checked with low level interaction
```shell
# cd /sys/class/gpio/
root@raspife3:/sys/class/gpio# echo 24 > export 
root@raspife3:/sys/class/gpio# cd gpio24/
root@raspife3:/sys/class/gpio/gpio24#  echo out > direction 
root@raspife3:/sys/class/gpio/gpio24# echo 1 > value
root@raspife3:/sys/class/gpio/gpio24# echo 0 > value
root@raspife3:/sys/class/gpio/gpio24# 
```
It worked perfectly and lit up then switch off the led, which seems to demonstrate (?) that the GPIO bus is OK.

AM I doing something wrong is there a problem with the kernel 4.8 ? 
Thanks for any help

michael-ring, 2016-12-12:
The main obstacle is that the Raspberry Pi 3 is not (yet) supported with mraa, currently the max. supported Device is Raspi 2.
When you have a look at /var/log/messages (or equivalent) yoju should see a message like:

raspberrypi: Failed to detect platform revision

g-vidal, 2016-12-12:
Hi,
everything (mraa upm) has worked perfectly on the raspberry 3 (even not supported;-)) until my last compilation where I noticed changes in mraa (mraa-gpio and mraa-i2c). The other difference is the kernel version I used 4.8 instead of 4.4. I am away from my raspi, I'll check ASAP but mraa compilation and execution went well as far as I can remember, I'll also check i2c that was working perfectly. here is a sketch of the jupyter notebook that has been executed by various colleagues [BME280notebook](https://frama.link/S6K8aCEv)

g-vidal, 2016-12-13:
I can confirm that mraa works perfectly with raspberry Pi3 but **not with any kernel** proposed on the  [raspberrypi/linux](https://github.com/raspberrypi/linux). 
My problem seems to come from the installation of kernel 4.8.12 I'll investigate that, if anybody has a hint please tell me.

I propose to close this trend even if the error remains with kernel 4.8 thinks work well with 4.4 which is the default

closed 2016-12-13 by g-vidal

cbosdo, 2017-03-03:
@g-vidal what do you have in the `/proc/cpuinfo` ? It would be good to also know what is in your `/proc/device-tree/compatible`.

zhangeven, 2018-12-03:
@g-vidal I have the same issue,but I resolved this by calling mraa.Gpio API like this:
pin = mraa.Gpio(2,owner=False,raw=True)