eclipse-mraa/mraa #621: ZFM20 get image from buffer to upper computer

bartekwozniak93, 2016-12-12:
I have problems with ZFM20 library. I am trying to get a raw image from the sensor to be able to synchronize fingerprints with more than one sensor later. I found that somebody had tried to do it as well (https://github.com/intel-iot-devkit/upm/issues/404) but there is no code provided. 

I succeeded in applying writeCmdPacket function but still don't know how to get a raw image data. This is my code: 

var pkt = new fingerprint_lib.uint8Array(1);
pkt.setitem(0, 0x0a); 
myFingerprintSensor.writeCmdPacket(pkt, 1); 
 ...

Could you help me, please? This is very important to me because this issue depends on my whole school project.

jontrulson, 2016-12-12:
There seems to be quite a lot of people who wanted to do that, google can help there.  In short I found these:

A Java program that can apparently do this:
https://github.com/mellongreen/ZFM20Module

A thread on Adafruit, which contains a link to an English datasheet.
https://forums.adafruit.com/viewtopic.php?f=19&t=77436
I think what you want is on page  17

A fairly long thread on the subject, with some description here:
https://forums.adafruit.com/viewtopic.php?f=22&t=33229&hilit=template&start=60#p359231

This should help you get going.  At the time we did this driver, we did not even have access to the english datasheet :)  So you should be able to figure out what is going on.  Good luck!

bartekwozniak93, 2016-12-12:
Thank you for your reply. I really appreciate your work (even more!-  you did it without English datasheet) 👍  

Unfortunately, I have still some problems to make it work. 
Before I posted my question, I had been browsing these websites over many hours. 

According to your link to a java program, using 'Getting raw image' is misleading because it is used for putting a fingerprint image to buffer not for uploading to the upper computer (pc).

With an English datasheet I managed to write the following code: 
...
var SerialPort = require("serialport");
var port = "/dev/ttyMFD1";
var serialPort= new SerialPort(port, {
 baudrate: 57600
});
//after enrolling fingerprint to sensor/storing fingerprint in buffer) 
...
var pkt = new fingerprint_lib.uint8Array(1);
pkt.setitem(0, 0x0a);
myFingerprintSensor.writeCmdPacket(pkt, 1);

//get data from UART
var isDataAvailable = myFingerprintSensor.dataAvailable(1000);
if(isDataAvailable){
 serialPort.on('open', function(error) {
  if(error) {
   console.log('error');
  } else {
   serialPort.on('data', function(data) {
    console.log(data);
   });
 }
});
...
}

The output of my program are few buffers eg: 
<Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...>
<Buffer 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 84 ef 01 ff ff ff ff 02 00 82 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 ...>
...

I don't know whether I got the proper result. 

I know that "The image is 256 pixels in width and 288 pixels in height, giving a total of 256*288 = 73728 pixels or bytes. the module takes the high nibble (4 bits) of one pixel and the low nibble of an adjacent pixel to create an 8-bit value that contains the data for both pixels. This way, the module transmits only 73728/2 = 36864 bytes.. " 

Could you give me some more specific hints to make it work? :)

alext-mkrs, 2016-12-13:
Shouldn't this be in upm's repo instead of mraa's?

arfoll, 2016-12-13:
Yes, the issue probably should... as long as @intel-iot-devkit/upm-core-dev are aware I think there's no real harm..

bartekwozniak93, 2016-12-13:
ups.. sorry Do you want me to recreate this issue?

arfoll, 2016-12-13:
Dont worry about it. Id open a new one in upm's issue tracker just for
clarity and link back to this for context

jontrulson, 2016-12-13:
In the Java example, look at the upImg() function and it's dependants.  That does look like what you need.

> Could you give me some more specific hints to make it work?

I'm sorry, I can't.  I haven't done it myself, nor do I have the time to try.

alext-mkrs, 2017-05-06:
@bartekwozniak93, is this still actual?

alext-mkrs, 2026-02-15:
Ok, answering my own question from 9 years ago - it's apparently not :) Closing.

closed 2026-02-15 by alext-mkrs