eclipse-mraa/mraa #622: Intel System Studio IoT for Windows 404 page (not an mraa issue)

claytongulick, 2016-12-19:
Hi Intel folks, I know this is only loosely related to mraa, but I can't use mraa or develop until it's resolved.

There's currently no way to download the Intel System Studio IoT for Windows, and hasn't been for a while now, following the links just gives a 404. I've tried leaving messages in the communities section, but haven't had any response.

I'm pretty sure Intel doesn't want 404's on their public site, especially for IoT right now, so I'm just trying to bring this to anyone's attention. Sorry about the github issue in the wrong place, but is there any way you could send an email to the correct folks to get the download working again? Thanks!!

arfoll, 2016-12-19:
Have you tried just grabbing the installer here:
https://software.intel.com/en-us/iot/hardware/joule/downloads
or:
https://software.intel.com/en-us/iot/hardware/edison/downloads

All seems to work?

arfoll, 2016-12-27:
Closing as no update

closed 2016-12-27 by arfoll

claytongulick, 2016-12-29:
Thanks @arfoll, I checked and the download is working again. For several days, it was broken and giving a 404. Thanks again!

arfoll, 2016-12-29:
No worries, I've shared your findings with our sysadmin team. Thanks for
notifying us and we'll try do better!