eclipse-mraa/mraa #624: Error 99 returned when mraa_i2c_write(m_i2c, data, length); is called, but i2cdetect can see device.

Pillager225, 2016-12-26:
I am using mraa on the Intel Edison with Arduino breakout board. I am using the library in its c++ distribution. I initialized an mraa::I2c object using the argument for bus 6. I compiled with g++ using the c++11 standard, and -lmraa argument. libmraa is installed on the Edison I am using. I would love to provide any other information you would need to help me. You can view my code at [here](https://github.com/KittsSeniorDesign/Pozyx_mraa), but specifically I am using mraa in the file named Wire.cpp. I am trying to make an Arduino library work for the Intel Edison. The library interfaces with an I2C device called Pozyx which does indoor localization and is visible at address 0x4B on bus 6. You can check out Pozyx [here](https://www.pozyx.io/) if you are interested, and their documentation on how to interface with it over I2C [here](https://www.pozyx.io/Documentation/Datasheet/Interfaces#i2c). Using gdb, I was able to see that the code I wrote can step through Pozyx, my Wire.cpp, and mraa::i2c, but when mraa_i2c_write(m_i2c, data, length); is called on line 206 in i2c.hpp, code 99 is returned which means error unspecified. How can I learn how to fix this error, or what causes it? Using gdb I have determined that m_i2c is not null. Please tell me what more I can tell you to help me. 

I am a college student doing this to get my bachelors degree. This is a very small part of my project, but it is holding my team up. We would very much like to be able to interface with Pozyx over I2C on the Intel Edison, and we hope that mraa is able to help us reach our goal. Thank you for the work you have done on mraa, and the help you will provide to us.

arfoll, 2016-12-27:
Where in your code are you getting stuck in the wire.cpp? I've only had a quick glance and couldn't quite work out what you're doing. But maybe start with something super basic in python like this:
```
import mraa as m
bus = m.I2c(6)
bus.address(0x4b)
print (int) bus.readReg(0x00)
```

From a quick read of the datasheet this should return 0x43 - https://www.pozyx.io/Documentation/Datasheet/RegisterOverview. At least this way we know if your code or if mraa is not working right. If it's your code then I suggest you send a list of only the mraa calls you are making and their order. But starting with checking the device responds correctly with a basic register read is always my first recommendation.

You also don't mention the version of mraa you're using, I'd recommend using 1.x or newer on an edison board. 1.5.1 is obviously recommended ;-)

Good luck with your project!

@intel-iot-devkit/upm-core-dev have you seen this device before? Looks quite cool

Pillager225, 2016-12-27:
In Wire.cpp it gets stuck in Wire::endTransmission(uint8_t), specifically line 28 where i2c->write(txBuffer, (int)txBufferLength); is called. Wire::endTransmission(uint8_t) is actually called by Pozyx::i2cWriteRead(uint8_t*, int, uint8_t*, int) in Pozyx_core.cpp on line 605. The program I am using to test the code is compiled from ready_to_localize.cpp, where the first thing it does is exactly what that python script you provided does (line 102).

After making sure I had mraa 1.5.1, I ran the python script you provided. Here was the output: 
`Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python2.7/site-packages/mraa.py", line 1175, in readReg
    return _mraa.I2c_readReg(self, reg)
ValueError: Unknown error in I2c::readReg()`
I assume the underlying error is code 99. Could it be because the way Pozyx wants I2C done? There documentation on their website is pretty thorough, but I do not know how mraa does its I2C.

arfoll, 2016-12-27:
Well that's good news really, we can reproduce in 4 lines :) This would indicate we're simply not seeing your i2c device.

I think actually this may due to mraa not respecting the arduino 'shield' behaviour. Try connecting your i2c lines on your shield to the i2c SCA/SCL lines directly. See if that works.

alext-mkrs, 2016-12-27:
@pillager225, you mentioned that i2cdetect can see the board, so it's unlikely to be the voltage mismatch, but just to be sure - please check that IOREF jumper is set to the 5V position and not the 3.3V one.

alext-mkrs, 2017-01-02:
Well, we are in the electronics land with this issue, so I guess yes, 0.4 volts may make a difference here (especially if there are further dropouts when the setup tries to communicate)

Pillager225, 2017-01-12:
So I wired the ioref pin to 5.1 volts. That 5.1 volts was relative to the same ground that the edison was being powered by. I still got error code 99. 

I figure that the best thing I can do is install the mraa c library with the source included so that I can debug that portion of the code. I'll let you know of any progress I make in that area.

Also, I know Intel is very close to my school, Santa Clara University, so if there is somebody willing to come by the SCU Robotic Systems Lab to check things out, please let me know.

arfoll, 2017-01-12:
Sadly I don't think we have anyone in the teams in SC, development of mraa is in Europe, the UPM team is in Portland, US.

The only thing I can recommend is grabbing a logic analyser and having a look at what's happening on the SCL/SDA lines. If you don't have one, you can gab very cheap USB based ones online, but I'm sure your university will have something you can use, especially in a robotics lab!

Pillager225, 2017-01-23:
I have yet to use a logic analyzer, but I will do that sometime this week most likely. 

Do you guys know compile mraa so that when you debug a program using mraa using gdb you can step through the c code that is executed? I would like to chase the error down as far as I can in the mraa library to get more information as to why I am getting the error.

arfoll, 2017-01-23:
Look at docs/building.md. there's instructions for building a debug version

alext-mkrs, 2017-02-25:
@Pillager225, is this still actual, how can we help you further on this if yes?

Pillager225, 2017-02-27:
I don't know how you guys can help. I have been unable to find the time to use the logic analyzer at my school and because this is part of a project related to my schoolwork, I needed to move forward. I did get images of the waveforms caused by the edison when trying to talk to pozyx, and they matched the ones that come out of a arduino which is able to communicate with pozyx over i2c. The only difference is that pozyx does not acknowledge the edison. I cannot, at this time, invest more time into trying to get i2c to work. Thank you guys for your bits of advice and help when trying to solve this issue. It is likely that the problem has more to do with pozyx then mraa.

closed 2017-02-27 by Pillager225