eclipse-mraa/mraa #625: FTDI : Combine GPIO and SPI

GuzMp, 2016-12-27:
Hi,

I use FT4222 with SPI and know I want to use also GPIO3 as Interrupt. So first I tried both separately, it works correctly but after if I use both in a same code, it fail at nivel of **x = mraa_gpio_init(515);** I can't init both in a same code ?

My code : 
_mraa_init();
  	spi = mraa_spi_init(512);
  	mraa_spi_mode(spi, MRAA_SPI_MODE0);
  	mraa_spi_frequency(spi, 5000000);

	mraa_gpio_context x;
    	x = mraa_gpio_init(515);
	
	if (x == NULL) {
		printf("Echec\n");
        	return 1;
   	}

mraa_gpio_dir(x, MRAA_GPIO_IN);
mraa_gpio_edge_t edge = MRAA_GPIO_EDGE_BOTH;
mraa_gpio_isr(x, edge, &DecaDuino::isr0, NULL);_

It's possible to combine both ?

In syslog error : 

_Dec 27 17:26:59 teosit-System-Product-Name libmraa[12311]: libmraa version v1.5.1 initialised by user 'root' with EUID 0
Dec 27 17:26:59 teosit-System-Product-Name libmraa[12311]: Platform not supported, not initialising
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: FT_GetDeviceInfoList returned 2 devices
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: mraa_ftdi_ft4222_init completed successfully
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: FT4222_GetVersion 42220200 01030178
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: libmraa initialised for platform 'Unknown platform + FTDI FT4222' of type 98
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: spi512_init: Using sub platform
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: spi_init: Selected bus 0
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: SPI MODE : 1363677943
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: gpio515: init: Using sub platform
Dec 27 17:27:00 teosit-System-Product-Name libmraa[12311]: gpio515: init: Sub platform not initialised_

Thank you.

arfoll, 2016-12-27:
Something v.odd there, on your branch are you sure you don't corrupt the suplatform board struct when you initialise the spi bus or something? There must be a bug somewhere as this works for I2c + gpio...

arfoll, 2016-12-29:
> Has anyone tested FT4222 SPI capability? I wrote most of the FT4222 code
> but couldn't test SPI operation and I didn't have any suitable hardware.

It's in the branch for now in @guzmp's fork. To test spi is easy if it does
full duplex, just short mosi to miso. Or walk down to the upm guys and
steal a sensor ;-)

alext-mkrs, 2017-02-25:
Is this still actual?

alext-mkrs, 2017-03-05:
Ping...

whbruce, 2017-03-05:
It seems that @GuzMp has got most of the way towards adding SPI support to FTDI sub platform but fallen at the last hurdle. I don't believe this a bug, more a request for help. I've starred his repo and will take a look, but won;t be able to get to this for some time. I'd tag this as "feature request" and "help needed".

alext-mkrs, 2017-03-07:
Thanks! Tagged.