eclipse-mraa/mraa #626: Build error

wallacejsb, 2017-01-02:
Running the command:  **npm install mraa**
Using node v7.3.0 and npm v4.0.5.
Result:
make: *** [mraa.target.mk:141: Release/obj.target/mraa/src/mraajsJAVASCRIPT_wrap.o] Error 1
make: Leaving directory '/usr/lib/node_modules/mraa/build'
gyp ERR! build error 
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.8.13-1-ARCH
gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /usr/lib/node_modules/mraa
gyp ERR! node -v v7.3.0
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm ERR! Linux 4.8.13-1-ARCH
npm ERR! argv "/usr/bin/node" "/usr/bin/npm" "install" "-g" "mraa"
npm ERR! node v7.3.0
npm ERR! npm  v4.0.5
npm ERR! code ELIFECYCLE

npm ERR! mraa@1.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@1.5.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the mraa package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs mraa
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls mraa
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /home/wallace/npm-debug.log

arfoll, 2017-01-02:
Mraa doesn't work with node 7.x. you'll have to downgrade to 6.x. see other
bugs

wallacejsb, 2017-01-02:
Using nvm, problem solved..

closed 2017-01-02 by wallacejsb