eclipse-mraa/mraa #627: Segfault when creating deb package on Ubuntu 16.04

whbruce, 2017-01-05:
Steps to reproduce
```
$ cmake -DIPK=ON ..
<snip>
$ make
<snip>
$ cpack
CPack: Create package using DEB
CPack: Install projects
CPack: - Run preinstall target for: mraa
CPack: - Install project: mraa
CPack: Create package
CPack: Package mraa_1.5.1.16_amd64.ipk generated.
Segmentation fault (core dumped)
```
IPK package has been successfully created, segfault would appear to be caused by deb packaging.
```
$ ls
arch.c               CPackConfig.cmake        examples                 src
CMakeCache.txt       _CPack_Packages          install_manifest.txt     tests
CMakeFiles           CPackSourceConfig.cmake  Makefile
cmake_install.cmake  CTestTestfile.cmake      mraa_1.5.1.16_amd64.ipk

```

Using gdb shows segfault is thrown by strlen()
```
Program received signal SIGSEGV, Segmentation fault.
strlen () at ../sysdeps/x86_64/strlen.S:106
106	../sysdeps/x86_64/strlen.S: No such file or directory.
(gdb) where
#0  strlen () at ../sysdeps/x86_64/strlen.S:106
#1  0x00000000004fab01 in cmCPackDebGenerator::createDeb() ()
#2  0x0000000000504ddb in cmCPackDebGenerator::PackageFiles() ()
#3  0x00000000004d3256 in cmCPackGenerator::DoPackage() ()
#4  0x00000000004c4f2d in main ()

```
Failure seems specific to DEB packaging, RPM package can be built successfully.

whbruce, 2017-01-05:
I get the same crash with UPM, so I guess I have a cmake/cpack problem.

tingleby, 2017-01-05:
We have some hacks in the cmake/modules that could be effecting this.

I also do NOT recommended packaging using CPack. Use the native package utilities. On Ubuntu the debian packaging tools "dh".

The latest debian/*
https://launchpad.net/~mraa/+archive/ubuntu/mraa/+files/mraa_1.5.1-2.debian.tar.xz

whbruce, 2017-01-05:
Sorry, didn't notice addition of Ubuntu install instructions.
Maybe we should remove cpack support.

arfoll, 2017-01-05:
Yeah probably but the ipk repos for Edison still work that way... It's one
of those terrible solutions that still works ehehe

closed 2017-01-25 by tingleby