eclipse-mraa/mraa #628: Support for Node js initJsonPlatform

nigeltiany, 2017-01-08:
This would be a great feature. Everything boils down to c++ with this library (correct me if I'm wrong) so initJsonPlatform from a higher level language would mean some cross compiling of some sort? Is this impossible? What need be done to have this feature for node js?

alext-mkrs, 2017-01-08:
JS/Node.js bindings are available for mraa out of the box, see examples dir for specifics. Is that what you're asking about?

nigeltiany, 2017-01-08:
No, I'm asking about the feature that allows initializing a new IOT board like the [example here](https://software.intel.com/en-us/blogs/2016/08/30/supporting-a-new-som-baseboard-with-mraas-json-platform-api). The example (in python) uses a method called initJsonPlatform which converts sysfs pins to mraa compatible pin mapping(I think). The mraa documentation says initJsonPlatform is not available for mraa through NPM. 

Follow up question...
How can I compile a version of mraa and include a json file with the unsupported board and have the compiled version usable with the new board?

arfoll, 2017-01-08:
ah right, you're 100% correct of course! So the reason you can't is because npm doesn't have conditional compilation (at least not easily...) so we would either require libjson-c-dev installed on all machines or disable it (which is what we chose).

Basically compile mraa from source properly as show in docs/building.md and you can build a version of the mraa.node with -DJSONPLAT=ON. It really shouldn't be that hard, but you do need the node.js dev headers on your board which is sometimes a little tricky. Note that the .travis.yml file with it's setting of -DNODE_ROOT_DIR:PATH="${NVM_BIN}/.. essentially shows you how to use nvm to do this, which is probably something I should document :)

alext-mkrs, 2017-02-25:
Actually, I'd recall my "enhancement" tag as this seems to be a question, already answered. @arfoll, do you think we can close this one, or you want to follow up, e.g. on
><...> setting of -DNODE_ROOT_DIR:PATH="${NVM_BIN}/.. essentially shows you how to use nvm to do this, which is probably something I should document :)

?

nigeltiany, 2017-06-12:
@alext-mkrs @arfoll Building the source needs to be done on the board? Please outline the steps I need to take. I have built from source with -DJSONPLAT=ON Goal: pass JSON file with pinMappings get javascript library

alext-mkrs, 2017-06-12:
So you've built it, now run "make install" and (assuming you've set prefix to /usr) everything will get installed into system-wide locations and after that it boils down to calling proper mraa's init function in your program  passing your JSON file to it.

nigeltiany, 2017-06-12:
@alext-mkrs I don't understand. I thought I can't call initJsonPlatform from compiled mraa javascript. I didn't have the JSON set up when I built the code. Assuming now the JSON pin mappings are in "/usr/PlatFormName.json" and I rebuild the code again, with javascript folder contain a javascript mraa package that works with my board?

alext-mkrs, 2017-06-13:
Okay, so looks like you misunderstand the way this functionality works :) After you have compiled mraa with the support for this functionality (and that's why you had to do it manually - the way `npm` builds mraa makes it impossible to enable this feature), you can just run the `initJsonPlatform()` function, passing it a JSON file you've prepared, and you should be done. It's not like the JSON file is somehow compiled into mraa, it's used dynamically and it's just a matter of enabling this feature.

nigeltiany, 2017-06-15:
@alext-mkrs Thanks, I understand now. Closing

closed 2017-06-15 by nigeltiany