eclipse-mraa/mraa #629: Installation failed on raspberry pi 2

ppopadiyn, 2017-01-10:
Hi,

I tried installing mraa on raspberry pi 2 via `npm install mraa`. And the installation failed at the following line: 
`mraa.target.mk:139: recipe for target 'Release/obj.target/mraa/src/mraajsJAVASCRIPT_wrap.o' failed`.

At the end is the full output of the 'npm install' command. I checked the various build requirements. Not sure what I need and what I don't, but this is what I have on the board:

- node v6.9.1
- npm v4.0.5
- swig 3.0.11
- Python 2.7.3
- GNU Make 4.0
- cmake version 2.8.9
- gcc version 4.8.2

Any ideas ? Thanks in advance.

Here is the complete npm output

```
> mraa@1.5.1 install /home/pi/work/test-mraa/node_modules/mraa
> node-gyp rebuild

make: Entering directory '/home/pi/work/test-mraa/node_modules/mraa/build'
  CC(target) Release/obj.target/mraa/src/x86/intel_gt_tuchuck.o
  CC(target) Release/obj.target/mraa/src/x86/up.o
  CC(target) Release/obj.target/mraa/src/x86/intel_cherryhills.o
  CC(target) Release/obj.target/mraa/src/x86/intel_sofia_3gr.o
  CC(target) Release/obj.target/mraa/src/x86/intel_minnow_byt_compatible.o
  CC(target) Release/obj.target/mraa/src/x86/intel_nuc5.o
  CC(target) Release/obj.target/mraa/src/x86/intel_de3815.o
  CC(target) Release/obj.target/mraa/src/x86/intel_edison_fab_c.o
../src/x86/intel_edison_fab_c.c: In function ‘mraa_intel_edison_aio_init_pre’:
../src/x86/intel_edison_fab_c.c:335:13: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     if (aio > plat->aio_count) {
             ^
../src/x86/intel_edison_fab_c.c: In function ‘is_arduino_board’:
../src/x86/intel_edison_fab_c.c:1213:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (i=0; i<(sizeof(gpiochip_idx)/sizeof(gpiochip_idx[0])); i++) {
                ^
  CC(target) Release/obj.target/mraa/src/x86/intel_galileo_rev_g.o
  CC(target) Release/obj.target/mraa/src/x86/intel_galileo_rev_d.o
  CC(target) Release/obj.target/mraa/src/x86/x86.o
  CC(target) Release/obj.target/mraa/src/arm/banana.o
../src/arm/banana.c: In function ‘mraa_banana_mmap_write’:
../src/arm/banana.c:110:24: warning: unused variable ‘addr’ [-Wunused-variable]
     volatile uint32_t* addr;
                        ^
../src/arm/banana.c: In function ‘mraa_banana’:
../src/arm/banana.c:562:52: warning: assignment discards ‘const’ qualifier from pointer target type [enabled by default]
         b->uart_dev[b->uart_dev_count].device_path = serialdev[uart3];
                                                    ^
../src/arm/banana.c:569:52: warning: assignment discards ‘const’ qualifier from pointer target type [enabled by default]
         b->uart_dev[b->uart_dev_count].device_path = serialdev[uart4];
                                                    ^
../src/arm/banana.c:575:52: warning: assignment discards ‘const’ qualifier from pointer target type [enabled by default]
         b->uart_dev[b->uart_dev_count].device_path = serialdev[uart7];
                                                    ^
../src/arm/banana.c:586:52: warning: assignment discards ‘const’ qualifier from pointer target type [enabled by default]
         b->uart_dev[b->uart_dev_count].device_path = serialdev[uart2];
                                                    ^
  CC(target) Release/obj.target/mraa/src/arm/beaglebone.o
../src/arm/beaglebone.c: In function ‘mraa_beaglebone_pwm_init_replace’:
../src/arm/beaglebone.c:329:13: warning: format ‘%s’ expects a matching ‘char *’ argument [-Wformat=]
             syslog(LOG_ERR, "pwm: Failed to open %s for writing, check access rights for user");
             ^
../src/arm/beaglebone.c: In function ‘mraa_beaglebone’:
../src/arm/beaglebone.c:379:18: warning: variable ‘uart3_enabled’ set but not used [-Wunused-but-set-variable]
     unsigned int uart3_enabled = 0;
                  ^
[... 180 lines cut ...]
gyp ERR! stack     at ChildProcess.onExit (/usr/local/lib/node_modules/npm/node_modules/node-gyp/lib/build.js:276:23)
gyp ERR! stack     at emitTwo (events.js:106:13)
gyp ERR! stack     at ChildProcess.emit (events.js:191:7)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:215:12)
gyp ERR! System Linux 4.1.19-v7+
gyp ERR! command "/usr/local/bin/node" "/usr/local/lib/node_modules/npm/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
gyp ERR! cwd /home/pi/work/test-mraa/node_modules/mraa
gyp ERR! node -v v6.9.1
gyp ERR! node-gyp -v v3.4.0
gyp ERR! not ok 
npm WARN test-mraa@1.0.0 No description
npm WARN test-mraa@1.0.0 No repository field.
npm ERR! Linux 4.1.19-v7+
npm ERR! argv "/usr/local/bin/node" "/usr/local/bin/npm" "install" "mraa"
npm ERR! node v6.9.1
npm ERR! npm  v4.0.5
npm ERR! code ELIFECYCLE

npm ERR! mraa@1.5.1 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the mraa@1.5.1 install script 'node-gyp rebuild'.
npm ERR! Make sure you have the latest version of node.js and npm installed.

```

AnandVetcha, 2017-01-15:
Following article mentions the steps to be followed to install MRAA on Rasberry Pi:
https://www.ics.com/blog/getting-started-mraa-raspberry-pi
Did you try steps mentioned in above link?

alext-mkrs, 2017-01-15:
Looks like there's something wrong with the swig-generated code (just one example, there are more errors):
```
../src/mraajsJAVASCRIPT_wrap.cxx: In function ‘SwigV8ReturnValue _wrap_I2c_read(const SwigV8Arguments&)’:
../src/mraajsJAVASCRIPT_wrap.cxx:7066:54: error: no matching function for call to ‘New(char*, int&)’
     jsresult = node::Buffer::New((char*) arg2, result);
```

In general, SWIG has problems with recent versions of v8, as they have [no active JS section maintainer](https://github.com/swig/swig/issues/804). Node 7, for example is broken. Try downgrading Node to something out of 5.x family, that used to work and is actually built for each mraa commit.

alext-mkrs, 2017-02-25:
@ppopadiyn, have you tried suggestions from this thread, is this still actual at all?

ppopadiyn, 2017-02-27:
Hi there,

After some pause (almost 2 months)... sorry for the delay, I tried installing mraa this morning again. I am using the latest distribution of raspbian jessy with pixel on raspberry 3. This time I tried using nodejs v4.8.0 as you suggested. Unfortunately, I am still not able to install mraa using the ```npm install mraa``` command. I am getting exactly the same error :(. That's why I tried to follow the instructions from here https://www.ics.com/blog/getting-started-mraa-raspberry-pi
I think I have some progress, but now I am getting the following error:

```
module.js:434
  return process.dlopen(module, path._makeLong(filename));
                 ^

Error: libmraa.so.1: cannot open shared object file: No such file or directory
    at Error (native)
    at Object.Module._extensions..node (module.js:434:18)
    at Module.load (module.js:343:32)
    at Function.Module._load (module.js:300:12)
    at Module.require (module.js:353:17)
    at require (internal/module.js:12:17)
    at Object.<anonymous> (/home/pi/Documents/Iot/mraa/examples/javascript/test.js:2:9)
    at Module._compile (module.js:409:26)
    at Object.Module._extensions..js (module.js:416:10)
    at Module.load (module.js:343:32)
```

arfoll, 2017-02-27:
So this error indicates you don't have libmraa.so.1 in your LD_LIBRARY_PATH? so did you maybe forget to add -DCMAKE_INSTALL_PREFIX=/usr and ended up with installing stuff in /usr/local?

alext-mkrs, 2017-02-27:
Most probably, that article instructs to run 'cmake .. ' and then just 'make install', without any additional parameters.

ppopadiyn, 2017-02-28:
Yes, I have't used -DCMAKE_INSTALL_PREFIX=/usr. After I did a second try with ```cmake -DCMAKE_INSTALL_PREFIX=/usr .. ```, everything seems to work fine for now and I managed to _require('mraa')_  in a javascript file.
At some point I would definitely try again to install it via ```npm install```, but I guess this should be enough for now to start playing with it.

closed 2017-02-28 by ppopadiyn