eclipse-mraa/mraa #630: mraa with firmata conflicts with modemmanager

skiselev, 2017-01-13:
modemmanager apparently causes Arduino 101 board to be reset in an endless loop. As the result firmata doesn't work. The workaround is to uninstall modemmanager (it might be also possible to remove some udev rules).

Platform: Intel NUC DE3815TYKE running Ubuntu 16.04 x86_64
MRAA version: 1.5.1-2 (installed from MRAA PPA repository)
Syslog:
Jan 13 14:36:24 skiselev-desktop kernel: usb 1-3: New USB device found, idVendor=8087, idProduct=0ab6
Jan 13 14:36:24 skiselev-desktop kernel: usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan 13 14:36:24 skiselev-desktop kernel: usb 1-3: Product: Atlas Edge
Jan 13 14:36:24 skiselev-desktop kernel: usb 1-3: Manufacturer: Intel
Jan 13 14:36:24 skiselev-desktop kernel: usb 1-3: SerialNumber: 00.01
Jan 13 14:36:24 skiselev-desktop kernel: cdc_acm 1-3:1.0: ttyACM0: USB ACM device
Jan 13 14:36:24 skiselev-desktop mtp-probe[8815]: checking bus 1, device 56: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3"
Jan 13 14:36:24 skiselev-desktop mtp-probe[8815]: bus: 1, device: 56 was not an MTP device
Jan 13 14:36:24 skiselev-desktop org.freedesktop.fwupd[673]: (fwupd:1720): libdfu-WARNING **: Failed to set new device: replugged device is not DFU-capable
Jan 13 14:36:27 skiselev-desktop org.gnome.evolution.dataserver.Sources5[1260]: ** (evolution-source-registry:1549): WARNING **: secret_service_search_sync: must specify at least one attribute to match
Jan 13 14:36:40 skiselev-desktop ModemManager[632]: <info>  Creating modem with plugin 'Generic' and '1' ports
Jan 13 14:36:40 skiselev-desktop ModemManager[632]: <warn>  Could not grab port (tty/ttyACM0): 'Cannot add port 'tty/ttyACM0', unhandled serial type'
Jan 13 14:36:40 skiselev-desktop ModemManager[632]: <warn>  Couldn't create modem for device at '/sys/devices/pci0000:00/0000:00:14.0/usb1/1-3': Failed to find primary AT port

alext-mkrs, 2017-01-15:
In CODK they simply [uninstall modemmanager](https://github.com/01org/CODK-M/blob/master/Makefile#L71) during the preparational target run. Not sure we'd want to do anything like that (and then again, this would only work for those who install from sources and not packages).

Sounds like something to add to the docs as a note to user to make sure it's uninstalled, what do you think, @arfoll?

arfoll, 2017-01-15:
Uhm it's a tricky one, modemmanager seems to just assume everthing in
/dev/ttyACMx is a modem, which uhm isn't an entirely bad assesment. Really
the arduino boards shoudl probably be asigned as something else. Likely the
best way to do this would be get users to create a udev rule to reassign
the firmata device node to smething like /dev/ttyFirmata0 rather than force
them to uninstall modemamanger.

alext-mkrs, 2017-01-16:
Oh, that's a good idea, I haven't looked too deep into this, but yeah, if all what ModemManager looks at is the device file name, your suggestion makes a lot of sense and is definitely better than asking people to uninstall it :)

tingleby, 2017-01-25:
Does this not effect the normal workflow for Arduinos too?

Propanu, 2017-02-02:
Confirmed, and fixed for my instance by disabling the ModemManager service. A custom udev rule would be a much nicer way to address this conflict.

w4ilun, 2017-04-06:
> Does this not effect the normal workflow for Arduinos too?

@tingleby I was wondering about that too, wouldn't this issue affect Arduinos in general, regardless of MRAA?

tingleby, 2017-04-07:
@w4ilun Yes Arduinos are blacklisted by ModemManager so they are ignored by it. We can add this upstream but could take some time

In the meantimeI have updated the ppa to mraa 1.6.1 and added some udev rules to the imraa package which should fix this.

w4ilun, 2017-04-11:
So I tried on a fresh install with Ubuntu Server 16.04.2. Once I have all the packages installed, I plugged in the 101 and rebooted. I then tried using mraa with mraa.Gpio(512+13) with no luck. I then tried it with mraa.addSubplatform... and it worked, so that confirms Firmata has been flashed but something else was wrong. I checked the logs and got the following:

Apr 10 20:44:27 ubuntu-nuc systemd[1]: Starting imraa...
Apr 10 20:44:27 ubuntu-nuc libmraa[1003]: libmraa version v1.6.1 initialised by user 'root' with EUID 0
Apr 10 20:44:27 ubuntu-nuc libmraa[1003]: Platform not supported, not initialising
Apr 10 20:44:27 ubuntu-nuc libmraa[1003]: imraa: Failed to open lock file
Apr 10 20:44:27 ubuntu-nuc libmraa[1003]: libmraa initialised for platform 'Unknown platform' of type 98
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: dfu-util 0.9
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Copyright 2010-2016 Tormod Volden and Stefan Schmidt
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: This program is Free Software and has ABSOLUTELY NO WARRANTY
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Please report bugs to http://sourceforge.net/p/dfu-util/tickets/
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: dfu-util: Invalid DFU suffix signature
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Opening DFU capable USB device...
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: ID 8087:0aba
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Run-time device DFU version 0011
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Claiming USB DFU Interface...
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Setting Alternate Setting #7 ...
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Determining device status: state = dfuIDLE, status = 0
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: dfuIDLE, continuing
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: DFU mode device DFU version 0011
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Device returned transfer size 2048
Apr 10 20:44:29 ubuntu-nuc imraa[1003]: Copying data from PC to DFU device
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: [1.7K blob data]
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Download done.
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Sent a total of 62876 bytes
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: state(2) = dfuIDLE, status(0) = No error condition is present
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Done!
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: dfu-util: can't detach
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Resetting USB to switch back to runtime mode
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: dfu-util: error resetting after download
Apr 10 20:44:31 ubuntu-nuc systemd[1]: mraa-imraa.service: Main process exited, code=exited, status=1/FAILURE
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: imraa version is 1.4.0 good
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: dfu location: /usr/bin
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: lock file location: /tmp/imraa.lock
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Starting to flash board
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Ardunio 101 Device Node Path: /dev/ttyACM0
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Waiting for device...
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Waiting for device...
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Device found!
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: flash cmd: /usr/bin/dfu-util -d ,8087:0ABA -D /usr/share/mraa/firmata101.ino.bin -v -a 7 -R
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: ERROR: Upload failed on /dev/ttyACM0
Apr 10 20:44:31 ubuntu-nuc systemd[1]: Failed to start imraa.
Apr 10 20:44:31 ubuntu-nuc systemd[1]: mraa-imraa.service: Unit entered failed state.
Apr 10 20:44:31 ubuntu-nuc systemd[1]: mraa-imraa.service: Failed with result 'exit-code'.
Apr 10 20:45:11 ubuntu-nuc libmraa[1278]: libmraa version v1.6.1 initialised by user 'root' with EUID 0
Apr 10 20:45:11 ubuntu-nuc libmraa[1278]: Platform not supported, not initialising
Apr 10 20:45:11 ubuntu-nuc libmraa[1278]: imraa: Failed to open lock file
Apr 10 20:45:11 ubuntu-nuc libmraa[1278]: libmraa initialised for platform 'Unknown platform' of type 98
Apr 10 20:45:19 ubuntu-nuc libmraa[1278]: gpio525: init: Using sub platform
Apr 10 20:45:19 ubuntu-nuc libmraa[1278]: gpio525: init: Sub platform not initialised
Apr 10 20:45:23 ubuntu-nuc libmraa[1278]: gpio: init: pin 13 beyond platform pin count (0)
Apr 10 20:45:27 ubuntu-nuc libmraa[1278]: gpio525: init: Using sub platform
Apr 10 20:45:27 ubuntu-nuc libmraa[1278]: gpio525: init: Sub platform not initialised
Apr 10 20:45:35 ubuntu-nuc libmraa[1285]: libmraa version v1.6.1 initialised by user 'root' with EUID 0
Apr 10 20:45:35 ubuntu-nuc libmraa[1285]: Platform not supported, not initialising
Apr 10 20:45:35 ubuntu-nuc libmraa[1285]: imraa: Failed to open lock file
Apr 10 20:45:35 ubuntu-nuc libmraa[1285]: libmraa initialised for platform 'Unknown platform' of type 98
Apr 10 20:46:22 ubuntu-nuc libmraa[1285]: mraa: Added firmata subplatform
Apr 10 20:46:57 ubuntu-nuc libmraa[1285]: gpio525: init: Using sub platform

Propanu, 2017-04-11:
@w4ilun is this with Node.js and MRAA installed via NPM? If that's the case it will never work unless you manually add the subplatform, because node-gyp builds its own mraa.node with a bundled static libmraa. C/C++ and Python should add it automatically though.

w4ilun, 2017-04-11:
@Propanu I tried this with C and Python after a clean install of Ubuntu and MRAA. I suspect it has something to do with the lockfile

Propanu, 2017-05-24:
@w4ilun, @tingleby, I figured it out. The udev rules that were added don't help in this case, because dfu fails on detaching the board, when switching back from DFU mode, as can be clearly seen from here:
```
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: dfu-util: can't detach
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: Resetting USB to switch back to runtime mode
Apr 10 20:44:31 ubuntu-nuc imraa[1003]: dfu-util: error resetting after download
```
And there's nothing preventing ModemManager to go bananas when that happens. In fact, if you look at the logs on a clean install you'll see the board keeps resetting.
The simplest solution I found would be to change `60-mraa-imraa.rules` to:
```
# Arduino 101
SUBSYSTEM=="usb", ATTR{idVendor}=="8087", ATTR{idProduct}=="0ab6", ENV{ID_MM_DEVICE_IGNORE}="1"
```
That's it, after the change I was able to see the subplatform added consistently over several reboots. No bugs in imraa/firmata code.

tingleby, 2017-05-24:
@Propanu so...
```
# Arduino 101 in DFU Mode
SUBSYSTEM=="tty", ENV{ID_REVISION}=="8087", ENV{ID_MODEL_ID}=="0ab6", MODE="0666", ENV{ID_MM_DEVICE_IGNORE}="1", ENV{ID_MM_CANDIDATE}="0"
SUBSYSTEM=="usb", ATTR{idVendor}=="8087", ATTR{idProduct}=="0aba", MODE="666", ENV{ID_MM_DEVICE_IGNORE}="1"
```
Should be replaced with?
```
# Arduino 101
SUBSYSTEM=="usb", ATTR{idVendor}=="8087", ATTR{idProduct}=="0ab6", ENV{ID_MM_DEVICE_IGNORE}="1"
```

Does this not effect other usages of the 101 too?
The rules as they are now are based on https://github.com/01org/corelibs-arduino101/blob/master/scripts/create_dfu_udev_rule

Is that wrong?

Propanu, 2017-05-24:
I've noticed the Arduino IDE rules, but in our case flashing works just fine. It's after that we run into issues. They're not wrong, just not relevant to our flow. I tested the process with all 3 rules in there, and it works like that too.

TBH, I've never heard of anyone wanting to use the 101 as a modem and I'd say it's a very remote scenario (especially if you install IMRAA right?). Best we can do is document this properly in IMRAA's documentation and perhaps add an extra comment in the rules file mentioning it:
```
# The device won't be usable as a modem with this rule in place
```

What do you think?

Propanu, 2017-05-25:
So removing the previous rules might affect flashing the board manually, I've pretty much relied on the service in my case and that's a pass. It might be best to keep all 3 rules in there after all.

tingleby, 2017-06-01:
Ok this is being built on the PPA right now.

tingleby, 2017-09-27:
@skiselev does this look resolved to you now?