eclipse-mraa/mraa #634: EDISON Error: Illegal arguments for construction of _exports_Gpio

Niko-La, 2017-01-21:
Running Intel Eddison using MRAA for interrupt. Code was working fine till today. It gives 

Error: Illegal arguments for construction of _exports_Gpio

MRAA Version: v1.1.1 also did a NPM install for the latest version but still getting same error

21.01.17 14:35:12 [-0500] /usr/src/app/main.js:100
21.01.17 14:35:12 [-0500] InterruptPin2 = new mraa.Gpio(32) // mraa8 -gp28
21.01.17 14:35:12 [-0500] ^
21.01.17 14:35:12 [-0500]
21.01.17 14:35:12 [-0500] Error: Illegal arguments for construction of _exports_Gpio
21.01.17 14:35:12 [-0500] at Error (native)
21.01.17 14:35:12 [-0500] at Object.<anonymous> (/usr/src/app/main.js:100:17)
21.01.17 14:35:12 [-0500] at Module._compile (module.js:570:32)
21.01.17 14:35:12 [-0500] at Object.Module._extensions..js (module.js:579:10)
21.01.17 14:35:12 [-0500] at Module.load (module.js:487:32)
21.01.17 14:35:12 [-0500] at tryModuleLoad (module.js:446:12)
21.01.17 14:35:12 [-0500] at Function.Module._load (module.js:438:3)
21.01.17 14:35:12 [-0500] at Module.runMain (module.js:604:10)
21.01.17 14:35:12 [-0500] at run (bootstrap_node.js:394:7)
21.01.17 14:35:12 [-0500] at startup (bootstrap_node.js:149:9)

alext-mkrs, 2017-01-24:
If the code was working - what has changed "today" in your environment? OS, Node, whatvere else updates/reconfigs?

alext-mkrs, 2017-02-25:
@Niko-La, is this still actual?

alext-mkrs, 2017-03-05:
Closing for no response, feel free to post here if this is still actual.

closed 2017-03-05 by alext-mkrs