eclipse-mraa/mraa #640: test: only accept 12 bit adc resolution

nicolalunghiemutex, 2017-02-06:
in test/platform_checks
```
class PlatformChecks(u.TestCase):
  def test_mraa_check_platform_no_of_pins(self):
    pinCount = m.getPinCount()
    self.assertEqual(pinCount, 20, "Wrong number. of pins. Check platform ...")

  def test_mraa_check_platform_ADC_max_resolution(self):
    self.p_ADC_mres = m.adcRawBits()
    print("Platform ADC max. resolution is: " + str(self.p_ADC_mres) + " bits")
    self.assertEqual(self.p_ADC_mres, 12, "Wrong ADC max. resolution. Check platform ...")

  def test_mraa_check_platform_ADC_resolution(self):
    self.p_ADC_res = m.adcSupportedBits()
    print("Platform ADC resolution is: " + str(self.p_ADC_res) + " bits")
self.assertEqual(self.p_ADC_res, 10, "Wrong ADC suported resolution. Check platform ...")
```

This is very wrong test! only 10 bit adc and 20 pin count??

alext-mkrs, 2017-02-06:
These are supposed to be run on a specific HW, Edison, I think.

nicolalunghiemutex, 2017-02-06:
there's also a script in mock that does the same except a par of "defines"

```
PLATFORM_PINCOUNT = 10
PLATFORM_STD_ADC_RES_BITS = 10
PLATFORM_MAX_ADC_RES_BITS = 12

class PlatformChecks(u.TestCase):
  def test_platform_pin_count(self):
    self.assertEqual(m.getPinCount(), PLATFORM_PINCOUNT, "Wrong number of pins reported by platform")

  def test_adc_std_res(self):
    adc_std_res = m.adcSupportedBits()
    print("Platform ADC standard resolution is: " + str(adc_std_res) + " bits")
    self.assertEqual(adc_std_res, PLATFORM_STD_ADC_RES_BITS, "Wrong ADC standard resolution")

  def test_adc_max_res(self):
    adc_max_res = m.adcRawBits()
    print("Platform ADC max. resolution is: " + str(adc_max_res) + " bits")
    self.assertEqual(adc_max_res, PLATFORM_MAX_ADC_RES_BITS, "Wrong ADC max. resolution")

```

alext-mkrs, 2017-02-07:
...because of the same thing - mock platform simulates a 10/12 bit ADC. What exact trouble does this cause you?

nicolalunghiemutex, 2017-02-08:
Sorry I've not understand what the mock platform is doing

alext-mkrs, 2017-02-13:
So you've closed and reopened it again. Do you have any further questions in this one?

nicolalunghiemutex, 2017-02-21:
maybe change the test with
```
@u.skipUnless(m.getPlatformName() == "Intel Edison", "Only for edison Board")
class PlatformChecksEdison(u.TestCase):
```

alext-mkrs, 2017-02-22:
That makes certain sense to me, but you still haven't shared what your end goal is :) If I know what are you coming at - i.e. just spotted a test, which is not universal and want to fix, or have a plan to add platforms/tests and want to create a better structure, etc - then I might be able to advise  more meaningfully.

nicolalunghiemutex, 2017-02-22:
Ok the plan is
I currently am developing support for a new board (up2) and we are developing an automated test suite
I have the board in place and I want to test mraa functionality and board specific functionality
I also have another C test suite that links to mraa and does integration test with other hats

I think that mraa should have a "generic" test part that test the generic linux library functions and a "platform" testing that is platform dependant

test
- common/mock
---- gpio
----basic function
----python check

-platform/platform1
----gpio
----uart
----other

and maybe the test could be switched on/off or via python support or with some CMAKE function

set(PLATFORM)

if(PLATFORM)
add_executable
add_test
...

arfoll, 2017-02-22:
That's pretty cool, however I think you're making this a bit too complex. Instead of hiding it under cmake, just put some guards on your tests, pu tthem in tests/up2/xxx and then they will only run on an up2? Adding too  much in cmake tends to break and requires alot of babying, so I'm trying to avoid throwing too much in there.

The problem with 'generic' tests is that unless we run them with mock, they don't mean much it's very hard to test anything since we don't know what's what etc... Although we could make tests that test every pin can be turned on etc, I guess we can query that kind of info...

alext-mkrs, 2017-02-23:
Yeah, +1 on that.

@nikolalunghiemutex, the problem with "common" you mentioned is there's really not much "common" or generic - aside from maybe getVersion() :) I'd second @arfoll's suggestion for putting your tests into a separate dir and plumbing it up in cmake. For the latter see e.g. how I've added tests for mock platform.

nicolalunghiemutex, 2017-02-23:
That's not true the common part is regardin linux part

see for example 

https://github.com/vsergeev/c-periphery/blob/master/tests/test_gpio.c
https://github.com/vsergeev/c-periphery/blob/master/tests/test_serial.c

that could be tested with the "mock" platform and is independent of the particular board

test for existance of gpio subsystem/spi/adc/mem
test for properly closing/opening gpio, set interrupt
test for permission?
check for initializing/deinitializing gpio
check for properly switching pin to different function, (ex I have a pin that is declared GPIO but is shared with spi function what happens when I request to enable spi?)

alext-mkrs, 2017-03-05:
Still not sure I understand what you're after here. From the mraa prospective, mock platform is still... a platform, i.e. it has specific parts. And it already has a comprehensive set of tests, for all of the features implemented.

alext-mkrs, 2017-05-20:
As there were other discussions on this and even some pull requests - @nicolalunghiemutex, is this one still actual?