eclipse-mraa/mraa #642: Refactor function for board definition

nicolalunghiemutex, 2017-02-07:
Some board (for example /src/x86/up.c) folder use these two functions

could be useful to insert those in a "global" file and then refactor code to use it
```
mraa_up_set_pininfo(b, 3, "I2C1_SDA", (mraa_pincapabilities_t){ 1, 1, 0, 0, 0, 1, 0, 0 }, 

mraa_up_get_pin_index(b, "I2C1_SDA", (int*) &(b->i2c_bus[0].sda));

```

like

```
static mraa_result_t
mraa_get_pin_index_from_name(mraa_board_t* board, char* name, int* pin_index)
{
    int i;
    for (i = 0; i < board->phy_pin_count; ++i) {
        if (strncmp(name, board->pins[i].name, MAX_LENGTH) == 0) {
            *pin_index = i;
            return MRAA_SUCCESS;
        }
    }
    return MRAA_ERROR_INVALID_RESOURCE;
}

mraa_set_pininfo(mraa_board_t* board, int mraa_index, char* name, mraa_pincapabilities_t caps, int sysfs_pin)
{
    if (mraa_index < board->phy_pin_count) {
        mraa_pininfo_t* pin_info = &board->pins[mraa_index];
        strncpy(pin_info->name, name, MAX_LENGTH);
        pin_info->capabilities = caps;
        if (caps.gpio) {
            pin_info->gpio.pinmap = sysfs_pin;
            pin_info->gpio.mux_total = 0;
        }
        if (caps.i2c) {
            pin_info->i2c.pinmap = 1;
            pin_info->i2c.mux_total = 0;
        }
        if (caps.pwm) {
            int controller = 0;
            if (strncmp(name, "PWM", 3) == 0 && strlen(name) > 3 && isdigit(name[3]))
                controller = name[3] - '0';
            pin_info->pwm.parent_id = (unsigned int) controller;
            pin_info->pwm.pinmap = 0;
            pin_info->pwm.mux_total = 0;
        }
        if (caps.spi) {
            pin_info->spi.mux_total = 0;
        }
        if (caps.aio) {
            pin_info->aio.mux_total = 0;
            pin_info->aio.pinmap = 0;
        }
        if (caps.uart) {
            pin_info->uart.mux_total = 0;
        }
        return MRAA_SUCCESS;
    }
    return MRAA_ERROR_INVALID_RESOURCE;
}
```

where is the most appropriate place to insert those?

arfoll, 2017-02-07:
So actually I think those should go in a new file like src/board.c and headers in include/board.h and then have all x86/ arm/ mock/ etc... boards able to use them. This should make it easier to write board files in mraa.

If you'd like to give that a go I'd happily merge it :)

nicolalunghiemutex, 2017-02-08:
I can do this but not this week :-)
Please assign to me and then I will send the patch to review