eclipse-mraa/mraa #651: Raspberry PI and AD8232

paull5, 2017-02-20:
Hi,
I am trying to program an AD8232 heart monitor using a Raspberry PI, I am using code from https://github.com/intel-iot-devkit/upm/blob/master/examples/python/ad8232.py. As far as I can make out I have all the right packages installed for ump and ADC ads1115. The problem I am having is when I try to run the code I get a Value Error:
File "/usr/lib/python2.7/dist-packages/upm/pyupm_ad8232.py", line 170, in __init__
    this = _pyupm_ad8232.new_AD8232(loPlus, loMinus, output, aref)
ValueError: UPM Invalid Argument: Invalid AIO pin specified - do you have an ADC?
Can you help with this problem,

Thank you.

arfoll, 2017-02-20:
Unfortunately it's not so simple as adding an ads1115 ADC and then connecting the AD8232 code together. You need in order for that to work to make mraa understand your ADC, which as of now is non trivial as we don't support ADCs on boards without them (it's somethign we'd like to do but haven't yet).

arfoll, 2017-02-20:
@Propanu @pylbert What would be really cool would be the ability to define either an IIO sensor to act as the mraa ADC or a UPM sensor to act as the underlying AIO driver for mraa. Food for thought :)

paull5, 2017-02-20:
Hi, Thank you i appreciate your help. Do you happen to know of a way I could use the AD8232 with the PI, or would I be better to use a different chip altogether? I am doing my final year of a computer science degree and this is part of my project, I really need all the help I can get. 
Thank.

arfoll, 2017-02-20:
@paull5 so you either need to load an IIO kernel module for your ads1115, then modify the correct raspberry pi mraa board file in src/arm/* to add the correct Aio entries and rebuild mraa. That would let you use the AD8232 with mraa/UPM.

Or you could load the UPM module for the ADS1115, then reimplement the AD8232 driver above the ADS1115.

paull5, 2017-02-20:
Ok thank you so much @arfoll I am very grateful for your advice.

whbruce, 2017-02-20:
@arfoll, @Propanu, @pylbert: I recently had a request for an ADS1115 to appear as AIO for a Minnowboard. A shared library exposing adv_func calls could be a neat solution to mraa AIO "plug-ins"??

arfoll, 2017-05-31:
Closing this as more of a missing feature/question at this point.

closed 2017-05-31 by arfoll

connect-javed001, 2018-04-26:
Hello guys, I am trying to program an AD8232 heart monitor using a Raspberry PI in python.
how to install upm module in raspberry pi for python?

dan119, 2018-11-30:
hi, did you manage to start the cardiograph?

Umairqureshi97, 2020-02-15:
@paull5 hi i am currently working on the same project as you could you tell me how did you connect  the ad8232 with rasberry pi i am stuck and i cant find a way to connect them.