eclipse-mraa/mraa #656: error with mraa library on Python from Galileo

luisafernandarj, 2017-02-27:
Hi man, I need your help,

When I try to use mraa library for python, I have this error:

import mraa
Traceback (most recent call last):
File “”, line 1, in
File “/usr/lib/python2.7/site-packages/mraa.py”, line 32, in
_mraa = swig_import_helper()
File “/usr/lib/python2.7/site-packages/mraa.py”, line 28, in swig_import_helper
_mod = imp.load_module(‘_mraa’, fp, pathname, description)
ImportError: File not found

I used the steps in this page: https://github.com/intel-iot-devkit/mraa/blob/master/README.md
Installing on Intel 32bit Yocto based opkg image

For true, I’ve Intel Galileo 1Gen, and this SDImage is this: https://software.intel.com/en-us/iot/hardware/galileo/downloads

Latest Yocto* Poky Image (210 MB)

Can you help me please?, Thanks!

arfoll, 2017-02-27:
> devkit/mraa/blob/master/README.md

What's the output of `ls -l /usr/lib/python2.7/site-packages/*mraa*`?

It looks like you havent got mraa installed correctly and are missing the
native part of the python module. I'm not sure why though.

luisafernandarj, 2017-02-27:
Uhhmmm... well.. this is the output...

root@clanton:/usr/lib/python2.7/site-packages# ls -l /usr/lib/python2.7/site-packages/*mraa*
-rwxr-xr-x    1 root     root        263924 Feb 22  2017 /usr/lib/python2.7/site-packages/_mraa.so
-rw-r--r--    1 root     root         43310 Jan  1 04:58 /usr/lib/python2.7/site-packages/mraa.py
-rw-r--r--    1 root     root         49077 Jan  1 04:52 /usr/lib/python2.7/site-packages/mraa.pyo

How I can install mraa, so? 
:(

arfoll, 2017-02-27:
So that looks very odd because it seems you have an updated _mraa.so but not an updated mraa.py by looking at the timestamps mraa.py should have the same as _mraa.so since that should be installed time. I'd recommend removing mraa from the package manager and installing it again, no idea what's gone wrong. Installing/compiling from source should give you a quick fix.

However i'm curious what's the ouput of `cat /etc/os-release`?

luisafernandarj, 2017-02-27:
I try to remove and installing again, with before comand, this is the output:

root@clanton:/usr/lib/python2.7/site-packages# ls -l /usr/lib/python2.7/site-packages/*mraa*
-rwxr-xr-x    1 root     root        263924 Feb 22  2017 /usr/lib/python2.7/site-packages/_mraa.so
-rw-r--r--    1 root     root         43310 Feb 22  2017 /usr/lib/python2.7/site-packages/mraa.py
root@clanton:/usr/lib/python2.7/site-packages# python
Python 2.7.3 (default, Oct 16 2014, 16:04:29) 
[GCC 4.7.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import mraa
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "mraa.py", line 32, in <module>
    _mraa = swig_import_helper()
  File "mraa.py", line 28, in swig_import_helper
    _mod = imp.load_module('_mraa', fp, pathname, description)
ImportError: File not found

I tryed to install mraa in my PC, and had the same problem, only difference was the sentence: ImportError: libmraa.1, so I installed from this page http://ppa.launchpad.net/mraa/mraa/ubuntu/pool/main/m/mraa/ , and work it... but in this case, I don't know which file is not there...

The output of cat /etc/os-release is:
root@clanton:/usr/lib/python2.7/site-packages# cat /etc/os-release
cat: can't open '/etc/os-release': No such file or directory

Thanks for you help!!!

arfoll, 2017-02-27:
Ok my guess is that you haven't flash the uSD card correctly and are on the internal SPI image - check out the instructions here: https://software.intel.com/en-us/get-started-galileo-linux-step1 (there are some for windows too) you need to use dd to flash the image you cannot just copy it to the uSD card.

luisafernandarj, 2017-02-27:
Okay, I'll try to install Linux again in the sd, and try the mraa library again. I will return to tell if it worked,

Thank you!!

luisafernandarj, 2017-02-27:
Yeah!!, That is The Solution, download the image and install it Completely. I forgot that I was using a copy-paste version to run only little sketch of Arduino and no-Python programs of the Intel Galileo.

root@galileo:~# cat /etc/os-release
ID=iot-devkit
NAME=iot-devkit (Intel IoT Development Kit)
VERSION=3.5 (tarsier)
VERSION_ID=3.5
PRETTY_NAME=iot-devkit (Intel IoT Development Kit) 3.5 (tarsier)

arfoll, 2017-02-27:
Glad that worked. Closing issue.

closed 2017-02-27 by arfoll